ZoneMinder/zoneminder #3451: No input for path, width, or height available when initiating new camera on raspberry pi

mkstarr, 2022-03-08:
**Describe Your Environment**
- release version zm
- installed using Download > Debian
- OS: raspbian 10
- browser: chromium Version 92.0.4515.98 

**Describe the bug**
No input area for path, width, or height but input required.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to ['...'](http://localhost/zm/index.php
2. Click on Console, ADD
3. Scroll down to bottom of screen, hit Save
4. See error

**Expected behavior**
For there to be preset path, width, and height fields to enter info, so that when I hit Save it proceeds normally.

**Debug Logs**
```
2022-03-08 08:12:36 | web_php |   | 1503 | WAR | Unknown function call Monitor->Importance from /usr/share/zoneminder/www/index.php:302 | /usr/share/zoneminder/www/includes/Monitor.php | 216
-- | -- | -- | -- | -- | -- | -- | --

```

hobson42, 2024-02-06:
This bug applies to zoneminder on Ubunto 22.04.03 with latest zoneminder v1.36.33.

This is a serious problem because it stops any new installation being set up.

connortechnology, 2024-02-06:
There are tabs at the left.  You need to go through them and fill out remaining info. 

Granted it would be nice if it automatically took you to the relevant tab.