ZoneMinder/zoneminder #3457: zmc gets signal 6 (aborted) for RTSP cam

hvillemoes, 2022-03-14:
**Describe Your Environment**
- Version of ZoneMinder [release version, development version, or commit] v.1.36.12
- How you installed ZoneMinder [e.g. PPA, RPMFusion, from-source, etc] https://zmrepo.zoneminder.com/debian/release-1.36
- Full name and version of OS Debian 10
- Browser name and version (if this is an issue with the web interface)

**If the issue concerns a camera**
- Make and Model pi zero with v4l2rtspserver
- frame rate 29.97 
- resolution 1280 x 720
- ZoneMinder Source Type: RTSP

**Describe the bug**
A clear and concise description of what the bug is.
No image - "Monitor is not capturing. We will be unable to provide an image"
Please note in the debug log: "signal 6 (Aborted), crashing"

VLG and ContaCam displays same cam flawless.
All hints are welcome.

**To Reproduce**
Steps to reproduce the behavior: In my environment the error is static.
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.
Images displayed.

**Debug Logs**
03/14/22 17:25:40.995395 zmc_m9[993].DB2-zm_remote_camera_rtsp.cpp/130 [Waiting for sources]
03/14/22 17:25:41.097112 zmc_m9[995].DB1-zm_comms.cpp/471 [connect(): Trying '10.9.10.49', family '2', proto '6']
03/14/22 17:25:41.119372 zmc_m9[995].DB2-zm_rtsp.cpp/43 [Sending RTSP message: OPTIONS rtsp://piz1.l.vrc.dk:8554/unicast RTSP/1.0
User-Agent: ZoneMinder/1.36.12
CSeq: 1

]
03/14/22 17:25:41.122705 zmc_m9[995].DB2-zm_rtsp.cpp/63 [Received RTSP response: RTSP/1.0 200 OK
CSeq: 1
Date: Mon, Mar 14 2022 16:25:41 GMT
Public: OPTIONS, DESCRIBE, SETUP, TEARDOWN, PLAY, PAUSE, GET_PARAMETER, SET_PARAMETER

 (152 bytes)]
03/14/22 17:25:41.122724 zmc_m9[995].DB2-zm_rtsp.cpp/43 [Sending RTSP message: DESCRIBE rtsp://piz1.l.vrc.dk:8554/unicast RTSP/1.0
User-Agent: ZoneMinder/1.36.12
CSeq: 2

]
03/14/22 17:25:41.132835 zmc_m9[995].ERR-zm_signal.cpp/50 [Got signal 6 (Aborted), crashing]
14/03/22 17:25:50.678596 zmstats[728].DB1 [main:84] [Deleted 0E0 log table entries by time]
14/03/22 17:25:50.679690 zmstats[728].DB1 [main:93] [Deleted 0E0 sessions]
14/03/22 17:25:50.945111 zmwatch[716].DB3 [ZoneMinder::Memory:309] [Shared data appears valid for monitor 9: 1]
14/03/22 17:25:50.945162 zmwatch[716].DB1 [main:110] [Monitor 9 LastWriteTime is 0.]

Thank you in advance

/ Harald

```

connortechnology, 2022-03-31:
Please test with the just released 1.36.13. I suspect it will fix this issue.

connortechnology, 2022-04-06:
I should also mention that we are only just barely maintaining the remote rtsp monitor type.  You should switch to the ffmpeg monitor type.

hvillemoes, 2022-04-08:
Thanks - 1.36.14 made it work.
Thank you for a fine piece of software.

closed 2022-04-08 by hvillemoes