ZoneMinder/zoneminder #3458: nph-zms hangs, no web streaming until I do killall nph-zms

acsawdey, 2022-03-18:
**Describe Your Environment**
- Version of ZoneMinder 1.32.12
- How you installed ZoneMinder: deb http://ppa.launchpad.net/iconnor/zoneminder-1.36/ubuntu focal main
- Full name and version of OS Ubuntu 20.04.4 LTS
- Browser name and version (if this is an issue with the web interface) Safari 15.3
This is running on an old supermicro 2U server box with 2x Xeon E5620 and 32GB memory. 
Web server is lighttpd so it can coexist with Pihole.

**If the issue concerns a camera**
- Make and Model anpviz IPC-D283
- frame rate 5
- resolution 3840x2160
- ZoneMinder Source Type: rtsp

**Describe the bug**
Web UI that includes streaming images doesn't work unless you kill off the existing nph-zms processes.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to console view
2. Click on monitor name
3. Monitor view comes up with fps figures below but no actual image.4. 
4. There will be one or more nph-zms processes:
```
root@sigma957:/var/log/lighttpd# ps auxww|grep nph-zms
www-data  594255  0.0  0.0 274012 27820 ?        SLl  18:47   0:00 /usr/lib/cgi-bin/nph-zms
www-data  598442 11.8  0.6 468460 222840 ?       SLl  20:00   0:06 /usr/lib/cgi-bin/nph-zms
```
5. run `killall nph-zms` and the streaming images will now start.6. 

I attached gdb to a nph-zms that seemed stuck, there were two threads with not very interesting backtraces:
```
(gdb) info threads
  Id   Target Id                                    Frame 
* 1    Thread 0x7f3afcc28580 (LWP 594255) "nph-zms" __pthread_clockjoin_ex (threadid=139891325409024, thread_return=0x0, clockid=<optimized out>, abstime=<optimized out>, block=<optimized out>)
    at pthread_join_common.c:145
  2    Thread 0x7f3afcc21700 (LWP 594256) "nph-zms" futex_wait_cancelable (private=<optimized out>, expected=0, futex_word=0x56373dfb3688) at ../sysdeps/nptl/futex-internal.h:183
(gdb) thread 2
[Switching to thread 2 (Thread 0x7f3afcc21700 (LWP 594256))]
#0  futex_wait_cancelable (private=<optimized out>, expected=0, futex_word=0x56373dfb3688) at ../sysdeps/nptl/futex-internal.h:183
183     ../sysdeps/nptl/futex-internal.h: No such file or directory.
(gdb) where
#0  futex_wait_cancelable (private=<optimized out>, expected=0, futex_word=0x56373dfb3688) at ../sysdeps/nptl/futex-internal.h:183
#1  __pthread_cond_wait_common (abstime=0x0, clockid=0, mutex=0x56373dfb3638, cond=0x56373dfb3660) at pthread_cond_wait.c:508
#2  __pthread_cond_wait (cond=0x56373dfb3660, mutex=0x56373dfb3638) at pthread_cond_wait.c:638
#3  0x00007f3b0461ce30 in std::condition_variable::wait(std::unique_lock<std::mutex>&) () at /lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x000056373de5087f in  ()
#5  0x00007f3b04622de4 in  () at /lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007f3b06a1a609 in start_thread (arg=<optimized out>) at pthread_create.c:477
#7  0x00007f3b0430f163 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
(gdb) thread 1
[Switching to thread 1 (Thread 0x7f3afcc28580 (LWP 594255))]
#0  __pthread_clockjoin_ex (threadid=139891325409024, thread_return=0x0, clockid=<optimized out>, abstime=<optimized out>, block=<optimized out>) at pthread_join_common.c:145
145     pthread_join_common.c: No such file or directory.
(gdb) where
#0  __pthread_clockjoin_ex (threadid=139891325409024, thread_return=0x0, clockid=<optimized out>, abstime=<optimized out>, block=<optimized out>) at pthread_join_common.c:145
#1  0x00007f3b04623047 in std::thread::join() () at /lib/x86_64-linux-gnu/libstdc++.so.6
#2  0x000056373de45886 in  ()
#3  0x00007f3b042140b3 in __libc_start_main (main=0x56373de44760, argc=1, argv=0x7ffcc7111e78, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7ffcc7111e68)
    at ../csu/libc-start.c:308
#4  0x000056373de4718e in  ()
```

 **Expected behavior**
Streaming images should work.

**Debug Logs**
```
root@sigma957:/var/log/zm# less zms_e9244.log
03/17/22 19:48:05.475840 zms_e9244[597709].ERR-zm_ffmpeg_input.cpp/190 [Unable to decode frame at frame 0: -11 Resource temporarily unavailable, continuing]
root@sigma957:/var/log/zm# less zms_e8854.log
03/15/22 12:50:45.065556 zms_e8854[454941].ERR-zm_ffmpeg_input.cpp/190 [Unable to decode frame at frame 0: -11 Resource temporarily unavailable, continuing]

```

acsawdey, 2022-03-18:
One other observation: montage review view does work without any fiddling but that is calling nph-zms with mode=single that seems just to return a single frame then exit.

connortechnology, 2022-03-23:
Will you please try out the proposed 1.36.13. I believe there is a fix in there for this.

deb http://ppa.launchpad.net/iconnor/zoneminder-proposed/ubuntu focal main

acsawdey, 2022-03-24:
Thanks for the build!

This is what I got installed:

```root@sigma957:/etc/apt/sources.list.d# apt info zoneminder
Package: zoneminder
Version: 1.36.12~20220323.49-focal1
Priority: optional
Section: net
Maintainer: Isaac Connor <isaac@zoneminder.com>
Installed-Size: 37.0 MB
```

Still seeing the same thing. nph-zms is running, the `State` modeline underneath is updating, but no video. Doing `killall nph-zms` results in one new frame being displayed.

If I get a chance I'll see if I can build that binary with symbols and poke at it a bit more.

connortechnology, 2022-04-05:
Please update to 1.36.14 and see if it fixes this issue.

acsawdey, 2022-04-06:
No luck with that.  Behavior is the same as before. Also with debug enabled from options->logging this is what we see:

```04/05/22 20:32:26.588853 zms_m1[1342231].DB1-zm_logger.cpp/224 [LogOpts: level=DB1 effective=DB1, screen=OFF, database=INF, logfile=DB9->/var/log/zm/zms_m1.log, syslog=INF]
04/05/22 20:32:26.588880 zms_m1[1342231].DB1-zm_utils.cpp/204 [Detected a x86\x86-64 processor with SSE4.2]
04/05/22 20:32:26.589955 zms_m1[1342231].DB1-zm_storage.cpp/80 [Loaded Storage area 2 'surveillance']
04/05/22 20:32:26.589980 zms_m1[1342231].DB1-zm_monitor.cpp/493 [Have camera type File]
04/05/22 20:32:26.589995 zms_m1[1342231].DB1-zm_monitor.cpp/2336 [Reloading linked monitors for monitor IPC-D283WD-S, '(null)']
04/05/22 20:32:26.590010 zms_m1[1342231].DB1-zm_packetqueue.cpp/660 [Setting pre_event_video_packet_count to 5]
04/05/22 20:32:26.590020 zms_m1[1342231].DB1-zm_packetqueue.cpp/654 [Setting max_video_packet_count to 300]
04/05/22 20:32:26.590030 zms_m1[1342231].DB1-zm_monitor.cpp/658 [mem.size(8) SharedData=760 TriggerData=560 VideoStoreData=4128 timestamps=48 images=3x33177600 = 99532800 total=99538360]
04/05/22 20:32:26.590043 zms_m1[1342231].DB1-zm_monitor.cpp/694 [Loaded monitor 1(IPC-D283WD-S), 0 zones]
04/05/22 20:32:26.590089 zms_m1[1342231].DB1-zm_monitor.cpp/968 [Unable to map file /dev/shm/zm.mmap.1 (99538360 bytes) to locked memory, trying unlocked]
04/05/22 20:32:26.590102 zms_m1[1342231].DB1-zm_monitor.cpp/971 [Mapped file /dev/shm/zm.mmap.1 (99538360 bytes) to unlocked memory]
04/05/22 20:32:26.590132 zms_m1[1342231].DB1-zm_stream.cpp/330 [Trying to open the lock on /run/zm/zms-998034.lock]
04/05/22 20:32:26.590161 zms_m1[1342231].DB1-zm_stream.cpp/353 [We have obtained a lock on /run/zm/zms-998034.lock fd: 7]
04/05/22 20:32:26.986123 zms_m1[1342231].DB1-zm_monitorstream.cpp/235 [Got QUERY command, sending STATUS]
04/05/22 20:32:28.014445 zms_m1[1342231].DB1-zm_monitorstream.cpp/235 [Got QUERY command, sending STATUS]
04/05/22 20:32:29.041306 zms_m1[1342231].DB1-zm_monitorstream.cpp/235 [Got QUERY command, sending STATUS]
04/05/22 20:32:30.165336 zms_m1[1342231].DB1-zm_monitorstream.cpp/235 [Got QUERY command, sending STATUS]
04/05/22 20:32:31.289469 zms_m1[1342231].DB1-zm_monitorstream.cpp/235 [Got QUERY command, sending STATUS]
04/05/22 20:32:32.496450 zms_m1[1342231].DB1-zm_monitorstream.cpp/235 [Got QUERY command, sending STATUS]
```

I nearly got 1.36.14 to build on my system, but I think ubuntu 20.04 has ffmpeg 5 and maybe zoneminder wants ffmpeg 4 libavcodec, at least that's the wisdom I find for the errors about a missing cur_dts member in AVCodec.

connortechnology, 2022-04-06:
ubuntu 20.04 does NOT have ffmpeg5.  You must have added from a ppa.  ffmpeg5 has made massive changes to the api that many projects depend on.  It's going to be a while before we see mass adoption of it. I think I can fix the cur_dts stuff fairly easily (and have commented it out in master).  But the AV_STRICT_TS stuff is needed..

You may have to crank up the debug level to at least 3.

connortechnology, 2022-10-27:
There have been several releases since this.  Please update and report.

closed 2023-01-24 by connortechnology