ZoneMinder/zoneminder #3459: NoDect issues after upgrade to v1.37.12

j-marz, 2022-03-20:
**Describe Your Environment**
- v1.37.12 
- PPA (deb http://ppa.launchpad.net/iconnor/zoneminder-master/ubuntu focal main)
- Ubuntu 20.04.4 LTS

**Describe the bug**
After upgrade to v1.37.12 i noticed high CPU usage on Zoneminder VM, which seemed to be caused by all cameras recording 10 minute long events every 10 minutes.

All cameras were previously configured with `nodect` function and triggered externally via zmtrigger events (20 second recording time). This option is no longer available in the??? Instead, the monitor is now set to `recording="on motion"` and `analysing="none"`

Another thing i've noticed is the before the upgrades, the motion events "cause" were limited to zmtrigger causes only, but now there are "motion: all" causes as well. So it looks like ZM is using built-in motion detection in addition to external zmtriggers, which may be leading to the long events.

I found this PR which changed the nodect function concept https://github.com/ZoneMinder/zoneminder/pull/3420

There doesn't appear to be a way to only record on external triggers anymore, which seems like a bug, but i could be missing something with the new function concept?

**To Reproduce**
Steps to reproduce the behavior:
1. configure monitor to use `nodect` in v1.37.11
2. upgrade to v1.37.12
3. monitor will be set to capturing=always, recording=on motion, analysing=none
4. zm motion detection will still record events irrespective of external zmtrigger

**Expected behavior**
Events should only record when triggered by zmtrigger event and only for the given time in the zmtrigger event (+ image buffers).

neillbell, 2022-03-22:
I'm curious about this one too. Most of my cameras are set up as NoDect and are triggered from linked low-resolution monitors using MoDect. I've had to fall back to 1.37.11 to keep things working.

j-marz, 2022-03-23:
Installed latest build and some cameras are now working correctly (motion only triggered by external zmtrigger) - i suspect due to https://github.com/ZoneMinder/zoneminder/commit/ed41e639809819829760184ed93ebb9455283e33. However, there are a few cameras that aren't recording motion from external zmtrigger. Logs state "Motion detection not enabled on monitor", but i can't spot a difference in monitor config between a camera that is working and one that isn't. 

Tagging @connortechnology given you made the commits and may have more info. :pray:

connortechnology, 2022-03-23:
That line should no longer exist. was replaced in 0850e0db70ef79bee93100597743c1d1b252d857.  Please update.

j-marz, 2022-03-23:
Thanks @connortechnology , i missed that commit. Hopefully it resolves the issue. When is the next PPA build scheduled?

neillbell, 2022-03-23:
I did some tests with the latest build and capturing triggered by linked monitors now works appropriately.

j-marz, 2022-03-24:
Confirming https://github.com/ZoneMinder/zoneminder/commit/0850e0db70ef79bee93100597743c1d1b252d857 fixed my remaining issue. Thanks @connortechnology :pray:

closed 2022-03-24 by j-marz