ZoneMinder/zoneminder #3460: Error with undefined 'AVPicture' when compiling 1.36.13

Nocifer, 2022-03-30:
**Environment**
Version: 1.36.13
OS: Arch Linux

**Describe the bug**
I'm getting this error (at the end of the snippet):

```
[ 76%] Building CXX object src/CMakeFiles/zm.dir/zm_packetqueue.cpp.o
/build/zoneminder/src/zoneminder-git/src/zm_ffmpeg_camera.cpp: In member function ‘int FfmpegCamera::OpenFfmpeg()’:
/build/zoneminder/src/zoneminder-git/src/zm_ffmpeg_camera.cpp:394:53: warning: invalid conversion from ‘const AVCodec*’ to ‘AVCodec*’ [-fpermissive]
  394 |     if ( (mVideoCodec = avcodec_find_decoder_by_name("h264_mmal")) == nullptr ) {
      |                         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~
      |                                                     |
      |                                                     const AVCodec*
/build/zoneminder/src/zoneminder-git/src/zm_ffmpeg_camera.cpp:402:39: warning: invalid conversion from ‘const AVCodec*’ to ‘AVCodec*’ [-fpermissive]
  402 |     mVideoCodec = avcodec_find_decoder(mVideoStream->
      |                   ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~
      |                                       |
      |                                       const AVCodec*
  403 | #if LIBAVCODEC_VERSION_CHECK(57, 64, 0, 64, 0)
      | ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  404 |         codecpar
      |         ~~~~~~~~                       
  405 | #else
      | ~~~~~                                  
  406 |         codec
      |         ~~~~~                          
  407 | #endif
      | ~~~~~~                                 
  408 |         ->codec_id);
      |         ~~~~~~~~~~~                    
/build/zoneminder/src/zoneminder-git/src/zm_ffmpeg_camera.cpp:536:45: warning: invalid conversion from ‘const AVCodec*’ to ‘AVCodec*’ [-fpermissive]
  536 |     if ( (mAudioCodec = avcodec_find_decoder(
      |                         ~~~~~~~~~~~~~~~~~~~~^
      |                                             |
      |                                             const AVCodec*
  537 | #if LIBAVCODEC_VERSION_CHECK(57, 64, 0, 64, 0)
      | ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  538 |             mAudioStream->codecpar->codec_id
      |             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ 
  539 | #else
      | ~~~~~                                        
  540 |             mAudioStream->codec->codec_id
      |             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~    
  541 | #endif
      | ~~~~~~                                       
  542 |             )) == nullptr ) {
      |             ~                                
[ 76%] Building CXX object src/CMakeFiles/zm.dir/zm_poly.cpp.o
/build/zoneminder/src/zoneminder-git/src/zm_mpeg.cpp: In member function ‘void VideoStream::SetupFormat()’:
/build/zoneminder/src/zoneminder-git/src/zm_mpeg.cpp:107:19: warning: invalid conversion from ‘const AVOutputFormat*’ to ‘AVOutputFormat*’ [-fpermissive]
  107 |         of = ofc->oformat;
      |              ~~~~~^~~~~~~
      |                   |
      |                   const AVOutputFormat*
/build/zoneminder/src/zoneminder-git/src/zm_mpeg.cpp: In member function ‘void VideoStream::SetupCodec(int, int, int, int, int, double)’:
/build/zoneminder/src/zoneminder-git/src/zm_mpeg.cpp:160:46: warning: invalid conversion from ‘const AVCodec*’ to ‘AVCodec*’ [-fpermissive]
  160 |     AVCodec *a = avcodec_find_encoder_by_name(codec_name);
      |                  ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~
      |                                              |
      |                                              const AVCodec*
/build/zoneminder/src/zoneminder-git/src/zm_mpeg.cpp:177:45: warning: invalid conversion from ‘const AVCodec*’ to ‘AVCodec*’ [-fpermissive]
  177 |                 codec = avcodec_find_encoder(codec_id);
      |                         ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~
      |                                             |
      |                                             const AVCodec*
/build/zoneminder/src/zoneminder-git/src/zm_mpeg.cpp: In member function ‘double VideoStream::ActuallyEncodeFrame(const uint8_t*, int, bool, unsigned int)’:
/build/zoneminder/src/zoneminder-git/src/zm_mpeg.cpp:618:23: warning: ‘void av_init_packet(AVPacket*)’ is deprecated [-Wdeprecated-declarations]
  618 |         av_init_packet( pkt );
      |         ~~~~~~~~~~~~~~^~~~~~~
In file included from /usr/include/libavcodec/avcodec.h:45,
                 from /build/zoneminder/src/zoneminder-git/src/zm_ffmpeg.h:127,
                 from /build/zoneminder/src/zoneminder-git/src/zm_mpeg.h:23,
                 from /build/zoneminder/src/zoneminder-git/src/zm_mpeg.cpp:20:
/usr/include/libavcodec/packet.h:506:6: note: declared here
  506 | void av_init_packet(AVPacket *pkt);
      |      ^~~~~~~~~~~~~~
/build/zoneminder/src/zoneminder-git/src/zm_mpeg.cpp:634:37: error: ‘AVPicture’ was not declared in this scope; did you mean ‘opicture’?
  634 |                 pkt->size = sizeof (AVPicture);
      |                                     ^~~~~~~~~
      |                                     opicture
```

I've turned -fpermissive errors into warnings, but the problem is that last error with AVPicture.

connortechnology, 2022-03-30:
You are trying to compile against ffmpeg 5.  See https://github.com/ZoneMinder/zoneminder/pull/3456

Nocifer, 2022-03-31:
Ah, I know about the issues with ffmpeg 5 but I saw this item "Fix compilation with ffmpeg 5.0" in the changelog of the new release and assumed the issues have been fixed. Alright then, If it's still a work in progress I'll just wait until it's ready, thanks.

connortechnology, 2022-04-05:
In 1.36.14 there are even more ffmpeg5 fixes.  Please give it a try.  It doesn't solve all the ffmpeg5 problems, but does fix the one listed here.

Nocifer, 2022-04-06:
The issue reported here seems indeed to be fixed, but the build still fails with an error about `AVOutputFormat::flags` being read-only (this also happens when building the latest master) and also with an error about `cur_dts` (which AFAICT by reading @exuvo's PR is a well known problem):
```
[ 94%] Building CXX object src/CMakeFiles/zm.dir/zm_storage.cpp.o
Manifying 22 pod documents
Manifying 23 pod documents
Manifying 23 pod documents
Manifying 21 pod documents
Manifying 21 pod documents
Manifying 22 pod documents
/build/zoneminder/src/zoneminder-git/src/zm_videostore.cpp: In member function ‘bool VideoStore::open()’:
/build/zoneminder/src/zoneminder-git/src/zm_videostore.cpp:143:22: error: assignment of member ‘AVOutputFormat::flags’ in read-only object
  143 |   oc->oformat->flags |= AVFMT_TS_NONSTRICT; // allow non increasing dts
      |                      ^
/build/zoneminder/src/zoneminder-git/src/zm_videostore.cpp:209:55: error: invalid conversion from ‘const AVCodec*’ to ‘AVCodec*’ [-fpermissive]
  209 |         video_out_codec = avcodec_find_encoder_by_name(codec_data[i].codec_name);
      |                           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                       |
      |                                                       const AVCodec*
/build/zoneminder/src/zoneminder-git/src/zm_videostore.cpp: In member function ‘void VideoStore::flush_codecs()’:
/build/zoneminder/src/zoneminder-git/src/zm_videostore.cpp:543:17: warning: ‘void av_init_packet(AVPacket*)’ is deprecated [-Wdeprecated-declarations]
  543 |   av_init_packet(&pkt);
      |   ~~~~~~~~~~~~~~^~~~~~
In file included from /usr/include/libavcodec/avcodec.h:45,
                 from /build/zoneminder/src/zoneminder-git/src/zm_ffmpeg.h:127,
                 from /build/zoneminder/src/zoneminder-git/src/zm_videostore.h:6,
                 from /build/zoneminder/src/zoneminder-git/src/zm_videostore.cpp:21:
/usr/include/libavcodec/packet.h:506:6: note: declared here
  506 | void av_init_packet(AVPacket *pkt);
      |      ^~~~~~~~~~~~~~
/build/zoneminder/src/zoneminder-git/src/zm_videostore.cpp: In member function ‘int VideoStore::writeVideoFramePacket(const std::shared_ptr<ZMPacket>&)’:
/build/zoneminder/src/zoneminder-git/src/zm_videostore.cpp:1128:19: warning: ‘void av_init_packet(AVPacket*)’ is deprecated [-Wdeprecated-declarations]
 1128 |     av_init_packet(&opkt);
      |     ~~~~~~~~~~~~~~^~~~~~~
In file included from /usr/include/libavcodec/avcodec.h:45,
                 from /build/zoneminder/src/zoneminder-git/src/zm_ffmpeg.h:127,
                 from /build/zoneminder/src/zoneminder-git/src/zm_videostore.h:6,
                 from /build/zoneminder/src/zoneminder-git/src/zm_videostore.cpp:21:
/usr/include/libavcodec/packet.h:506:6: note: declared here
  506 | void av_init_packet(AVPacket *pkt);
      |      ^~~~~~~~~~~~~~
/build/zoneminder/src/zoneminder-git/src/zm_videostore.cpp:1194:19: warning: ‘void av_init_packet(AVPacket*)’ is deprecated [-Wdeprecated-declarations]
 1194 |     av_init_packet(&opkt);
      |     ~~~~~~~~~~~~~~^~~~~~~
In file included from /usr/include/libavcodec/avcodec.h:45,
                 from /build/zoneminder/src/zoneminder-git/src/zm_ffmpeg.h:127,
                 from /build/zoneminder/src/zoneminder-git/src/zm_videostore.h:6,
                 from /build/zoneminder/src/zoneminder-git/src/zm_videostore.cpp:21:
/usr/include/libavcodec/packet.h:506:6: note: declared here
  506 | void av_init_packet(AVPacket *pkt);
      |      ^~~~~~~~~~~~~~
/build/zoneminder/src/zoneminder-git/src/zm_videostore.cpp: In member function ‘int VideoStore::writeAudioFramePacket(const std::shared_ptr<ZMPacket>&)’:
/build/zoneminder/src/zoneminder-git/src/zm_videostore.cpp:1273:21: warning: ‘void av_init_packet(AVPacket*)’ is deprecated [-Wdeprecated-declarations]
 1273 |       av_init_packet(&opkt);
      |       ~~~~~~~~~~~~~~^~~~~~~
In file included from /usr/include/libavcodec/avcodec.h:45,
                 from /build/zoneminder/src/zoneminder-git/src/zm_ffmpeg.h:127,
                 from /build/zoneminder/src/zoneminder-git/src/zm_videostore.h:6,
                 from /build/zoneminder/src/zoneminder-git/src/zm_videostore.cpp:21:
/usr/include/libavcodec/packet.h:506:6: note: declared here
  506 | void av_init_packet(AVPacket *pkt);
      |      ^~~~~~~~~~~~~~
/build/zoneminder/src/zoneminder-git/src/zm_videostore.cpp:1291:19: warning: ‘void av_init_packet(AVPacket*)’ is deprecated [-Wdeprecated-declarations]
 1291 |     av_init_packet(&opkt);
      |     ~~~~~~~~~~~~~~^~~~~~~
In file included from /usr/include/libavcodec/avcodec.h:45,
                 from /build/zoneminder/src/zoneminder-git/src/zm_ffmpeg.h:127,
                 from /build/zoneminder/src/zoneminder-git/src/zm_videostore.h:6,
                 from /build/zoneminder/src/zoneminder-git/src/zm_videostore.cpp:21:
/usr/include/libavcodec/packet.h:506:6: note: declared here
  506 | void av_init_packet(AVPacket *pkt);
[... 10 lines cut ...]
 1316 |     Debug(1, "undef dts, fixing by setting to stream cur_dts %" PRId64, stream->cur_dts);
      |     ^~~~~
/build/zoneminder/src/zoneminder-git/src/zm_videostore.cpp:1317:24: error: ‘AVStream’ {aka ‘struct AVStream’} has no member named ‘cur_dts’
 1317 |     pkt->dts = stream->cur_dts;
      |                        ^~~~~~~
[... cut ...]
 1319 |     Debug(1, "non increasing dts, fixing. our dts %" PRId64 " stream cur_dts %" PRId64, pkt->dts, stream->cur_dts);
      |     ^~~~~
/build/zoneminder/src/zoneminder-git/src/zm_videostore.cpp:1320:24: error: ‘AVStream’ {aka ‘struct AVStream’} has no member named ‘cur_dts’
 1320 |     pkt->dts = stream->cur_dts;
      |                        ^~~~~~~
Manifying 20 pod documents
make[2]: *** [src/CMakeFiles/zm.dir/build.make:888: src/CMakeFiles/zm.dir/zm_videostore.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
```

Nocifer, 2022-05-12:
Issue fixed with 1.36.15, good job.

closed 2022-05-12 by Nocifer