ZoneMinder/zoneminder #3461: ZMC fails to start for monitors with Decoding=none

klaasvakie, 2022-04-04:
**Describe Your Environment**
- Version of ZoneMinder [release version, development version, or commit]

1.37.13~20220401124110-focal

- How you installed ZoneMinder [e.g. PPA, RPMFusion, from-source, etc]

PPA from deb http://ppa.launchpad.net/iconnor/zoneminder-master/ubuntu focal main

- Full name and version of OS

DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=20.04
DISTRIB_CODENAME=focal
DISTRIB_DESCRIPTION="Ubuntu 20.04.4 LTS"

**Describe the bug**

Cameras that have the following:

Decoding=None
Video Writer=Camera Passthrough
Capturing=Always
Motion Detection=None

don't automatically start recording after rebooting. Function state in Console view is "Not Capturing, Recording" and icon remains red. No new events are generated.

**To Reproduce**

Configure a camera with above properties and reboot the machine. Log into ZM and observe state in the Console view. Observe event count not increasing over time.

**Expected behavior**

Cameras should start normally and generate Events after a reboot. 

**Fix**

Reverting commit: https://github.com/ZoneMinder/zoneminder/commit/f95d184f02c212b1cfd283e1ac0fa74881d080ae makes these cameras start normally for me.

**Workaround**

After booting, just clicking on the Source/Settings for the camera and clicking "Save" without changing anything causes the camera to start recording and generating events. Of course you will have to do this every time after rebooting and for each monitor

connortechnology, 2022-04-05:
I think the fix actually to do with ServerId and not being in a multi-server environment.
Fixed by 5b824f46c4b66c867959fdf905b7d4d571145d17. Please verify.

klaasvakie, 2022-04-05:
I can confirm that https://github.com/ZoneMinder/zoneminder/commit/5b824f46c4b66c867959fdf905b7d4d571145d17 fixes this for me.

Thank you very much @connortechnology !

closed 2022-04-05 by connortechnology