ZoneMinder/zoneminder #3463: ZMC crashes when save JPEG is enabled

xeroiv, 2022-04-08:
- Version: 1.36.14
- Installation: Installed from ZM Repo(https://wiki.zoneminder.com/Debian_11_Bullseye_with_Zoneminder_1.36.x)
- OS: Debian 11 LXC on Proxmox 7.1
- Storage: FreeNAS nfs share

**Describe the bug**
When the option for save JPEG is enabled ZMC constantly crashes causing many single frame events to be saved instead of the 10 minute segments that are configured as the minimum event length.

**To Reproduce**
Steps to reproduce the behavior:
1. Go Monitor 1 Edit Screen.
2. Set "save JPEG" to something other than disabled.
3. Save Configuration.
4. See error

**Expected behavior**
ZMC for the monitor that saves JPEG not crashing.

**Debug Logs**
```

[zmc_m1.log](https://github.com/ZoneMinder/zoneminder/files/8447521/zmc_m1.log)

```

baudneo, 2022-04-08:
Your logs show that you are passing through h264 stream to the videowriter ->

```
04/07/22 19:42:29.510443 zmc_m1[580].DB1-zm_event.cpp/182 [Using storage area at /shared/security/1/2022-04-07/4068]
04/07/22 19:42:29.510488 zmc_m1[580].DB1-zm_event.cpp/199 [Writing video file to /shared/security/1/2022-04-07/4068/4068-video.mp4]
04/07/22 19:42:29.510505 zmc_m1[580].DB4-zm_swscale.cpp/27 [SWScale object created]
04/07/22 19:42:29.510521 zmc_m1[580].DB1-zm_videostore.cpp/115 [Opening video storage stream /shared/security/1/2022-04-07/4068/4068-video.mp4 format: mp4]
```

xeroiv, 2022-04-10:
So after further reading some of the documents that setting would appear to be at odds with the passthrough command and should probably remain off.  There is an issue though that someone CAN turn both those settings on and crashes ZMC.

connortechnology, 2022-09-21:
So is this still an ongoing issue?  If you can't show actual crashing I'll have to close it.

xeroiv, 2022-09-21:
Hi, I have worked around the issue by using the correct setup options.  I
haven't done troubleshooting since I opened the issue.

RTechSn, 2023-03-22:
I have the same issue. 
After a couple of seconds of recording log shows
`INF	'zmc -m 1' crashed, signal 8	zmdc.pl	-`
and zoneminder is restarting.
It's happening only when i use NFS storage and "Save jpegs" is checked.

> Hi, I have worked around the issue by using the correct setup options. I haven't done troubleshooting since I opened the issue.
> […](#)

Could you please tell me, how you have solved it?