ZoneMinder/zoneminder #3464: Camera FPS and streaming FPS are not same.

amimbitassu, 2022-04-13:
- Version of ZoneMinder 1.36.14
- Installed from PPA
- Ubuntu 18.06
- Edge

- Hikvision
- 6 FPS
- 1920x1080
- RTSP - FFMPEG - H265+

**Describe the bug**

When watching stream from montage view from the browser or zmninja from an offsite location, due to network lag or bandwidth, the frames can not be delivered. Warnings start to appear about throttling FPS.

The issue is, cameras are set to 6fps, no FPS limit is set on the zoneminder, but the throttling messages start from 20fps then 13.33 and then 8.89. After that the stream is stopped. Why is the streaming set to an higher FPS than the FPS of the camera? If all the camera were set to 6fps streaming; then there may be no throttling? (There are 25 cameras in total.)

**To Reproduce**

**Expected behavior**

**Debug Logs**
```

<!--
"<html>
<body>
<!--StartFragment-->

2022-04-13 10:19:23 | zms_m14 |   | 406079 | WAR | Frame send time 305 msec too slow, throttling maxfps to 20.00 | zm_monitorstream.cpp | 458
-- | -- | -- | -- | -- | -- | -- | --

<!--EndFragment-->
</body>
</html>
-->

```
"

connortechnology, 2022-04-13:
Good question.  There actually is an fps limit set in ZM, and it is set at Options->High B/W->WEB_H_VIDEO_MAXFPS. There are similar options under the other bandwidth options as well.  This defaults to 30.  

One of the issues with the code is that once it reduces the framerate, it never goes back up. So a temporary network pause will cause the stream to stay slow. I fixed this in the current development version (1.37.*).  

The question is why does it stop? It shouldn't stop, it should just stay slow. Need more logs.

amimbitassu, 2022-04-14:
Thank you very much for this information. It greatly improved live view performance.

I was waiting to be at the office to create logs for the errors to share, I produced them yesterday night during live view with zmninja. (frame throttle from 6fps to 4fps happened once and then errors started to pop up if I remember correctly.) 

BUT, today when I try live view, I don't even get the throttling.

Not sure what happened, maybe the offsite internet quality is much better compared to yesterday. Satellite internet :(

Note: I can not figure out how to export a section of logs from previous days. I can not filter for date. Is there a guide on this? Is it possible to export whole 7 day of logs? I could only filter for err and find the time. You can ignore zmc_m5 errors, there is a problem with the cable of the camera I guess.

[logs.csv](https://github.com/ZoneMinder/zoneminder/files/8487053/logs.csv)

"<html>
<body>

2022-04-13 19:46:57 | web_php |   | 442100 | ERR | No data to read from socket | /usr/share/zoneminder/www/includes/functions.php | 1904
-- | -- | -- | -- | -- | -- | -- | --
2022-04-13 19:46:57 | web_php |   | 442100 | ERR | Timed out waiting for msg /run/zm/zms-243761s.sock | /usr/share/zoneminder/www/ajax/stream.php | 95
2022-04-13 19:46:57 | web_php |   | 442045 | ERR | No data to read from socket | /usr/share/zoneminder/www/includes/functions.php | 1904
2022-04-13 19:46:57 | web_php |   | 442045 | ERR | Timed out waiting for msg /run/zm/zms-009411s.sock | /usr/share/zoneminder/www/ajax/stream.php | 95
2022-04-13 19:46:57 | web_php |   | 442035 | ERR | No data to read from socket | /usr/share/zoneminder/www/includes/functions.php | 1904
2022-04-13 19:46:57 | web_php |   | 442035 | ERR | Timed out waiting for msg /run/zm/zms-200545s.sock | /usr/share/zoneminder/www/ajax/stream.php | 95
2022-04-13 19:46:57 | web_php |   | 442022 | ERR | No data to read from socket | /usr/share/zoneminder/www/includes/functions.php | 1904
2022-04-13 19:46:57 | web_php |   | 442022 | ERR | Timed out waiting for msg /run/zm/zms-462765s.sock | /usr/share/zoneminder/www/ajax/stream.php | 95
2022-04-13 19:46:57 | web_php |   | 441884 | ERR | Socket /run/zm/zms-720519s.sock does not exist. This file is created by zms, and since it does not exist, either zms did not run, or zms exited early. Please check your zms logs and ensure that CGI is enabled in apache and check that the PATH_ZMS is set correctly. Make sure that ZM is actually recording. If you are trying to view a live stream and the capture process (zmc) is not running then zms will exit. Please go to http://zoneminder.readthedocs.io/en/latest/faq.html#why-can-t-i-see-streamed-images-when-i-can-see-stills-in-the-zone-window-etc for more information. | /usr/share/zoneminder/www/includes/functions.php | 1904
2022-04-13 19:46:55 | zms_m9 |   | 442007 | ERR | Can't sendto on sd 7: No such file or directory | zm_monitorstream.cpp | 314

</body>
</html>"

Note2: I have identified a minor bug in camera addition page. When there is a "!" character in the rtsp camera address, an error pops up about invalid character, but the camera is saved and is working. Shall I create a new issue for this?

connortechnology, 2022-04-14:
No new issue.  ! is not valid.  It needs to be %encoded.  You can get away with ! but other characters will REALLY screw everything up.

amimbitassu, 2022-04-15:
I've already changed my camera passwords including "!".

I was able to recreate the throttling and errors.

[tableExport (2).csv](https://github.com/ZoneMinder/zoneminder/files/8493841/tableExport.2.csv)