ZoneMinder/zoneminder #3465: will not install on Debian buster because of repository expired certificate

AlexisPierre, 2022-04-15:
Attempting to install /debian/release-1.36/buster :

```
https://zmrepo.zoneminder.com/debian/release-1.36 buster/ Release
  Certificate verification failed: The certificate is NOT trusted. 
The certificate chain uses expired certificate.  Could not handshake: Error in the certificate verification. [IP: 158.69.226.113 443]

connortechnology, 2022-04-15:
Not getting that here. cert doesn't expire until june 1023.  Is your clock set right?

AlexisPierre, 2022-04-15:
Thanks, current time is sat 00:32 GMT+1.

AlexisPierre, 2022-04-15:
I've also tried copying archive-keyring.gpg and all files from zoneminder to use my localhost as repository, in oder to avoid the certificates issue, but apt does not find the InRelease file and then does not find zoneminder package. In other repositories Release is above the directory with the packge debs etc, but I don't know about the package formats.
Thanks!

gheja, 2022-04-15:
I believe the problem is that the certificate on `zmrepo.zoneminder.com` is signed by Let's Encrypt, and recently they are using an intermediate certificate to sign which is signed by two CA certificates, `DST Root CA X3` and `ISRG Root X1`. The DST cert has expired in 2021.

There was a bug in OpenSSL that handled this in the wrong way - it returned "expired" status and did not bother to check the other one.

More info here: https://serverfault.com/questions/1079199/client-on-debian-9-erroneously-reports-expired-certificate-for-letsencrypt-issue

If you can't or don't want to upgrade your system, OpenSSL or ca-certificates, then removing the expired root CA cert, installing the new and updating ca-certificates cache should fix it.

But of course, a disclaimer: you're taking advice from a random guy on the internet, these are system wide settings, and you should have a usable backup at hand, just in case.

Run these as root:
```bash
# remove the expired DST Root CA
rm /usr/share/ca-certificates/mozilla/DST_Root_CA_X3.crt

# comment it out in the ca-certificates.conf
sed -r -i 's,^(mozilla/DST_Root_CA_X3.crt),# \1,g' /etc/ca-certificates.conf

# update your CA certificate cache
update-ca-certificates --fresh
```

If the problem still persist, install the new certificate, run as root:
```bash
# create a directory for the new certificate
mkdir -p /usr/local/share/ca-certificates/extra

# download the new certificate
wget --no-check-certificate -O /usr/local/share/ca-certificates/extra/isrgrootx1.crt https://letsencrypt.org/certs/isrgrootx1.pem.txt

# update your CA certificate cache
update-ca-certificates --fresh
```

AlexisPierre, 2022-04-15:
Thank you. In the meantime I managed to install from my own localhost repository, bypassing the certificates issue. It took my some time as I had left out the buster.db file.

connortechnology, 2022-04-15:
That cert is not present and is deselected on our servers.  I suspect the issue is entirely on @AlexisPierre's machine. I'm no expert of course and wait to be proven wrong.

gheja, 2022-04-16:
The certificate follows specs only the OpenSSL library had a bug when validating it. It was fixed.

More about the problem:

> Unfortunately this does not apply to OpenSSL 1.0.2 which always prefers the untrusted chain and if that chain contains a path that leads to an expired trusted root certificate (DST Root CA X3), it will be selected for the certificate verification and the expiration will be reported.

https://www.openssl.org/blog/blog/2021/09/13/LetsEncryptRootCertExpire/

https://bugs.launchpad.net/ubuntu/+source/openssl/+bug/1928989

aval13, 2022-04-17:
Just my 2 cents.

@AlexisPierre 
on my Buster (10.12) server, apt update works without any issues.

About the **key signing the repository**:
As @connortechnology said, the key signing the repo is valid until 2023.
apt-key list shows:
pub   rsa2048 2018-01-12 [SC] [expires: 2023-01-11]
      6BF5 D08D 3846 AB63 DCE6  F0AE E79B FCE5 B226 8263
uid           [ unknown] Isaac Connor <isaac@zoneminder.com>
pub   rsa2048 2018-01-12 [SC] [expires: 2023-01-11]
      E605 ABF9 5A57 00ED 1EAF  108B 2BEB 7FDD 03AB 47CC
uid           [ unknown] Isaac Connor <isaac@zoneminder.com>

About the **certificates presented on the Debian repo URL**:
@gheja The expired certificate was valid until 2021-09-30, more than 6 months ago, this issue was opened 2 days ago, so if it was that particular issue, most likely it would have "hit" sooner.
Also Debian Buster uses openssl 1.1.1n, not the affected 1.0.2.

I checked the chain based on what my server has and:
openssl s_client on the repo's url returns ok on my server (chain ok: DST Root CA X3 -> ISRG Root X1 -> R3 -> server)
although DST Root CA X3 is indeed expired in the system.
However, in Debian we also get ISRG Root X1 (trusted root CA, not signed by X3) and the Authority Key reported for the R3 cert matches the installed trusted X1 CA.
Even if the intermediate served by the server presents the X1 signed by X3, the key of that cert is the same as the root CA X1 so the chain check stops before reaching the expired X3.
So that's why the system trusts the repo's URL.

If anybody has more questions, I'd be happy to help is needed.

closed 2022-05-18 by connortechnology