ZoneMinder/zoneminder #3474: Email issue ZM 1.36.XX

phuongpham, 2022-04-29:
Not sure if this is a bugs or a "feature"
But after upgrading to ZM1.36 I had to add email details in the Event filter section.
The problem I have is I'm using ZM togheter Event Server and the email is sending out to quickly.
It means ES is not finished to processing, before ZM already sent the email.
It's somewhere to change so the email is not sending after ES is finished.

I'm using ALPR, object recognition and Face Recognition in ES.

@connortechnology  I tried to registred new profile in the Zoneminder forum but never got the "activation email". Have already looked inside the spam inbox etc. but nothing there.

phuongpham, 2022-04-29:
> Add an EndTime is not null rule to the filter. This will make it wait until the event is over before sending the email.
> 
> Yes there is something weird going on with forum email.

Great! I will try this out now

phuongpham, 2022-04-30:
@connortechnology 
I added the additional filter as you explain.
The problem now is ZM doesn't send email any longer

[image]

connortechnology, 2022-04-30:
Sorry, my mistake, End StartDate/Time.

phuongpham, 2022-04-30:
> Sorry, my mistake, End StartDate/Time.

I changed to uppercase and seems like it work. But still end time not equal to NULL

connortechnology, 2022-05-02:
Please post another screen shot. Also... I guess I was a little scatter brained on the weekend... the field is End Date/Time.  So End Date/Time IS NOT NULL.  When you select IS NOT, the text field is supposed to only allow NULL but for some reason isn't. So I'll have to fix that.  Please type in uppercase NULL (it should be case insensitive, but just in case, make it upper case).

phuongpham, 2022-05-02:
> Please post another screen shot. Also... I guess I was a little scatter brained on the weekend... the field is End Date/Time. So End Date/Time IS NOT NULL. When you select IS NOT, the text field is [...]

This is what I have today

[image]

connortechnology, 2022-05-02:
You still have End Time instead of End Date/Time.  End Time is just the time component, like 8:43.  We need the entire field. Which is End Date/Time

closed 2022-10-11 by connortechnology