ZoneMinder/zoneminder #3475: monitors.json API matches uses incorrect Id to return Manufacturer and CameraModel

MarsRover75, 2022-04-30:
ZoneMinder 3.17.13 API 2.0
PPA install on Ubuntu Server 20.04 LTS
**BUGS:**
**1. monitors.json and monitor/<id>.json use incorrect Id (mathing Monitor.Id instead of Monitor.ManufacturerId) to return Manufacturer and CameraModel:**

```
/usr/bin/curl -sS --noproxy localhost http://localhost/zm/api/monitors/3.json | jq .
{
  "monitor": {
    "Monitor": {
      "Id": "3",
      "Name": "third",
      "Notes": "",
      "ServerId": "0",
      "StorageId": "3",
      "ManufacturerId": "12",
      "ModelId": "27",
      "Type": "Ffmpeg",
------------- cut -------------------
    },
    "Monitor_Status": {
      "MonitorId": "3",
      "Status": "Connected",
      "CaptureFPS": "25.00",
      "AnalysisFPS": "0.00",
      "CaptureBandwidth": "194361"
    },
    "Event_Summary": {
      "MonitorId": "3",
      "TotalEvents": "2097",
      "TotalEventDiskSpace": "238373534279",
      "HourEvents": "11",
      "HourEventDiskSpace": "0",
      "DayEvents": "169",
      "DayEventDiskSpace": "3466844284",
      "WeekEvents": "883",
      "WeekEventDiskSpace": "79973436275",
      "MonthEvents": "2097",
      "MonthEventDiskSpace": "238373534279",
      "ArchivedEvents": "0",
      "ArchivedEventDiskSpace": "0"
    },
    "Manufacturer": {
      "Id": "3",
      "Name": "Airlink101"
    },
    "CameraModel": {
      "Id": "3",
      "Name": "A24",
      "ManufacturerId": "1"
    }
  }
}

```

**2. /host/getDiskPercent.json shows unrelated info in space**
```
{
  "usage": {
    "pin": {
      "space": "0",
      "color": "#2615b6"
    },
    "forche": {
      "space": "3.814697265625E-6",
      "color": "#42b147"
    },
    "third": {
      "space": "3.814697265625E-6",
      "color": "#5eb348"
    "top_1": {
      "space": "3.814697265625E-6",
      "color": "#11fa51"
    },
    "top_2": {
      "space": "3.814697265625E-6",
      "color": "#dc9f57"
    },
    "counter": {
      "space": "3.814697265625E-6",
      "color": "#6e35ed"
    },
    "upper": {
      "space": "0",
      "color": "#44e284"
    },
    "vily": {
      "space": "0",
      "color": "#514082"
    },
    "Total": {
      "space": "102.70781326294",
      "color": "#F7464A"
    }
  }
}

```

connortechnology, 2022-05-01:
The getDiskPercent function is documented as deprecated..  I have to assume that the color field is used by zmNinja.  I agree it probably shouldn't be there, but I'm not going to remove it without being sure.

closed 2022-05-01 by connortechnology

MarsRover75, 2022-05-04:
> The getDiskPercent function is documented as deprecated.. I have to assume that the color field is used by zmNinja. I agree it probably shouldn't be there, but I'm not going to remove it without being sure.

OK, but what about incorrect Manufacturer and CameraModel matching?

connortechnology, 2022-05-06:
Fixed by https://github.com/ZoneMinder/zoneminder/commit/2a5a94b995580a442652dfa2cedb73f4277c9b88

commit 2a5a94b9 (2022-05-01): Fix monitors.json and monitor/.json use incorrect Id (mathing Monitor.Id instead of Monitor.ManufacturerId) to return Manufacturer and CameraModel: Fixes #3475