ZoneMinder/zoneminder #3478: SQLSTATE[HY000] [2002] on multiserver setup with MariaDB SSL enabled

arekscode, 2022-05-04:
I provide some more information on this forum post: https://forums.zoneminder.com/viewtopic.php?t=31717

**Describe Your Environment**
- Version of ZoneMinder [release version, development version, or commit]
Tried on 1.37.14~20220501132939-focal and 1.36.14-focal1
- How you installed ZoneMinder [e.g. PPA, RPMFusion, from-source, etc]
PPA
- Full name and version of OS
Ubuntu 20 LTS for ZoneMinder, Debian 11 for database (mariadb --version = 15.1 Distrib 10.5.15-MariaDB)

**Describe the bug**
ZoneMinder seems to be unable to connect to a database if SSL is enabled. I am simply presented with SQLSTATE[HY000] [2002]. There is nothing other than `ZoneMinder will retry connection in <n> seconds.` after that. 

I have no problem connecting using `mariadb -u zmuser -h <host> -p`.

**To Reproduce**

1. Enable SSL on MariaDB
2. Create a user on database with 'require ssl'
3. Add certs and keys to ZoneMinders 

**Expected behavior**
Just connect to the database. 

**Debug Logs**
Not sure what to include. ZoneMinder doesn't log anything about this from what I can tell. 

`systemctl status mysql` (Database server)
```
[Warning] Aborted connection <n> to db: 'unconnected' user: 'unauthenticated' host: '<ip of ZoneMinder server>' (This connection closed normally without authentication)
```

`/etc/zm/conf.d/03-certs.conf` (ZoneMinder server)
```
# SSL CA certificate for ZoneMinder database
ZM_DB_SSL_CA_CERT=/etc/mysql/ssl/ca-cert.pem

# SSL client key for ZoneMinder database
ZM_DB_SSL_CLIENT_KEY=/etc/mysql/ssl/client-key.pem

# SSL client cert for ZoneMinder database
ZM_DB_SSL_CLIENT_CERT=/etc/mysql/ssl/client-cert.pem
```

`/etc/mysql/mariadb.conf.d/50-mysql-clients.cnf` (Database server)
```
...
[mysql]
ssl-ca=/etc/mysql/ssl/ca-cert.pem
ssl-cert=/etc/mysql/ssl/client-cert.pem
ssl-key=/etc/mysql/ssl/client-key.pem
...
```

`mariadb -u root -e "SELECT User, Host FROM mysql.user;"`  (Database server)
```
User    Host    Password
...
zmuser  %
zmuser  localhost
zmuser  192.168.0.0/255.255.255.0
```

bkjaya2020, 2022-05-16:
I have faced the same problem with mysql 8 and zoneminder master 1.37.16~20220513115825-focal installed on ubuntu focal
[image]

arekscode, 2022-05-16:
> I have faced the same problem with mysql 8 and zoneminder master 1.37.16~20220513115825-focal installed on ubuntu focal
> [image]

I ended up switching to frigate and have not looked back. 
Everything is done through a web ui so you only need to set up a https reverse proxy or open up a port on your router. 
It integrates with home assistant very well, way way way better than zoneminder, and comes with object detection out of the box, which uses a lot less cpu than zoneminder (even without the recommended coral tpu).
It’s all configured in text files, so it can be a bit fiddly, but it’s definitely worth it.

closed 2022-05-16 by arekscode