ZoneMinder/zoneminder #3480: Can not set OPT_USE_API after upgrade to 1.37.15_20220504130434-focal

MarsRover75, 2022-05-06:
**Describe Your Environment**
- Version 1.37.15_20220504130434-focal_amd64.deb 
- deb from https://launchpad.net/~iconnor/+archive/ubuntu/zoneminder-master/+build/23609245
- Ubuntu Server 20.04 AMD64 LTS with latest updates

**If the issue concerns a camera**
It doesn't

**Describe the bug**
After upgrading ZM from 1.37.13 to 1.37.15_20220504130434-focal_amd64.deb OPT_USE_API turned itself off.
Any attempts to turn it back on silently end up with OPT_USE_API=0.
The only solution I've found was to downgrade to 1.37.14_20220502095655-focal for a minute just to turn it back on. Then it stays ON until I change anything in /zm/index.php?view=options&tab=system with saving. The parameter I change is saved successfully, but OPT_USE_API turns OFF and no way to change it to ON without doing a "downgrade to 1.37.14 -> turn on OPT_USE_API -> upgrade to 1.37.15" procedure.

The same happened on another identical server, but the upgrade was done from 1.34.26.

**To Reproduce**
See description above.

**Expected behavior**
OPT_USE_API=1

**Debug Logs**
```

<insert debug logs here, please make sure they are within the ``` quotes so they are formatted properly>

```

connortechnology, 2022-05-06:
This was fixed yesterday.   zoneminder - 1.37.15~20220506114546-focal  is the latest build.

MarsRover75, 2022-05-06:
Thanks, that was quick!

MarsRover75, 2022-05-07:
unfortunately,. problem persists in 1.37.15_20220506114546-focal, couldn't enable OPT_USE_IP in web-UI.
```
~# dpkg -i zoneminder_1.37.15_20220506114546-focal_amd64.deb
(OK)
~# /usr/bin/curl -sS --noproxy localhost http://localhost/zm/api/monitors/4.json
{"success":false,"data":{"name":"API Disabled","message":"API Disabled","url":"\/zm\/api\/monitors\/4.json","exception":{"class":"UnauthorizedException","code":401,"message":"API Disabled"}}}
~# mysql -e "select Name, Value from Config where Name='ZM_OPT_USE_API';"
+----------------+-------+
| Name           | Value |
+----------------+-------+
| ZM_OPT_USE_API | 0     |
+----------------+-------+
1 row in set (0,00 sec)

mysql> UPDATE zm.Config SET Value=1 WHERE Name='ZM_OPT_USE_API';
Query OK, 1 row affected (0,01 sec)
Rows matched: 1  Changed: 1  Warnings: 0

mysql> select Name, Value from Config where Name='ZM_OPT_USE_API';
+----------------+-------+
| Name           | Value |
+----------------+-------+
| ZM_OPT_USE_API | 1     |
+----------------+-------+
1 row in set (0,00 sec)

~# /usr/bin/curl -sS --noproxy localhost http://localhost/zm/api/monitors/4.json
{"monitor":{"Monitor":{"Id":"4","Name":"status","Notes":"","ServerId":"0","StorageId":"3", ...... cut ...
```
no need to restart ZM

closed 2022-06-17 by connortechnology