ZoneMinder/zoneminder #3484: Montage Review Video Download failed

Bjplms, 2022-05-14:
**Describe Your Environment**
- Version of ZoneMinder 1.36.12
- How you installed ZoneMinder [repository]
- Full name and version of OS Debian 11
- Browser name and version Microsoft Edge and Google Chrome

**Describe the bug**
When I try to download a video about Montage Review which is longer than 10 minutes nothing happens except the website shows all the time that it is being created, in /var/tmp I see that the zip or tar folder is created and the size also approx. fits with what is stated on the website, but unfortunately there is no download.
If I download it manually from /var/tmp, everything is there and OK.
It looks different if I only download 8 minutes, for example, this works immediately.

thank you for your help

connortechnology, 2022-06-01:
Please test with 1.36.17

connortechnology, 2022-09-21:
Closed due to potential fix and lack of response.

closed 2022-09-21 by connortechnology