ZoneMinder/zoneminder #3488: Add an option to disable timestamp and make timestamp as an opt-in option (not opt-out)

AIGeneratedUsername, 2022-05-21:
- [ ] Add an option to disable timestamp because there is no such option on the "Monitor config --> Timestamp" page.
- [ ] Make timestamp as an opt-in option (not opt-out). An ugly large default timestamp overlaps a bottom part of a video and prevents to view what it going on on the video. My proposal is to disable timestamp be default. Who wants it - s(he) will be able to enable it.

connortechnology, 2022-05-21:
Just clear the format line, timestamp goes away.

AIGeneratedUsername, 2022-05-21:
> Just clear the format line, timestamp goes away.

I have an empty "Timestamp Label Format" field. I even rebooted my system to make sure that changes are applied. Timestamp is still attached on videos in "Events".
[image]
[image]

connortechnology, 2022-05-21:
Ah... that's not actually in the video (if you downloaded the mp4 and played it in vlc or something it wouldn't be there).  It's just video.js overlaying it.  I agree it should be turn off-able.

commit 569c4337 (2023-01-04): Hide timestamp caption from bottom of video.js event view.  It serves no purpose.  Fixes #3488
commit 61b9dec3 (2023-01-04): Hide timestamp caption from bottom of video.js event view.  It serves no purpose.  Fixes #3488
closed 2023-01-04 by connortechnology