ZoneMinder/zoneminder #3492: v1.36.16  view=montage does not work correctly

arpitest, 2022-05-27:
**Describe Your Environment**
ZM v1.36.16 release, from ppa on Ubuntu focal 20.04.4 LTS
Firefox 100.0.2, 1920x1080

**Describe the bug**
Since upgrading from 1.36.15, view=montage does not work correctly:
-  clicking on a camera picture does not open the one-camera view as previous versions
- '3 wide' selected (i have 6 cameras), but only 2 displayed on second row, the last one is on 3. row, screenshot attached
  (bug is probably in MonitorStream.js, replacing that file from 1.36.15 fixes this, but then images are not updating)

[image]

arpitest, 2022-05-27:
small update: clearing all cookies in browser fixes it for one time, but after reload it's wrong again. cookie clear fixes it again, until next reload...

ManlyJack, 2022-05-29:
I have a similar issue, used to have 4 cameras 4 wide, now they show stacked, 1 wide and don't live-stream just a static image. also not getting live view on any other pages.
Drop down boxes also do nothing

connortechnology, 2022-05-29:
@ManlyJack sounds like a javascript problem.  Please check javascript console for errors.  

@arpitest do the other layout options work?

connortechnology, 2022-06-01:
Please test with 1.36.17

arpitest, 2022-06-01:
thanks, both problems seems to be fixed in .17

closed 2022-06-01 by connortechnology