ZoneMinder/zoneminder #3493: Live view/preview not loading

AdBol05, 2022-05-28:
**Describe Your Environment**
- Version of ZoneMinder 1.36.12
- Installed on TrueNAS as plugin
- TrueNAS CORE 12.0-U8
- Tried with Crome and Internet Explorer

**Describe the bug**
Succesfully installed zoneminder on TrueNAS, and connected my cameras to it. Everything was working fine.
After addind users and shutting the server down (so I could wire up the cameras) and starting zoneminder again the server still reciesves streams from camera, saves events an so on, but the live view and preview isn't working now. Only showing this icon:
[image]

So I am unable to set my detection zones because I only get this:
[image]

connortechnology, 2022-05-29:
Check that AUTH_HASH_LOGINS is also checked and that your timezone settings are correct.

AdBol05, 2022-05-29:
Thanks that worked out.

closed 2022-05-29 by AdBol05