ZoneMinder/zoneminder #3505: (FreeBSD)(CLANG) Sendfile7 support is broken

abishai, 2022-06-12:
I try to build ZM 1.36.19 and after recent changes in https://github.com/ZoneMinder/zoneminder/blob/1.36.19/src/zm_sendfile.h it doesn't compile

```
/usr/ports/multimedia/zoneminder/work/zoneminder-1.36.19/src/zm_sendfile.h:29:17: error: no matching function for call to 'sendfile'
  ssize_t err = sendfile(in_fd, out_fd, *offset, size, nullptr, &size, 0);
                ^~~~~~~~
/usr/include/sys/socket.h:705:5: note: candidate function not viable: no known conversion from 'size_t *' (aka 'unsigned long *') to 'off_t *' (aka 'long *') for 6th argument
int     sendfile(int, int, off_t, size_t, struct sf_hdtr *, off_t *, int);
```

I can revert offending argument to off_t back, but is it viable solution?

abishai, 2022-06-17:
```
--- src/zm_sendfile.h.orig	2022-06-12 09:18:29 UTC
+++ src/zm_sendfile.h
@@ -17,7 +17,7 @@ extern "C" {
 
 /* Function to send the contents of a file. Will use sendfile or fall back to reading/writing */
 
-ssize_t zm_sendfile(int out_fd, int in_fd, off_t *offset, size_t size) {
+ssize_t zm_sendfile(int out_fd, int in_fd, off_t *offset, off_t size) {
 #ifdef HAVE_SENDFILE4_SUPPORT
   ssize_t err = sendfile(out_fd, in_fd, offset, size);
   if (err < 0) {
```

^ My idea was to change argument, but if you want to keep it, casting works as well. I'll apply yours variant then. Thank you!

connortechnology, 2022-06-17:
The issue is that we were abusing the stack variable size.  Which really.... we shouldn't do.  So I think this looks better.  Does it compile and work there? If so I'll merge it.

abishai, 2022-06-18:
The code compiles, but something is broken in zms. Monitor live view is constantly refreshes.

The logs is filled with
```
2022-06-18 13:18:57 | web_php |   | 3124 | ERR | Unable to get semaphore. | includes/functions.php | 1907
-- | -- | -- | -- | -- | -- | -- | --
2022-06-18 13:18:57 | zms_m1 |   | 273810 | WAR | Unable to send stream frame: Broken pipe | zm_monitorstream.cpp | 412
2022-06-18 13:18:56 | web_php |   | 3124 | ERR | Unable to get semaphore. | includes/functions.php | 1907
```

I'm building with DHAVE_SENDFILE=0
Without it I'm receiving zmc crash by signal 11 on startup,  but I set this knob before. I was unable to solve this issue since I ported 1.30 to FreeBSD.  I remember ZM has a fallback if sendfile is not available. Maybe it's broken now? Or flickering is another regression? It looks like zms sends one frame and than reloads and start to send initial frame one more time and so on.

The last version that works is v1.36.12. Probably, I should try bisect or revert some obvious places to see if it helps.

abishai, 2022-06-18:
Probably, it's not zms, but zmc.

```
Found locked packet when trying to free up video packets. This basically means that decoding is not keeping up.
```
It looks like zmc provides broken video to all subsystems.

abishai, 2022-06-18:
Semaphore issue was simple enough - for semaphore support I need php74-sysvsem PHP extension. Stream is working now.
On the other side, I see a lot of 
```
"Found locked packet when trying to free up video packets. This basically means that decoding is not keeping up." warning messages.
```
and 
```
Analysis daemon for 8 Monitor-8 needs restarting, time since last analysis 66 seconds (1655563844-1655563778)
```

for every monitor in Record state (with or without analysis enabled)

"Locked packets" are filling logs very fast. Monitors are ffmpeg 5MP with 6 FPS 6000 VBR in passthrough mode. Both of them are in Record mode with analysis disabled. Other 10 monitors in Mocord throws no such messages.

connortechnology, 2022-06-20:
My experience with 5MP cams is that we cannot decode, and do motion detection on them fast enough.  

Yes you do need semaphore support.

Can we please leave this discussion to the issue at hand?  Being sendfile implementation.  What is the status of that?  Is it working with SENDFILE=1?

abishai, 2022-06-21:
I agree, should I create separate issues for the problems discovered? Issues with record only monitors is rather serious.

As for sendfile implementation:
1. With the patch, ZoneMinder compiles.
2. Without SENDFILE=0 zmc crashes on strartup with signal 11. I think, we need stack trace, right?

badrpc, 2022-12-02:
I think sendfile on FreeBSD can only work with sockets. When I built it with sendfile support, I wasn't able to watch recorded events with the following error in logs:

```
zms_e95.log:12/01/22 22:11:12.762041 zms_e95[101577].WAR-zm_eventstream.cpp/1091 [Unable to send raw frame 1: Socket operation on non-socket 356482 remaining]
zms_e95.log:12/01/22 22:11:12.762191 zms_e95[101577].ERR-zm_eventstream.cpp/730 [Can't send /var/db/zoneminder/events/3/2022-11-29/95/00001-capture.jpg: Socket operation on non-socket]
```

ERRORS section of the [manual](https://www.freebsd.org/cgi/man.cgi?query=sendfile&apropos=0&sektion=0&manpath=FreeBSD+13.1-RELEASE+and+Ports&arch=default&format=html) lists
``` 
     [ENOTSOCK]		The s argument is not a	socket.
```
Here `s` is the second argument of `sendfile`.

connortechnology, 2022-12-06:
@abishai what is the state of this ?  I there something I can merge to master?

VVD, 2023-01-09:
@abishai, I'm installed 1.36.12 today!
It have a lot of Deprecated warnings with php 8.1.
Begin to update port to 1.36.32, got same build error, create patch, want to create issue… and found this one. :-D

My patch is:
```
--- src/zm_sendfile.h.orig
+++ src/zm_sendfile.h
@@ -17,7 +17,7 @@ extern "C" {
 
 /* Function to send the contents of a file. Will use sendfile or fall back to reading/writing */
 
-ssize_t zm_sendfile(int out_fd, int in_fd, off_t *offset, size_t size) {
+ssize_t zm_sendfile(int out_fd, int in_fd, off_t *offset, ssize_t size) {
 #ifdef HAVE_SENDFILE4_SUPPORT
   ssize_t err = sendfile(out_fd, in_fd, offset, size);
   if (err < 0) {
```

Let's try together update port!

abishai, 2023-01-12:
Is 1.36.32 has deprecated warnings fixed?

Can you check that Record mode works as well?
I've experienced watchdog issue for every monitor in Record mode in 1.36.19. But, as I doubt that this one FreeBSD specific, maybe it's fixed in 1.36.32
```
Analysis daemon for 8 Monitor-8 needs restarting, time since last analysis 66 seconds (1655563844-1655563778)
```

connortechnology, 2023-01-12:
I would expect those to either be fixed, or the issue is outside ZM. 

php8.1 deprecation warnings are simply turned off in 1.36.32.

connortechnology, 2023-01-12:
So this patch compiles fine here.... I'd be happy to merge it...

hjf, 2023-01-12:
I compiled 1.36.32 with [this patch](https://github.com/ZoneMinder/zoneminder/issues/3505#issuecomment-1159198007) and I don't see any issues so far, save for icons not visible in Web UI. Recordings and montage work fine, and no errors in the log. I sent an email to @abishai  with the ports tree patch

badrpc, 2023-01-12:
@hjf can you watch recorded events?

hjf, 2023-01-12:
> @hjf can you watch recorded events?

yes when watching as MP4. it acts weird as MJPEG, but i'm not sure if it's my setup.

abishai, 2023-01-12:
@connortechnology May I ask you another offtopic question here ? Some of locales in lang looks suspicious. For example, en_gb.php contains LC_TIME=en_GB.utf8 . I think this is incorrect, it should be UTF-8.

abishai, 2023-01-12:
nm, it's fixed already.

abishai, 2023-01-12:
@hjf Received the patch, will try it. Thanks!

VVD, 2023-01-14:
> Is 1.36.32 has deprecated warnings fixed?

I don't see them anymore.
 
> Can you check that Record mode works as well? I've experienced watchdog issue for every monitor in Record mode in 1.36.19. But, as I doubt that this one FreeBSD specific, maybe it's fixed in 1.36.32

I'm use Modect with Camera Passthrough.

After I install 1.36.12 and configure 3 cams I got #3414, then added ram 16GB+swap 16GB, but OOM was almost after every alarm. Then it disappeared and I don't know what helped. It work fine for 3 days with a lot of alarms even with 4GB of ram without use of swap.
Now I update to 1.36.32 and got same on one cam - every alarm = OOM.
Set Maximum Image Buffer Size (frames) to 200 and OOM replaced with messages in log:
```
Jan 14 18:14:45 cctv zmwatch[78559]: WAR [Analysis daemon for 2 Cam-2 needs restarting, time since last analysis 48 seconds (1673709285-1673709237)]
Jan 14 18:14:46 cctv zmc_m2[96548]: WAR [zmc_m2] [Unable to get starting packet lock]
Jan 14 18:14:46 cctv zmc_m2[96548]: ERR [zmc_m2] [ALARM but no event]
Jan 14 18:14:57 cctv zmdc[78510]: WAR ['zmc -m 2' has not stopped at 23/01/14 18:14:57 after 10 seconds. Sending KILL to pid 96548]
Jan 14 18:18:14 cctv zmc_m2[98348]: WAR [zmc_m2] [You have set the max video packets in the queue to 200. The queue is full. Either Analysis is not keeping up or your camera's keyframe interval 21 is larger than this setting.]
Jan 14 18:18:16 cctv zmc_m2[98348]: WAR [zmc_m2] [Found locked packet when trying to free up video packets. This basically means that decoding is not keeping up.]
```
Analysis FPS 2, FPS from cam 10, video stream from cams 2048x1536 h264.

Top look like this most time:
```
Mem: 1951M Active, 260K Inact, 1277M Laundry, 693M Wired, 392M Buf, 20M Free
Swap: 16G Total, 841M Used, 15G Free, 5% Inuse, 4096B In, 3840K Out

  PID USERNAME    THR PRI NICE   SIZE    RES STATE    C   TIME    WCPU COMMAND
   11 root          3 155 ki31     0B    48K RUN      0  58,2H 244,15% [idle]
96282 www           4  20    0   687M   358M select   0   7:25  20,12% /usr/local/bin/zmc -m 3
96266 www           4  20    0   655M   362M select   0   7:26  19,05% /usr/local/bin/zmc -m 1
98348 www           4  20    0  3389M  2455M select   1   1:48  15,86% /usr/local/bin/zmc -m 2
```

abishai, 2023-01-14:
Analysys FPS (and other FPS settings) are very dangerous.  I think they should be unset...

abishai, 2023-01-14:
Is ajax request works for you in monitor view ? (Request that loads monitor alarm state and fps). It throws HTTP=500, something is wrong with semaphores again, however I have semaphore php extensions.

hjf, 2023-01-14:
I have had OOM but I think they are related to my server not being able to compress h.264 fast enough. but I tried to revert back to JPEG and the MJPEG recordings don't play. there is a segfault in logs.

abishai, 2023-01-14:
Well, looks like 1.36.12 is the last one that works under FreeBSD. 
1.36.32 has the same issue that stopped updating to 1.36.19

Any attempt to write stream produces zmc crash. While monitor in detect mode and detection not triggered everything looks OK.

```

1/14/23, 6:50:22 PM GMT+3 | zmdc |   | 12729 | ERR | 'zmc -m 6' exited abnormally, exit status 11 | zmdc.pl | -
-- | -- | -- | -- | -- | -- | -- | --
1/14/23, 6:50:22 PM GMT+3 | zmc_m6 |   | 583650 | ERR | Signal address is 0x125, from 0x27f012 | zm_signal.cpp | 80
1/14/23, 6:50:22 PM GMT+3 | zmc_m6 |   | 583650 | ERR | Got signal 11 (Segmentation fault), crashing | zm_signal.cpp | 50
1/14/23, 6:49:54 PM GMT+3 | zmc_m6 |   | 569575 | WAR | Found locked packet when trying to free up video packets. This basically means that decoding is not keeping up. | zm_packetqueue.cpp | 156
1/14/23, 6:49:54 PM GMT+3 | zmc_m6 |   | 569575 | WAR | You  have set the max video packets in the queue to 120. The queue is full.  Either Analysis is not keeping up or your camera's keyframe interval 0  is larger than this setting. | zm_packetqueue.cpp | 139
1/14/23, 6:48:57 PM GMT+3 | zmdc |   | 12729 | WAR | 'zmc -m 6' has not stopped at 23/01/14 18:48:57 after 10 seconds. Sending KILL to pid 31612 | zmdc.pl | -
1/14/23, 6:48:46 PM GMT+3 | zmc_m6 |   | 582186 | WAR | Unable to get starting packet lock | zm_monitor.cpp | 2692
1/14/23, 6:48:46 PM GMT+3 | zmwatch |   | 12864 | WAR | Analysis daemon for 6 Monitor-6 needs restarting, time since last analysis 47 seconds (1673711326-1673711279) | zmwatch.pl | -
1/14/23, 6:44:07 PM GMT+3 | zmc_m6 |   | 101695 | WAR | Found locked packet when trying to free up video packets. This basically means that decoding is not keeping up. | zm_packetqueue.cpp | 156
1/14/23, 6:44:07 PM GMT+3 | zmc_m6 |   | 101695 | WAR | You  have set the max video packets in the queue to 120. The queue is full.  Either Analysis is not keeping up or your camera's keyframe interval 0  is larger than this setting. | zm_packetqueue.cpp | 139
1/14/23, 6:43:22 PM GMT+3 | zmdc |   | 12729 | ERR | 'zmc -m 6' exited abnormally, exit status 11 | zmdc.pl | -
1/14/23, 6:43:22 PM GMT+3 | zmc_m6 |   | 581980 | ERR | Signal address is 0xbd, from 0x27f012 | zm_signal.cpp | 80
1/14/23, 6:43:22 PM GMT+3 | zmc_m6 |   | 581980 | ERR | Got signal 11 (Segmentation fault), crashing | zm_signal.cpp | 50
1/14/23, 6:43:21 PM GMT+3 | zmc_m6 |   | 276253 | WAR | Found locked packet when trying to free up video packets. This basically means that decoding is not keeping up. | zm_packetqueue.cpp | 156
1/14/23, 6:43:21 PM GMT+3 | zmc_m6 |   | 276253 | WAR | You  have set the max video packets in the queue to 120. The queue is full.  Either Analysis is not keeping up or your camera's keyframe interval 0  is larger than this setting. | zm_packetqueue.cpp | 139
1/14/23, 6:42:41 PM GMT+3 | zmdc |   | 12729 | ERR | 'zmc -m 6' exited abnormally, exit status 11 | zmdc.pl | -
1/14/23, 6:42:41 PM GMT+3 | zmc_m6 |   | 581790 | ERR | Signal address is 0xdd, from 0x27f012 | zm_signal.cpp | 80
1/14/23, 6:42:41 PM GMT+3 | zmc_m6 |   | 581790 | ERR | Got signal 11 (Segmentation fault), crashing | zm_signal.cpp | 50

```

VVD, 2023-01-14:
> Analysys FPS (and other ZM FPS settings) are very dangerous. I think they should be unset...

Recommendations got this #3414.

> Is ajax request works for you in monitor view ? (Request that loads monitor alarm state and fps). It throws HTTP=500, something is wrong with semaphores again, however I have semaphore php extensions.

Montage preview? Or what?
```
php81-ctype-8.1.14             The ctype shared extension for php
php81-gd-8.1.14                The gd shared extension for php
php81-intl-8.1.14              The intl shared extension for php
php81-opcache-8.1.14           The opcache shared extension for php
php81-pdo-8.1.14               The pdo shared extension for php
php81-pdo_mysql-8.1.14         The pdo_mysql shared extension for php
php81-pecl-APCu-5.1.22         APC User Caching
php81-session-8.1.14           The session shared extension for php
php81-sockets-8.1.14           The sockets shared extension for php
```

> signal 11 is always comes after 'locked packed' warning. Probably, memory corruption after zmc memory exhausted.
Where do you get this log? What file name?

abishai, 2023-01-14:
> Montage preview? Or what?
?view=watch

VVD, 2023-01-14:
> > Montage preview? Or what?
> 
> When you click on monitor name ?view=watch

I see video, but apache spam php errors in `error_log`:
```
[Sat Jan 14 19:19:27.173855 2023] [php:warn] [pid 99226] [client MYIP:10718] PHP Warning:  fopen(/usr/local/www/zoneminder/includes/functions.php.sem.590549): Failed to open stream: Permission denied in /usr/local/www/zoneminder/includes/functions.php on line 2239, referer: http://SERVER/zm/index.php?view=watch&mid=1
[Sat Jan 14 19:19:27.173914 2023] [php:error] [pid 99226] [client MYIP:10718] PHP Fatal error:  Uncaught TypeError: flock(): Argument #1 ($stream) must be of type resource, bool given in /usr/local/www/zoneminder/includes/functions.php:2242\nStack trace:\n#0 /usr/local/www/zoneminder/includes/functions.php(2242): flock(false, 2)\n#1 /usr/local/www/zoneminder/ajax/stream.php(28): sem_acquire(false, 1)\n#2 /usr/local/www/zoneminder/index.php(269): require_once('/usr/local/www/...')\n#3 {main}\n  thrown in /usr/local/www/zoneminder/includes/functions.php on line 2242, referer: http://SERVER/zm/index.php?view=watch&mid=1
```
Ofc process with owner `www` can't open /usr/local/www/zoneminder/includes/functions.php.sem.590549:
```
ls -l /usr/local/www/zoneminder/includes/
total 352
-rw-r--r--  1 root  wheel   7367 Nov 18 22:21 Control.php
-rw-r--r--  1 root  wheel  23610 Nov 18 22:21 Event.php
-rw-r--r--  1 root  wheel  21270 Nov 18 22:21 Filter.php
-rw-r--r--  1 root  wheel  14208 Nov 18 22:21 FilterTerm.php
-rw-r--r--  1 root  wheel   1162 Nov 18 22:21 Frame.php
-rw-r--r--  1 root  wheel   7225 Nov 18 22:21 Group.php
-rw-r--r--  1 root  wheel    565 Nov 18 22:21 Group_Monitor.php
-rw-r--r--  1 root  wheel  23994 Nov 18 22:21 Monitor.php
-rw-r--r--  1 root  wheel    564 Nov 18 22:21 MontageLayout.php
-rw-r--r--  1 root  wheel  16434 Nov 18 22:21 Object.php
-rw-r--r--  1 root  wheel   4234 Nov 18 22:21 Server.php
-rw-r--r--  1 root  wheel   1954 Nov 18 22:21 Snapshot.php
-rw-r--r--  1 root  wheel   5129 Nov 18 22:21 Storage.php
-rw-r--r--  1 root  wheel   1464 Nov 18 22:21 User.php
-rw-r--r--  1 root  wheel   1686 Nov 18 22:21 Zone.php
drwxr-xr-x  2 root  wheel   1024 Jan 14 16:58 actions
-rw-r--r--  1 root  wheel  17024 Nov 18 22:21 auth.php
-rw-r--r--  1 root  wheel   9610 Jan 14 16:49 config.php
-rw-r--r--  1 root  wheel  27654 Nov 18 22:21 control_functions.php
drwxr-xr-x  2 root  wheel    512 Jan 14 16:58 csrf
-rw-r--r--  1 root  wheel  11257 Nov 18 22:21 database.php
-rw-r--r--  1 root  wheel  77700 Nov 18 22:21 functions.php
-rw-r--r--  1 root  wheel   2212 Nov 18 22:21 lang.php
-rw-r--r--  1 root  wheel  16647 Nov 18 22:21 logger.php
-rw-r--r--  1 root  wheel   6211 Nov 18 22:21 session.php
```

abishai, 2023-01-14:
Probably some temp directory or config is not set. On another way, php issue are not so serious. But zmc crash is. All I know version 1.36.12 works, 1.36.32 (and as I know for sure - 1.36.19) has zmc issue.

VVD, 2023-01-14:
/usr/local/www/zoneminder/includes/functions.php:
```
// The following works around php not being built with semaphore functions.
if ( !function_exists('sem_get') ) {
  function sem_get($key) {
    return fopen(__FILE__ . '.sem.' . $key, 'w+');
  }
  function sem_acquire($sem_id) {
    return flock($sem_id, LOCK_EX);
  }
  function sem_release($sem_id) {
    return flock($sem_id, LOCK_UN);
  }     
}
```

abishai, 2023-01-14:
That __FILE__ is obviously wrong. It must point to temporary directory.

abishai, 2023-01-14:
```
diff -u /usr/local/poudriere/ports/workshop/multimedia/zoneminder/Makefile /usr/local/poudriere/ports/server/multimedia/zoneminder/Makefile
--- /usr/local/poudriere/ports/workshop/multimedia/zoneminder/Makefile	2023-01-14 16:20:08.164847000 +0000
+++ /usr/local/poudriere/ports/server/multimedia/zoneminder/Makefile	2023-01-14 10:03:23.578477000 +0000
@@ -1,6 +1,5 @@
 PORTNAME=	zoneminder
-DISTVERSION=	1.36.12
-PORTREVISION=	1
+DISTVERSION=	1.36.32
 CATEGORIES=	multimedia
 
 MAINTAINER=	bsd@abinet.ru
@@ -39,9 +38,9 @@
 USE_GITHUB=	yes
 GH_TUPLE=	FriendsOfCake:crud:1429237:crud/web/api/app/Plugin/Crud \
 		ZoneMinder:CakePHP-Enum-Behavior:ea90c0c:crud_plugin/web/api/app/Plugin/CakePHP-Enum-Behavior \
-		ZoneMinder:RtspServer:cd7fd49:rtsp_server/dep/RtspServer
+		ZoneMinder:RtspServer:eab3285:rtsp_server/dep/RtspServer
 USE_RC_SUBR=	zoneminder
-USE_PHP=	json pdo_mysql session gd sockets ctype opcache openssl hash
+USE_PHP=	json pdo_mysql session gd sockets ctype opcache openssl hash sysvsem
 
 OPTIONS_DEFINE=	NLS V4L
 OPTIONS_SUB=	yes
@@ -110,6 +109,7 @@
 	${MKDIR} ${STAGEDIR}/var/db/zoneminder
 	${MKDIR} ${STAGEDIR}/var/run/zm
 	${MKDIR} ${STAGEDIR}/var/tmp/zm
+	${MKDIR} ${STAGEDIR}/var/log/zm
 
 post-install: # zmsystemctl.pl is only for systend-based systems
 	@cd ${STAGEDIR}${PREFIX} && \
diff -u /usr/local/poudriere/ports/workshop/multimedia/zoneminder/distinfo /usr/local/poudriere/ports/server/multimedia/zoneminder/distinfo
--- /usr/local/poudriere/ports/workshop/multimedia/zoneminder/distinfo	2023-01-14 16:20:08.164963000 +0000
+++ /usr/local/poudriere/ports/server/multimedia/zoneminder/distinfo	2023-01-14 09:46:03.780182000 +0000
@@ -1,9 +1,9 @@
-TIMESTAMP = 1654708449
-SHA256 (zoneminder-zoneminder-1.36.12_GH0.tar.gz) = 8f2772c9aa891e3e3de6b73e4ce95189af4d5faa886ac13d7ce20972d11c31dd
-SIZE (zoneminder-zoneminder-1.36.12_GH0.tar.gz) = 11350267
+TIMESTAMP = 1673689563
+SHA256 (zoneminder-zoneminder-1.36.32_GH0.tar.gz) = 1bf0306a42f114734c9f8d8c52caed1f457a4990a690d7a50df531e5247ba20d
+SIZE (zoneminder-zoneminder-1.36.32_GH0.tar.gz) = 11497953
 SHA256 (FriendsOfCake-crud-1429237_GH0.tar.gz) = 4dc57f344623d3c0e735b53ed583e5382aa9e43012215f641cbd8ebad381aadc
 SIZE (FriendsOfCake-crud-1429237_GH0.tar.gz) = 73018
 SHA256 (ZoneMinder-CakePHP-Enum-Behavior-ea90c0c_GH0.tar.gz) = 3a9006c7c59f2b864ce628df4725b241fc23cfd9abc56e746acbd22d850f5ab1
 SIZE (ZoneMinder-CakePHP-Enum-Behavior-ea90c0c_GH0.tar.gz) = 3500
-SHA256 (ZoneMinder-RtspServer-cd7fd49_GH0.tar.gz) = 0a397ca5810ecff1c66955348b65903114c5da8f390e3af137a4d0fa33672762
-SIZE (ZoneMinder-RtspServer-cd7fd49_GH0.tar.gz) = 1395694
+SHA256 (ZoneMinder-RtspServer-eab3285_GH0.tar.gz) = 8e37a14b4895e2e1d76e323875ae515f4ac4018b50d579842848718de804c142
+SIZE (ZoneMinder-RtspServer-eab3285_GH0.tar.gz) = 1396598
Common subdirectories: /usr/local/poudriere/ports/workshop/multimedia/zoneminder/files and /usr/local/poudriere/ports/server/multimedia/zoneminder/files
diff -u /usr/local/poudriere/ports/workshop/multimedia/zoneminder/pkg-plist /usr/local/poudriere/ports/server/multimedia/zoneminder/pkg-plist
--- /usr/local/poudriere/ports/workshop/multimedia/zoneminder/pkg-plist	2023-01-14 16:20:08.171348000 +0000
+++ /usr/local/poudriere/ports/server/multimedia/zoneminder/pkg-plist	2023-01-14 10:03:40.370280000 +0000
@@ -6175,6 +6175,7 @@
 %%PERL5_MAN3%%/ZoneMinder::Trigger::Connection::Example.3.gz
 %%PERL5_MAN3%%/ZoneMinder::User.3.gz
 %%PERL5_MAN3%%/ZoneMinder::Zone.3.gz
+share/applications/zoneminder.desktop
 share/man/man8/zmaudit.pl.8.gz
 share/man/man8/zmc.8.gz
 share/man/man8/zmcamtool.pl.8.gz
@@ -6194,7 +6195,6 @@
 share/man/man8/zmvideo.pl.8.gz
 share/man/man8/zmwatch.pl.8.gz
 share/man/man8/zmx10.pl.8.gz
-share/applications/zoneminder.desktop
 %%DATADIR%%/db/triggers.sql
 %%DATADIR%%/db/zm_create.sql
 %%DATADIR%%/db/zm_update-0.0.1.sql
[... 84 lines cut ...]
+%%WWWDIR%%/skins/classic/js/jquery-ui-1.13.2/images/ui-icons_777620_256x240.png
+%%WWWDIR%%/skins/classic/js/jquery-ui-1.13.2/images/ui-icons_777777_256x240.png
+%%WWWDIR%%/skins/classic/js/jquery-ui-1.13.2/images/ui-icons_cc0000_256x240.png
+%%WWWDIR%%/skins/classic/js/jquery-ui-1.13.2/images/ui-icons_ffffff_256x240.png
+%%WWWDIR%%/skins/classic/js/jquery-ui-1.13.2/index.html
+%%WWWDIR%%/skins/classic/js/jquery-ui-1.13.2/jquery-ui.css
+%%WWWDIR%%/skins/classic/js/jquery-ui-1.13.2/jquery-ui.js
+%%WWWDIR%%/skins/classic/js/jquery-ui-1.13.2/jquery-ui.min.css
+%%WWWDIR%%/skins/classic/js/jquery-ui-1.13.2/jquery-ui.min.js
+%%WWWDIR%%/skins/classic/js/jquery-ui-1.13.2/jquery-ui.structure.css
+%%WWWDIR%%/skins/classic/js/jquery-ui-1.13.2/jquery-ui.structure.min.css
+%%WWWDIR%%/skins/classic/js/jquery-ui-1.13.2/jquery-ui.theme.css
+%%WWWDIR%%/skins/classic/js/jquery-ui-1.13.2/jquery-ui.theme.min.css
+%%WWWDIR%%/skins/classic/js/jquery-ui-1.13.2/package.json
 %%WWWDIR%%/skins/classic/js/jquery.min.js
 %%WWWDIR%%/skins/classic/js/jquery.min.js.URL
 %%WWWDIR%%/skins/classic/js/moment.js
@@ -8154,3 +8154,4 @@
 @dir(%%WWWOWN%%,%%WWWGRP%%,) /var/db/zoneminder
 @dir(%%WWWOWN%%,%%WWWGRP%%,) /var/run/zm
 @dir(%%WWWOWN%%,%%WWWGRP%%,) /var/tmp/zm
+@dir(%%WWWOWN%%,%%WWWGRP%%,) /var/log/zm
```

Here is patch against current FreeBSD port tree.

VVD, 2023-01-14:
ZM_DIR_TEMP is the good replace:
```
--- includes/functions.php.orig
+++ includes/functions.php
@@ -2236,7 +2236,7 @@
 // The following works around php not being built with semaphore functions.
 if ( !function_exists('sem_get') ) {
   function sem_get($key) {
-    return fopen(__FILE__ . '.sem.' . $key, 'w+');
+    return fopen(ZM_DIR_TEMP . '/sem.' . $key, 'w+');
   }
   function sem_acquire($sem_id) {
     return flock($sem_id, LOCK_EX);
```

> Patch against current port tree https://pastebin.com/bzGxWYNB

1. /var/cache/zoneminder/events => /var/db/zoneminder/events
Check this too: https://bugs.freebsd.org/bugzilla/show_bug.cgi?id=267304#c3
2. Require USE_PHP= intl.

VVD, 2023-01-14:
Installed devel/php81-sysvsem, restarted apache, all work fine now.

VVD, 2023-01-14:
My patch is: https://pastebin.com/bRCug87r

badrpc, 2023-01-14:
Regarding sendfile: As I have said in https://github.com/ZoneMinder/zoneminder/issues/3505#issuecomment-1335877465 the way zoneminder uses sendfile is not supported on FreeBSD. zms uses zm_sendfile to send data back to fcgiwrap and the file descriptor there is not a socket. I used a hacky[1] approach to disable sendfile and got a working (except the OOM problem) 1.36.32 package. I'll try to find time to disable it properly.

Regarding sem_get, the php80-sysvsem (or php81-/php82- to match the version of php) package has solved the problem for me. The port must be modified to list it as a dependency, but in the meantime I just installed it manually.

The 1.36.32 runs out of memory for me too. Easily kills 32G machine on first alarm. It looks like the one reported in https://github.com/ZoneMinder/zoneminder/issues/3614, I'm not sure how to trace this one down.

1. Hacky patch to disable sendfile
```
--- work/zoneminder-1.36.32/src/zm_sendfile.h.orig	2022-11-18 19:21:30.000000000 +0000
+++ work/zoneminder-1.36.32/src/zm_sendfile.h	2022-12-01 22:39:22.542680000 +0000
@@ -5,32 +5,33 @@
 extern "C" {
 #endif

-#ifdef HAVE_SENDFILE4_SUPPORT
-#include <sys/sendfile.h>
-#elif HAVE_SENDFILE7_SUPPORT
+//#ifdef HAVE_SENDFILE4_SUPPORT
+//#include <sys/sendfile.h>
+//#elif HAVE_SENDFILE7_SUPPORT
 #include <sys/types.h>
 #include <sys/socket.h>
-#include <sys/uio.h>
-#else
+//#include <sys/uio.h>
+//#else
 #include <unistd.h>
-#endif
+//#endif

 /* Function to send the contents of a file. Will use sendfile or fall back to reading/writing */

 ssize_t zm_sendfile(int out_fd, int in_fd, off_t *offset, size_t size) {
-#ifdef HAVE_SENDFILE4_SUPPORT
-  ssize_t err = sendfile(out_fd, in_fd, offset, size);
-  if (err < 0) {
-    return -errno;
-  }
-  return err;
-
-#elif HAVE_SENDFILE7_SUPPORT
-  ssize_t err = sendfile(in_fd, out_fd, *offset, size, nullptr, &size, 0);
-  if (err && errno != EAGAIN)
-    return -errno;
-  return size;
-#else
+//#ifdef HAVE_SENDFILE4_SUPPORT
+//  ssize_t err = sendfile(out_fd, in_fd, offset, size);
+//  if (err < 0) {
+//    return -errno;
+//  }
+//  return err;
+//
+//#elif HAVE_SENDFILE7_SUPPORT
+//  off_t ssize;
+//  ssize_t err = sendfile(in_fd, out_fd, *offset, size, nullptr, &ssize, 0);
+//  if (err && errno != EAGAIN)
+//    return -errno;
+//  return ssize;
+//#else
   uint8_t buffer[size];
   ssize_t err = read(in_fd, buffer, size);
   if (err < 0) {
@@ -38,13 +39,13 @@
     return -errno;
   }

-  err = fwrite(out_fd, buffer, size);
+  err = write(out_fd, buffer, size);
   if (err < 0) {
     Error("Unable to write %zu bytes: %s", size, strerror(errno));
     return -errno;
   }
   return err;
-#endif
+//#endif
 }

 #ifdef __cplusplus
```

abishai, 2023-01-14:
@badrpc sendfile is already disabled in port config via cmake option, no need to patch is furhter if it compiles.

badrpc, 2023-01-14:
Sorry, that's probably a recent update and I didn't notice it. Need to update my tree I suppose. Sorry again.

abishai, 2023-01-14:
No, it was from the beginning. Under FreeBSD ZM never used sendfile. Any attempt to turn it on would crash zms.
It's just stopped compile at all, so we were fixing compilation issue, not sendfile itself.

badrpc, 2023-01-14:
Hmm.... That (disabled sendfile) doesn't match my experience. I can indeed see that ports Makefile sets `-DHAVE_SENDFILE=0` but I'm pretty sure that it was compiling with `HAVE_SENDFILE7_SUPPORT` and then zms process was reporting `zms_e95.log:12/01/22 22:11:12.762191 zms_e95[101577].ERR-zm_eventstream.cpp/730 [Can't send /var/db/zoneminder/events/3/2022-11-29/95/00001-capture.jpg: Socket operation on non-socket]`. I will check and report back.

Regarding the memory usage of zmc, yes I agree. But it might be worth opening another issue for that as this one starts to have multiple parallel threads and becomes harder to follow.

connortechnology, 2023-01-14:
So I'd like to get to the bottom of all this. I used to have a freebsd vm,m but it's long gone.  I'm downloading 13.1.  Can you guys do me a favour and document the steps needed to go from a fresh install to getting zm installed?   Doesn't have to be perfect, and we can stick it all in the official docs so that FreeBSD can become a more supported platform.

abishai, 2023-01-14:
Simply installed or you need to play with building as well. I'll try to describe the process, but from my experience it scares Linux folks =/

Port has step-by-step installation procedure documented step-by-step
https://github.com/freebsd/freebsd-ports/blob/main/multimedia/zoneminder/files/pkg-message.in

The only trick is to switch fresh FreeBSD installation to latest packaged editing /etc/pkg/FreeBSD.conf

```
FreeBSD: {
  url: "pkg+http://pkg.FreeBSD.org/${ABI}/latest",
  mirror_type: "srv",
  signature_type: "fingerprints",
  fingerprints: "/usr/share/keys/pkg",
  enabled: yes
}
```

and install necessary software with

```
pkg install mysql57-server fcgiwrap nginx
```

You can setup ZM according documentation after that. (The latest version in port 1.36.12).

After you've got everything running, stop ZM and delete it

```
pkg delete zoneminder
```

Install port collection with 
```
portsnap auto
```

ZM build recipe will be in /usr/ports/multimedia/zoneminder

Try to build it with 

```
make 
``

make will pull necessary build dependencies and compile everything. (If it try to compile something big like llvm, you can install it from packages).

```
make clean
```
will clean all built material 

After that, apply my patch https://pastebin.com/bzGxWYNB on top, this will bump port to 1.36.32.
Try to make clean it to verify everything still builds.

Useful make commands

```
make clean - cleans workdirs
make - compiles
make install - build + install
make deinstall - removes installed
make extract - extract sources to workdir
make patch - extract sources and apply patches (specific to FreeBSD - from files/)
```

abishai, 2023-01-14:
(I can prepare everything if you can provide ssh access)

VVD, 2023-01-14:
> fcgiwrap nginx

I can try to add description how to configure with apache.
But better discuss this in IRC or in https://bugs.freebsd.org/bugzilla/.

abishai, 2023-01-15:
I've come to idea of serial tests using exactly the same dependencies I have in production (it's very stable on 1.36.12). I copy everything and begin step by step version bump of ZM alone in test environment. 

This, we'll find when bug occurs (or not if bug in dependency or how ZM works with updated dependency).

As for now, 1.36.14 doesn't have the problem, Logs are green.

abishai, 2023-01-15:
1.36.15 doesn't look not good, first compilation failure in zm_time, fixedf with https://github.com/ZoneMinder/zoneminder/commit/df414fc028aebddd0105c7405dbccfe61dc7458a Now I have linker issue because of curl I disabled long time ago in cmake... Going to investigate it. As 1.36.16 compiles normally, I suppose 1.36.15 was buggy...

abishai, 2023-01-15:
```
abishai@observer-test:~ % doas -u www zmc -m 6
01/15/23 16:55:49.342352 zmc_m6[570458].ERR-zm_monitor.cpp/804 [Can't open memory map file /dev/shm/zm.mmap.6: No such file or directory]

01/15/23 16:55:49.342445 zmc_m6[570458].WAR-zmc.cpp/241 [Couldn't connect to monitor 6]

01/15/23 16:55:49.342513 zmc_m6[570458].ERR-zm_signal.cpp/50 [Got signal 11 (Segmentation fault), crashing]

01/15/23 16:55:49.342568 zmc_m6[570458].ERR-zm_signal.cpp/80 [Signal address is 0x60, from 0x23831e]
```

I see. We don't have /dev/shm . We have tmpfs for that and can mount it to directory if needed. I suspect 1.36.15 is development branch then? In 1.36.xxx we can set shm directory in ZM config

abishai, 2023-01-15:
OK, as I can't test 1.36.15, we can assume that the issue was introduced somewhere between 1.36.14 and 1.36.16. If monitor writes, frame queue is overflowed for some reason. In 1.36.16 zmc is killed by watchdog, in 1.36.32 things become even worse, it exhausts the memory and crashes.

Any idea of suspicious places to look?

PS: If 1.37.xxx really has /dev/shm hardcoded, this will be another issue in future. We need mechanism to change the path and move it out of /dev. But I wouldn't worry about this now.

connortechnology, 2023-01-15:
I don't think there is much point in testing each version, but if you are going to do so, use git bisect.  

/dev/shm is not hard coded.  It gets the value from /etc/zm.conf or other files in /etc/zm/conf.d.  Generally these files are pre-populated by cmake build system.  

ANyways, I will get my build system running today and likely have it all figured out soon.  

I am concerned about this lack of curl.  We really need it for modern features.

abishai, 2023-01-15:
The lack of curl was arbitrary.. When I ported 1.30 I had no idea how to test it and simply disabled it with ZM_NO_CURL=ON (with VLC). They looked like marginal backends superseded by ffmpeg
I'll keep CURL ON if you tell it's important.

VVD, 2023-01-17:
Every 10 minutes:
```
2023-01-17 17:51:40 | zmc_m2 |   | 154295 | ERR | Signal address is 0xaa, from 0x28d612 | zm_signal.cpp | 80
-- | -- | -- | -- | -- | -- | -- | --
2023-01-17 17:51:40 | zmc_m2 |   | 154295 | ERR | Got signal 11 (Segmentation fault), crashing | zm_signal.cpp | 50
2023-01-17 17:51:40 | zmc_m2 |   | 147930 | WAR | Found locked packet when trying to free up video packets. This basically means that decoding is not keeping up. | zm_packetqueue.cpp | 156
-- | -- | -- | -- | -- | -- | -- | --
2023-01-17 17:51:38 | zmc_m2 |   | 147930 | WAR | You  have set the max video packets in the queue to 100. The queue is full.  Either Analysis is not keeping up or your camera's keyframe interval 21  is larger than this setting. | zm_packetqueue.cpp | 139
```
1.36.32 buggy.

connortechnology, 2023-01-17:
In the directions linked to, some additions/comments:

You missed the step where you actually start fcgiwrap
In the nginx config, should index index.php not end in a semi colon?

I have my system up and running with 1.36.32,  It has some serious issues.  First off, it seems to be trying to save event data to /var/db/zoneminder/events but that dir didn't get created/ownership assigned.

VVD, 2023-01-17:
Check this: https://bugs.freebsd.org/bugzilla/show_bug.cgi?id=267304

abishai, 2023-01-17:
@connortechnology 
You are right, manual can be improved.
As for directory, I was sure ZM creates it, providing it owns /var/db/zoneminder. I'll add directory.

Is port runs for you or you need some help?

connortechnology, 2023-01-17:
Have it installed and running, but it is crashing. Working on it.

connortechnology, 2023-01-17:
How can I get the debug symbols from ports ?

connortechnology, 2023-01-17:
I have solved the crash I was experiencing. Something I had fixed in master.  I will release 1.36.33 this week with the fix.  
779ae5051e43629189b7f6780978676a7c6a21ca

abishai, 2023-01-17:
> How can I get the debug symbols from ports ?

```
echo WITH_DEBUG_PORTS= multimedia/zoneminder > /etc/make.conf
```

abishai, 2023-01-17:
For me, situation doesn't improve with the patch. Monitors in record and mocord mode are crashing. Modect mode works outside of alarm state.

```
1/17/23, 11:22:55 PM GMT+3 | zmdc |   | 88869 | ERR | 'zmc -m 6' exited abnormally, exit status 11 | zmdc.pl | -
-- | -- | -- | -- | -- | -- | -- | --
1/17/23, 11:22:55 PM GMT+3 | zmc_m6 |   | 715820 | ERR | Signal address is 0xdb, from 0x27f012 | zm_signal.cpp | 80
1/17/23, 11:22:55 PM GMT+3 | zmc_m6 |   | 715820 | ERR | Got signal 11 (Segmentation fault), crashing | zm_signal.cpp | 50
1/17/23, 11:22:53 PM GMT+3 | zmdc |   | 88869 | ERR | 'zmc -m 8' exited abnormally, exit status 11 | zmdc.pl | -
1/17/23, 11:22:52 PM GMT+3 | zmc_m8 |   | 715886 | ERR | Signal address is 0xa5, from 0x27f012 | zm_signal.cpp | 80
1/17/23, 11:22:52 PM GMT+3 | zmc_m8 |   | 715886 | ERR | Got signal 11 (Segmentation fault), crashing | zm_signal.cpp | 50
1/17/23, 11:22:48 PM GMT+3 | zmc_m8 |   | 586423 | WAR | Found locked packet when trying to free up video packets. This basically means that decoding is not keeping up. | zm_packetqueue.cpp | 156
1/17/23, 11:22:48 PM GMT+3 | zmc_m8 |   | 586423 | WAR | You  have set the max video packets in the queue to 80. The queue is full.  Either Analysis is not keeping up or your camera's keyframe interval 0  is larger than this setting. | zm_packetqueue.cpp | 139
1/17/23, 11:22:40 PM GMT+3 | zmwatch |   | 88989 | WAR | Restarting capture daemon for 5 Monitor-5, time since last capture 53 seconds (1673986960-1673986907) | zmwatch.pl | -
1/17/23, 11:22:35 PM GMT+3 | zmc_m6 |   | 523438 | WAR | Found locked packet when trying to free up video packets. This basically means that decoding is not keeping up. | zm_packetqueue.cpp | 156
1/17/23, 11:22:35 PM GMT+3 | zmc_m6 |   | 523438 | WAR | You  have set the max video packets in the queue to 80. The queue is full.  Either Analysis is not keeping up or your camera's keyframe interval 0  is larger than this setting. | zm_packetqueue.cpp | 139
1/17/23, 11:22:09 PM GMT+3 | zmwatch |   | 88989 | WAR | Restarting capture daemon for 6 Monitor-6, time since last capture 47 seconds (1673986929-1673986882) | zmwatch.pl | -
1/17/23, 11:22:01 PM GMT+3 | zmc_m4 |   | 610469 | WAR | Found locked packet when trying to free up video packets. This basically means that decoding is not keeping up. | zm_packetqueue.cpp | 156
1/17/23, 11:21:59 PM GMT+3 | zmc_m4 |   | 610469 | WAR | You  have set the max video packets in the queue to 80. The queue is full.  Either Analysis is not keeping up or your camera's keyframe interval 9  is larger than this setting. | zm_packetqueue.cpp | 139
1/17/23, 11:21:49 PM GMT+3 | zmdc |   | 88869 | ERR | 'zmc -m 5' exited abnormally, exit status 11 | zmdc.pl | -
1/17/23, 11:21:48 PM GMT+3 | zmc_m5 |   | 715360 | ERR | Signal address is 0x181, from 0x27f012 | zm_signal.cpp | 80
1/17/23, 11:21:48 PM GMT+3 | zmc_m5 |   | 715360 | ERR | Got signal 11 (Segmentation fault), crashing | zm_signal.cpp | 50
1/17/23, 11:21:47 PM GMT+3 | zmdc |   | 88869 | ERR | 'zmc -m 8' exited abnormally, exit status 11 | zmdc.pl | -
1/17/23, 11:21:46 PM GMT+3 | zmc_m8 |   | 715593 | ERR | Signal address is 0xb8, from 0x27f012 | zm_signal.cpp | 80
1/17/23, 11:21:46 PM GMT+3 | zmc_m8 |   | 715593 | ERR | Got signal 11 (Segmentation fault), crashing | zm_signal.cpp | 50

```

connortechnology, 2023-01-17:
I can now view jpeg saved events after applying 8300a7b .  What I am seeing now is the that opening a new event is taking forever.  .2 seconds to do the event insert into the db.  So the capture queues fill up.  

Also, decoding is not keeping up.  I don't know why freebsd is so slow for decoding. It can't keep up with a 1080p stream in this vm.  Similar Ubuntu vm's have no trouble.

abishai, 2023-01-18:
I have https://github.com/ZoneMinder/zoneminder/commit/779ae5051e43629189b7f6780978676a7c6a21ca applied only.

The decoding is not keeping up since 1.32.16. Currently, my installation contains 12 5MP cameras on 4fps and they consume 4-5 cores. The same warnings exist in 1.36.12 zm_packetqueue.cpp but logs are silent.

I'll try to enable debug logs then to see timings.

[12 more replies]

commit 577c827b (2023-01-17): Upgrade sendfile7 to not mash size on stack.  offset is a nullptr most of the time, don't dereference it.  When falling back to read/write, just read in chunks of 4096, because I don't know how to make llvm use c99 dynamic arrays. Check for HAVE_SENDFILE. Fixes #3505
commit 8300a7b4 (2023-01-17): Upgrade sendfile7 to not mash size on stack.  offset is a nullptr most of the time, don't dereference it.  When falling back to read/write, just read in chunks of 4096, because I don't know how to make llvm use c99 dynamic arrays. Check for HAVE_SENDFILE. Fixes #3505
closed 2023-04-26 by connortechnology