ZoneMinder/zoneminder #3507: Update Ubuntu zoneminder 1.37.17 montage stopped streamining

jaapf4fs, 2022-06-16:
After update zoneminder on Ubuntu to 1.37.17 montage stopped streaming.
Log:

16/06/2022 - 13:08:28 | web_js |   | 24010 | ERR | TypeError: this.streamQuery is undefined | zm/cache/js_MonitorStream-dark-1655312991.js | 167

Best regards
Jaap

jaapf4fs, 2022-06-16:
OK
But I don' t expect this is a big issue so I thought just a quick note would be enough, sorry..

Upgraded ZM 5 minutes ago same problem (log):
16/06/2022 - 17:27:33 | web_js |   | 4186 | ERR | TypeError: this.streamQuery is undefined | zm/cache/js_MonitorStream-dark-1655387944.js | 167

I will use the template tomorrow when I have more time.

Best regards,
Jaap

connortechnology, 2022-06-16:
I suspect it is just taking a while for the packages to build. Check back when you can.  I do believe the issue is fixed.

commit 58fedd8c (2022-06-17): Fix streamQuery => streamCmdQuery. Fixes #3507
closed 2022-06-17 by connortechnology