ZoneMinder/zoneminder #3510: ZM Event web paging issue

basildane, 2022-06-22:
**Describe Your Environment**
- Version of ZoneMinder  [1.36.19]
- How you installed ZoneMinder [PPA]
- Full name and version of OS [Ubuntu 20.04]
- Browser name and version [Brave 1.40.105]

**If the issue concerns a camera**

**Describe the bug**
When "paging", using the page controls at the bottom of the events page, the current page is remembered between sessions and events.  For example, you open a camera and browse to page 5.  Open another camera, the NEW camera you just opened is displaying page 5.  It should default to page 1.  Even worse, when visiting the page even days later, it remembers what page you were on in the last session and puts you into that page by default.  So you end up looking at events that are days old wondering why, then you have to go to the bottom and click "1" - every single time.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Console
2. Click on any channel
3. Click on a page (ie 5)
4. Go to another channel
5. You are on page 5 on the Other channel

**Expected behavior**
When opening the application fresh, or when changing to a new channel, you should start at page 1.

**Debug Logs**

connortechnology, 2023-01-03:
The issue is that we use this cookie feature to remember the visible columns, etc as well. It would be best to not use it for paging I think.

connortechnology, 2023-01-03:
SO I looked into it.  We can turn off the cookie for the pagenumber no problem.  However then if we do a page reload, it puts us back on page 1.  

I kinda think that is preferable to the current behaviour.  @knight-of-ni do you have any thoughts on the matter?

basildane, 2023-01-03:
Another option would be to include the channel number in the cookie, so the cookie would intelligently follow the page for the correct channel.  But, this sounds good too.

commit 8370ae6c (2023-02-16): Set a short expiry 1min and set the cookie name to include the filter so that each and every filter gets it;s own pagination saved.  Fixes #3510
commit a2e9cc23 (2023-02-16): Set a short expiry 1min and set the cookie name to include the filter so that each and every filter gets it;s own pagination saved.  Fixes #3510
closed 2023-02-16 by connortechnology