ZoneMinder/zoneminder #3511: Zoneminder camera won't run after OS update

PaperOrb, 2022-06-24:
**Describe Your Environment**
- ZoneMinder 1.36.12+dfsg1-1
- installed from PPA
- was popOS 21.10 when it was working. Stop working after upgrading to 22.04.
- Browser name is firefox 101.01

**If the issue concerns a camera**
- Chicony Electronics Co., Ltd HD Webcam (info from lspci. it's an inbuilt webcam)
- fps = unsure
- resolution = 720p max
- ZoneMinder Source Type: local

**Describe the bug**
I updated from popOS 21.10 to 22.04 and now my /dev/video0 camera shows as not running. Checked err and war in debug logs but couldn't make heads or tails of it. I restarted ZM from the terminal as well as the web interface. 

**To Reproduce**
Steps to reproduce the behavior:
Update from popOS 21.10 to 22.04.

**Expected behavior**
Camera should be running.

**Debug Logs**
[tableExport.txt](https://github.com/ZoneMinder/zoneminder/files/8972188/tableExport.txt)

baudneo, 2022-06-24:
Is the path of the camera still as it was before the upgrade?

PaperOrb, 2022-06-24:
> Is the path of the camera still as it was before the upgrade?
Yep, still the same.

baudneo, 2022-06-24:
Can you post some of the logs, if it's a large paste do a link to Pastebin or a GitHub gist.

PaperOrb, 2022-06-24:
> Can you post some of the logs, if it's a large paste do a link to Pastebin or a GitHub gist.

Logs are in the OP unless you mean a different set of logs?

baudneo, 2022-06-24:
Sorry didn't see the logs first time. So it doesn't really give much info besides saying it crashed. I do see that the font location is incorrect, try fixing that first and see if same behavior. 

Run find for the font file and then in the zm web gui options there is a setting for font path. I think it's in the system options. You may have to look around to find the font config though.

Do you know how to turn on debug logging for zmc?

PaperOrb, 2022-06-24:
Looking for the font still. I switched LOG_LEVEL_SYSLOG, LOG_LEVEL_FILE, LOG_LEVEL_WEBLOG, LOG_LEVEL_DATABASE to debug level. Here's the log file:
[tableExport(1).txt](https://github.com/ZoneMinder/zoneminder/files/8972263/tableExport.1.txt)

PaperOrb, 2022-06-24:
Apparently ZM is looking for the font at  /usr/share/zoneminder/fonts/default.zmfnt but there's not fonts directory inside zoneminder. Any way to get it to generate this?

baudneo, 2022-06-24:
Turn file logging level to debug, I wouldn't have database or syslog set to debug. It will fill your drive fast.

Enable debug logging by making sure the debug logging checkbox is ticked then in the logging components textbox add _zmc and then click save. After saving you must restart zoneminder. Use systemctl restart zoneminder.service.  after zm restarts it will start actually logging DBG level messages.

Did you run find for default.zmfnt? It's nowhere systemwide?

PaperOrb, 2022-06-24:
I'm not sure which option you mean by "in the logging components textbox add _zmc". Also I found and copied default.zmfnt. I edited my prior post with the details and the log.

Also this is how my UI looks if I enable the log_debug box. 
[image]

baudneo, 2022-06-24:
LOG_DEBUG_TARGET textbox, it should be right below the enable debug logging checkbox.

[image]

baudneo, 2022-06-24:
I would maybe try reinstalling ZM `sudo apt install --reinstall zoneminder` to see if that fixes those other issues your WEB GUI is showing as well.

PaperOrb, 2022-06-24:
Looks like nothing is being logged now. The GUI issues were fixed after I put in _zmc like you said. Here's my logging settings:
[image]
[image]
[image]

baudneo, 2022-06-24:
Your log debug level is set to 1 which is 1 step above off, the scale is 0 (off) to 9, set the level to 5 for now, click save, restart ZM using `sudo systemctl restart zoneminder.service` and then go into a terminal and `tail -f /var/log/zm/zmc_m1.log`

PaperOrb, 2022-06-24:
> Your log debug level is set to 1 which is 1 step above off, the scale is 0 (off) to 9, set the level to 5 for now, click save, restart ZM using `sudo systemctl restart zoneminder.service` and then go into a terminal and `tail -f /var/log/zm/zmc_m1.log`

Thanks, changed it. After restarting via that command, tail reveals nothing in the zmc_m1.log (that's the correct ID though). Also the log inside the ZM web GUI is showing nothing but a single message:
```
2022-06-23 21:27:03 | zmc_dvideo0 |   | 106520 | FAT | Can't find swscale format for palette 1 | zm_local_camera.cpp | 146
```

baudneo, 2022-06-24:
What permissions are set to the `/dev/video0` file? My guess is that the ZM user (usually www-data in debian based distros) may not have access to the file after the upgrade. try `ls -al /dev/video0` and post output.

Turn off file debugging, turn on syslog debugging, remove zmc from the component list, click save and restart ZM. All components will have debug logging on so there will be alot of logs in syslog. You can tail the syslog file to follow it. `sudo tail -f /var/log/syslog`

baudneo, 2022-06-24:
the name of the logfile should be zmc_dvideo0 instead of zmc_m1 btw.

PaperOrb, 2022-06-24:
> What permissions are set to the `/dev/video0` file? My guess is that the ZM user (usually www-data in debian based distros) may not have access to the file after the upgrade. try `ls -al /dev/video0` and post output. Could try `sudo -u www-data cat /dev/video0` to see if it gets a permission denied error, that will tell you if www-data has access to that file.
> 
> Turn off file debugging, turn on syslog debugging, remove zmc from the component list, click save and restart ZM. All components will have debug logging on so there will be alot of logs in syslog. You can tail the syslog file to follow it. `sudo tail -f /var/log/syslog`

Output from `ls -al /dev/video0` is `crw-rw----+ 1 root video 81, 0 Jun 23 12:33 /dev/video0`.

Here's a fraction of the text inside /var/log/syslog: [edited_log.txt](https://github.com/ZoneMinder/zoneminder/files/8972424/edited_log.txt)

baudneo, 2022-06-24:
`Can't find swscale format for palette 1` is what seems to be crashing things. I do not have the knowledge of ZM to debug that issue. One thing I would try is adding the www-data user to the video group though by issuing this command `sudo usermod -aG video www-data` and then restarting ZM. You may have to restart the machine for the new permissions to take full effect, YMMV.

baudneo, 2022-06-24:
swscale search on google came up with `Package swscale performs highly optimized image scaling and colorspace and pixel format conversion operations.`

I also noticed in the logs that the monitor is set to an 'Unknown' type. You may want to create a new monitor and re-set up the device. Upgrading may have borked some stuff in weird ways. Also try reinstalling ZM, if you only have 1 monitor it shouldnt be too hard to reconfigure ZM. You could also sqldrop your DB and then import it upon reinstall so all your config is saved and reimported. See the docs and the Backup/Backup config only sections -> https://wiki.zoneminder.com/MySQL

PaperOrb, 2022-06-24:
> swscale search on google came up with `Package swscale performs highly optimized image scaling and colorspace and pixel format conversion operations.`
> 
> I also noticed in the logs that the monitor is set to an 'Unknown' type. You may want to create a new monitor and re-set up the device. Upgrading may have borked some stuff in weird ways. Also try reinstalling ZM, if you only have 1 monitor it shouldnt be too hard to reconfigure ZM. I wouldnt mess around with the SQL stuff right now to backup and restore unless you are somewhat experianced with ZM and SQL.

Thanks, it looks like creating a new monitor with the /dev/video0 camera worked.

closed 2022-06-24 by PaperOrb