ZoneMinder/zoneminder #3512: FCM eol?

mfauvain, 2022-06-25:
**Describe Your Environment**
- Version of ZoneMinder 1.36.19
- How you installed ZoneMinder : AUR
- Full name and version of OS : Arch kernel 5.18.6

Hi, getting a fcm server error and not receiving alerts anymore. Is this a problem on my side or is it general due to zmeventnotification EOL?
Below is the error in zmeventnotification.log:
```
fcmv1: FCM push message error 
<html><head>
<meta http-equiv="content-type" content="text/html;charset=utf-8">
<title>500 Server Error</title>
</head>
<body text=#000000 bgcolor=#ffffff>
<h1>Error: Server Error</h1>
<h2>The server encountered an error and could not complete your request.<p>Please try again in 30 seconds.</h2>
<h2></h2>
</body></html>]
```

baudneo, 2022-06-25:
Zmninja went end of life. Asker deactivated his fcm cloud server. This is expected behavior. You can spin up your own instance or, I think ic0n has set up zoneminders own fcm instance. See the zoneminder forum post about zmninja end of life, I think there are instructions on setting up your own fcm server.

Another option is to use the pushhapi_pushover.py script provided in ZMES to use pushover as the push notification client.

baudneo, 2022-06-25:
https://forums.zoneminder.com/viewtopic.php?t=30996&sid=fd597ce519a8bc03a85afbb981f751af

mfauvain, 2022-06-25:
Ok, thx for confirming. Will check how to setup my own fcm.

jaredvacanti, 2022-11-12:
@mfauvain , can you expand on if your homebridge setup gets APNS push notifications? I'm assuming you use https://github.com/Sunoo/homebridge-camera-ffmpeg or https://github.com/seydx/homebridge-camera-ui to view ZM cameras. ZMeventnotification sends messages to mosquitto or embedded mqtt server in homebridge, how does that translate to a push notification on your device?

mfauvain, 2022-11-23:
yeah, using [https://github.com/Sunoo/homebridge-camera-ffmpeg](https://github.com/Sunoo/homebridge-camera-ffmpeg) and zmeventnotification MQTT. However, zmevent output has to be parsed and re-contributed to mqtt so that homebridge camera plugin 'reads' it correctly. I use mqtt-launcher and a python script to parse the zmevent json.
With this, alerts popup on iPhone.
I also use a MQTT daemon on my laptop so I receive alerts + images here also.
Hope this helps.