ZoneMinder/zoneminder #3513: Zoneminder web interface not working

CaverJ, 2022-06-26:
**Describe Your Environment**
- The latest version updated on June 25th, 2022 (web interface is not working, so I cannot find out the version number)
- dnf upgrade
- Rocky Linux 8.6
- Firefox (many different versions) and Chrome

**If the issue concerns a camera**
- Make and Model
- frame rate
- resolution
- ZoneMinder Source Type:

**Describe the bug**
The web interface is not coming up. Zone minder is also not working with zmNinja on my Android phone.

**To Reproduce**
Steps to reproduce the behavior:
1. Update to latest from June 25th, 2022
2. The web interface is not coming up

**Expected behavior**
A clear and concise description of what you expected to happen.

**Debug Logs**
```

<insert debug logs here, please make sure they are within the ``` quotes so they are formatted properly>

```

baudneo, 2022-06-26:
Need logs! We can't really help you without knowing what's going on.

ZM debug level logs and apache/nginx logs depending on what http server is being used to serve ZM web gui.

CaverJ, 2022-06-26:
> Need logs! We can't really help you without knowing what's going on.
> 
> ZM debug level logs and apache/nginx logs depending on what http server is being used to serve ZM web gui.

[zmdc.log](https://github.com/ZoneMinder/zoneminder/files/8986842/zmdc.log)
[zmfilter_1.log](https://github.com/ZoneMinder/zoneminder/files/8986843/zmfilter_1.log)
[zmfilter_2.log](https://github.com/ZoneMinder/zoneminder/files/8986844/zmfilter_2.log)
[zmpkg.log](https://github.com/ZoneMinder/zoneminder/files/8986845/zmpkg.log)

baudneo, 2022-06-26:
Those are all INFO level logs. The problem being that you can't use the web gui to configure debug level logs though.

Try the Apache logs at `/var/log/apache2/error.log` (your log for may be elsewhere but it should be similar)

Also try restarting apache2.service as that's the default http server zm uses. I've had issues where restarting Apache brought web gui back up after upgrading.

Another option is to configure Apache to output debug level logs in it's configs. Also take a look at syslog to see if there are any clues in there.

The only other way I know or configuring debt level logs for zmc would be using some SQL to modify the DB to enable debug logs.

CaverJ, 2022-06-26:
[error_log.log](https://github.com/ZoneMinder/zoneminder/files/8986886/error_log.log)

CaverJ, 2022-06-26:
Found the issue - The database was outdated (was 1.36.12), and needed to be updated (now 1.36.19) using:
sudo zmupdate.pl

Then restarted Apache and Zoneminder:
sudo systemctl restart httpd
sudo systemctl start zoneminder

closed 2022-06-26 by CaverJ

connortechnology, 2022-06-27:
That error_log indicates a complete lack of apache2 configuration for zoneminder.  However that may be a false flag.
1.36.17 had an issue where single quotes or apostrophes in some config entries would break javascript.  1.36.18/.19 should fix it.  

If you think this could be the case, maybe downgrade, edit config, then upgrade again.