ZoneMinder/zoneminder #3516: LDAP auth vs database case sensitivity leads to HTTP 500

megasaturnv, 2022-07-01:
To allow Zoneminder to authenticate with LDAP an admin will set AUTH_TYPE to "remote" and configure apache to authenticate with the LDAP server. Then, the same username is created in Zoneminder. Problem is, LDAP authentication is case insensitive a user can login with a valid account which doesn't match the case used for Zoneminder's user.

e.g. user in Zoneminder's database might be "John.Doe" but LDAP would allow "john.doe" to authenticate

After authenticating with the correct username but doesn't match the exact case used for the user in the Zoneminder database, the user is given a HTTP error 500

A temporary solution we have is to change
https://github.com/ZoneMinder/zoneminder/blob/2e22cd5c3b489ee05a2cf6ff749df8995f7e6d38/web/includes/auth.php#L323
to `$user = dbFetchOne($sql, NULL, array(strtolower($_SERVER['REMOTE_USER'])));` and then only create lower-case usernames in Zoneminder's database. But, we have to change this file manually after every update. I was thinking of making a pull request for this, but it would break users that just use simple "htpasswd" authentication (and don't use lower case usernames)

Are there any long-term solutions for this? Thanks

connortechnology, 2022-08-26:
I think we would simply add an additional configuration option, telling ZM to use this code.  Something like ZM_CASE_INSENSITIVE_USERNAMES. Then people can do whatever they need to do.   Pretty trivial. Do you want to fund an hour of my time to get it into the next ZM release?

megasaturnv, 2022-09-01:
Thanks. I'll fork it and do a PR

megasaturnv, 2022-10-10:
Just re-opening this issue to check something. Do I need to do a PR specifically to merge these changes into release-1.36 (rather than master), or are these changes scheduled to be in the next major release?

I don't mind doing a fresh PR for this and testing again if needed, since there are major differences between master and release-1.36.

Thanks!

connortechnology, 2022-10-11:
It has been merged to release-1.36 and will be released in 1.36.29 today.

closed 2022-10-11 by connortechnology