ZoneMinder/zoneminder #3533: zmc Segfault on 1.36.20

apalacheno, 2022-07-17:
On Ubuntu 20.04 using the official PPA, since the 1.36.20 update ZoneMinder can't record video footage anymore and _zmc_ gets restarted every few minutes. The issue occured right after the automatic package update on 2022-07-14 and persists as of today.

The cam in question, a Hikvision DS-2CD2147G2-SU, is using Modect. Before the update, everything worked fine, no problems with 0B recordings. No change on the setup or the cams has been done. A host and cam restart has been performed, but without any change in behavior.

A typical error log looks like this:

**zmc_m1.log:**
```
07/17/22 22:06:24.728923 zmc_m1[946774].INF-zm_monitor.cpp/1680 [Hauseingang: 1240 - Capturing at 24.99 fps, capturing bandwidth 477564bytes/sec Analysing at 24.99 fps]
07/17/22 22:06:25.929817 zmc_m1[946774].INF-zm_monitor.cpp/1680 [Hauseingang: 1270 - Capturing at 24.98 fps, capturing bandwidth 478316bytes/sec Analysing at 24.98 fps]
07/17/22 22:06:27.129041 zmc_m1[946774].INF-zm_monitor.cpp/1680 [Hauseingang: 1300 - Capturing at 25.02 fps, capturing bandwidth 527548bytes/sec Analysing at 25.02 fps]
07/17/22 22:06:28.327972 zmc_m1[946774].INF-zm_monitor.cpp/1680 [Hauseingang: 1330 - Capturing at 25.02 fps, capturing bandwidth 851106bytes/sec Analysing at 25.02 fps]
07/17/22 22:06:28.712768 zmc_m1[946853].INF-zm_monitor.cpp/1980 [Hauseingang: 1339 - Gone into alarm state PreAlarmCount: 0 > AlarmFrameCount:1 Cause:Motion Garage3]
07/17/22 22:06:29.529049 zmc_m1[946774].INF-zm_monitor.cpp/1680 [Hauseingang: 1360 - Capturing at 24.98 fps, capturing bandwidth 608786bytes/sec Analysing at 8.33 fps]
07/17/22 22:06:30.196461 zmc_m1[946853].INF-zm_monitor.cpp/1987 [Hauseingang: 1338 - Opening new event 10049, alarm start]
07/17/22 22:06:30.196895 zmc_m1[947414].INF-zm_videostore.cpp/532 [some options not used, turn on debugging for a list.]
07/17/22 22:06:30.647503 zmc_m1[947414].ERR-zm_signal.cpp/50 [Got signal 11 (Segmentation fault), crashing]
07/17/22 22:06:30.647545 zmc_m1[947414].ERR-zm_signal.cpp/80 [Signal address is (nil), from 0x5651d63cf90d]
07/17/22 22:06:30.647695 zmc_m1[947414].ERR-zm_signal.cpp/101 [Backtrace 0: /usr/bin/zmc(+0xc43d1) [0x5651d63453d1]]
07/17/22 22:06:30.647704 zmc_m1[947414].ERR-zm_signal.cpp/101 [Backtrace 1: /lib/x86_64-linux-gnu/libpthread.so.0(+0x14420) [0x7f6f6bfd3420]]
07/17/22 22:06:30.647716 zmc_m1[947414].ERR-zm_signal.cpp/101 [Backtrace 2: /usr/bin/zmc(+0x14e90d) [0x5651d63cf90d]]
07/17/22 22:06:30.647728 zmc_m1[947414].ERR-zm_signal.cpp/101 [Backtrace 3: /usr/bin/zmc(+0xfb078) [0x5651d637c078]]
07/17/22 22:06:30.647825 zmc_m1[947414].ERR-zm_signal.cpp/101 [Backtrace 4: /usr/bin/zmc(+0xfc8bc) [0x5651d637d8bc]]
07/17/22 22:06:30.647831 zmc_m1[947414].ERR-zm_signal.cpp/101 [Backtrace 5: /lib/x86_64-linux-gnu/libstdc++.so.6(+0xd6de4) [0x7f6f69bc5de4]]
07/17/22 22:06:30.647837 zmc_m1[947414].ERR-zm_signal.cpp/101 [Backtrace 6: /lib/x86_64-linux-gnu/libpthread.so.0(+0x8609) [0x7f6f6bfc7609]]
07/17/22 22:06:30.647842 zmc_m1[947414].ERR-zm_signal.cpp/101 [Backtrace 7: /lib/x86_64-linux-gnu/libc.so.6(clone+0x43) [0x7f6f698b2133]]
07/17/22 22:06:30.647847 zmc_m1[947414].INF-zm_signal.cpp/108 [Backtrace complete, please execute the following command for more information: addr2line -e /usr/bin/zmc 0x5651d63453d1 0x7f6f6bfd3420 0x5651d63cf90d 0x5651d637c078 0x5651d637d8bc 0x7f6f69bc5de4 0x7f6f6bfc7609 0x7f6f698b2133]
07/17/22 22:06:30.662663 zmc_m1[946774].INF-zm_monitor.cpp/3060 [Hauseingang: image_count:1388 - Closing event 10049, shutting down]
07/17/22 22:06:36.526535 zmc_m1[947471].INF-zmc.cpp/220 [Starting Capture version 1.36.20]
07/17/22 22:06:42.288626 zmc_m1[947471].INF-zm_monitor.cpp/1680 [Hauseingang: 0 - Capturing at 0.00 fps, capturing bandwidth 14bytes/sec Analysing at 0.00 fps]
07/17/22 22:06:43.478383 zmc_m1[947471].INF-zm_monitor.cpp/1680 [Hauseingang: 160 - Capturing at 134.48 fps, capturing bandwidth 3585016bytes/sec Analysing at 48.75 fps]
07/17/22 22:06:44.678439 zmc_m1[947471].INF-zm_monitor.cpp/1680 [Hauseingang: 190 - Capturing at 25.00 fps, capturing bandwidth 535039bytes/sec Analysing at 65.00 fps]
07/17/22 22:06:45.888624 zmc_m1[947471].INF-zm_monitor.cpp/1680 [Hauseingang: 220 - Capturing at 24.79 fps, capturing bandwidth 525046bytes/sec Analysing at 67.76 fps]
07/17/22 22:06:47.084399 zmc_m1[947471].INF-zm_monitor.cpp/1680 [Hauseingang: 250 - Capturing at 25.09 fps, capturing bandwidth 529509bytes/sec Analysing at 25.09 fps]
```
**web_php.log:**
```
7/16/22, 7:57:59 PM GMT+2.551221 web_php[2769].ERR [10.30.0.11] [Can't create frame images from video for this event 9715-video.mp4

                Command was: /usr/bin/ffmpeg -ss 1 -i /video/1/2022-07-16/9715/9715-video.mp4 -frames:v 1 /video/1/2022-07-16/9715/snapshot.jpg 2>&1

                Output was: ffmpeg version 4.2.7-0ubuntu0.1 Copyright (c) 2000-2022 the FFmpeg developers
  built with gcc 9 (Ubuntu 9.4.0-1ubuntu1~20.04.1)
  configuration: --prefix=/usr --extra-version=0ubuntu0.1 --toolchain=hardened --libdir=/usr/lib/x86_64-linux-gnu --incdir=/usr/include/x86_64-linux-gnu --arch=amd64 --enable-gpl --disable-stripping --enable-avresample --disable-filter=resample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librsvg --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opencl --enable-opengl --enable-sdl2 --enable-libdc1394 --enable-libdrm --enable-libiec61883 --enable-nvenc --enable-chromaprint --enable-frei0r --enable-libx264 --enable-shared
  libavutil      56. 31.100 / 56. 31.100
  libavcodec     58. 54.100 / 58. 54.100
  libavformat    58. 29.100 / 58. 29.100
  libavdevice    58.  8.100 / 58.  8.100
  libavfilter     7. 57.100 /  7. 57.100
  libavresample   4.  0.  0 /  4.  0.  0
  libswscale      5.  5.100 /  5.  5.100
  libswresample   3.  5.100 /  3.  5.100
  libpostproc    55.  5.100 / 55.  5.100
[mov,mp4,m4a,3gp,3g2,mj2 @ 0x56112bd7e6c0] Could not find codec parameters for stream 0 (Video: h264 (avc1 / 0x31637661), none, 2688x1520): unspecified pixel format
Consider increasing the value for the 'analyzeduration' and 'probesize' options
/video/1/2022-07-16/9715/9715-video.mp4: could not seek to position 1.000
Input #0, mov,mp4,m4a,3gp,3g2,mj2, from '/video/1/2022-07-16/9715/9715-video.mp4':
  Metadata:
    major_brand     : isom
    minor_version   : 512
    compatible_brands: isomiso2avc1iso6mp41
    title           : Zoneminder Security Recording
    encoder         : Lavf58.29.100
  Duration: N/A, bitrate: N/A
    Stream #0:0(und): Video: h264 (avc1 / 0x31637661), none, 2688x1520, SAR 1:1 DAR 168:95, 90k tbr, 90k tbn, 180k tbc (default)
    Metadata:
      handler_name    : VideoHandler
Stream mapping:
  Stream #0:0 -> #0:0 (h264 (native) -> mjpeg (native))
Press [q] to stop, [?] for help
Cannot determine format of input stream 0:0 after EOF
Error marking filters as finished
Conversion failed!] at /usr/share/zoneminder/www/views/image.php line 200
```

connortechnology, 2022-07-18:
You are going to have to turn on debug logging in order to give us a better idea of where it is crashing.  option->LOgging->DEBUG.  Set level to 4.

apalacheno, 2022-07-19:
With debug _level=4_ there is a lot of output. Not sure where to start, but I try to paste the milliseconds before and after the segfault. See timestamp codes **08:17:56.263370** and **08:17:56.273250**.

If you need more lines before or after the event, please let me know. I've got 12 GB of logs at hand :)

```07/18/22 08:17:56.240331 zmc_m1[34394].DB1-zm_zone_stats.h/54 [ZoneStat: GetStats zone_id: 5 pixel_diff=0 alarm_pixels=0 alarm_filter_pixels=0 alarm_blob_pixels=0 alarm_blobs=0 min_blob_size=0 max_blob_size=0 alarm_box=(0,0=>0,0) alarm_center=(0,0) score=0]
07/18/22 08:17:56.240337 zmc_m1[34394].DB1-zm_zone_stats.h/54 [ZoneStat: After detect motion zone_id: 5 pixel_diff=0 alarm_pixels=0 alarm_filter_pixels=0 alarm_blob_pixels=0 alarm_blobs=0 min_blob_size=0 max_blob_size=0 alarm_box=(0,0=>0,0) alarm_center=(0,0) score=0]
07/18/22 08:17:56.240343 zmc_m1[34394].DB3-zm_monitor.cpp/1872 [After motion detection, score:0 last_motion_score(2), new motion score(0)]
07/18/22 08:17:56.240353 zmc_m1[34394].INF-zm_monitor.cpp/2019 [Hauseingang: 65959 - Gone into alert state]
07/18/22 08:17:56.240369 zmc_m1[34394].DB1-zm_monitor.cpp/2120 [Blending]
07/18/22 08:17:56.242707 zmc_m1[34350].DB1-zm_db.cpp/190 [Success running sql query INSERT INTO `Logs` ( `TimeKey`, `Component`, `ServerId`, `Pid`, `Level`, `Code`, `Message`, `File`, `Line` ) VALUES ( 1658125076.219110, 'zmc_m1', 0, 79049, 0, 'INF', 'some options not used, turn on debugging for a list.', 'zm_videostore.cpp', 532 )]
07/18/22 08:17:56.248275 zmc_m1[34394].DB1-zm_monitor.cpp/2122 [Done Blending]
07/18/22 08:17:56.248311 zmc_m1[34394].DB3-zm_packetqueue.cpp/204 [stream index 0 ?= video_stream_id 0, keyframe 0, keep_keyframes 1,  counts 35 > pre_event_count 25 at begin 1]
07/18/22 08:17:56.248320 zmc_m1[34394].DB2-zm_packetqueue.cpp/497 [Incrementing 0x559ab625ffe0, queue size 70, end? 0]
07/18/22 08:17:56.248327 zmc_m1[34394].DB2-zm_packetqueue.cpp/504 [Incrementing 0x559ab625ffe0, 0x559ab625ffe0 still not at end, so returning true]
07/18/22 08:17:56.248335 zmc_m1[34394].DB4-zm_packetqueue.cpp/403 [Locking in get_packet using it 0x559ab625ffe0 queue end? 0]
07/18/22 08:17:56.248342 zmc_m1[34394].DB4-zm_packetqueue.cpp/409 [Have Lock in get_packet]
07/18/22 08:17:56.248348 zmc_m1[34394].DB3-zm_packetqueue.cpp/422 [get_packet using it 0x559ab625ffe0 locking index -1]
07/18/22 08:17:56.248355 zmc_m1[34394].DB4-zm_packet.h/106 [TryLocking packet -1]
07/18/22 08:17:56.248361 zmc_m1[34394].DB2-zm_packetqueue.cpp/427 [Locked packet -1, unlocking packetqueue mutex]
07/18/22 08:17:56.248368 zmc_m1[34394].DB3-zm_monitor.cpp/1724 [Motion detection is enabled signal(1) signal_change(0) trigger state(Cancel) image index -1]
07/18/22 08:17:56.248374 zmc_m1[34394].DB3-zm_monitor.cpp/1730 [Have event lock]
07/18/22 08:17:56.248380 zmc_m1[34394].DB4-zm_monitor.cpp/1734 [Trigger not OFF state is (0)]
07/18/22 08:17:56.248386 zmc_m1[34394].DB3-zm_packetqueue.cpp/204 [stream index 1 ?= video_stream_id 0, keyframe 0, keep_keyframes 1,  counts 35 > pre_event_count 25 at begin 1]
07/18/22 08:17:56.248392 zmc_m1[34394].DB2-zm_packetqueue.cpp/497 [Incrementing 0x559ab625ffe0, queue size 70, end? 0]
07/18/22 08:17:56.248398 zmc_m1[34394].DB2-zm_packetqueue.cpp/504 [Incrementing 0x559ab625ffe0, 0x559ab625ffe0 still not at end, so returning true]
07/18/22 08:17:56.248405 zmc_m1[34394].DB4-zm_packetqueue.cpp/403 [Locking in get_packet using it 0x559ab625ffe0 queue end? 0]
07/18/22 08:17:56.248411 zmc_m1[34394].DB4-zm_packetqueue.cpp/409 [Have Lock in get_packet]
07/18/22 08:17:56.248417 zmc_m1[34394].DB3-zm_packetqueue.cpp/422 [get_packet using it 0x559ab625ffe0 locking index -1]
07/18/22 08:17:56.248423 zmc_m1[34394].DB4-zm_packet.h/106 [TryLocking packet -1]
07/18/22 08:17:56.248429 zmc_m1[34394].DB2-zm_packetqueue.cpp/427 [Locked packet -1, unlocking packetqueue mutex]
07/18/22 08:17:56.248435 zmc_m1[34394].DB3-zm_monitor.cpp/1724 [Motion detection is enabled signal(1) signal_change(0) trigger state(Cancel) image index -1]
07/18/22 08:17:56.248441 zmc_m1[34394].DB3-zm_monitor.cpp/1730 [Have event lock]
07/18/22 08:17:56.248447 zmc_m1[34394].DB4-zm_monitor.cpp/1734 [Trigger not OFF state is (0)]
07/18/22 08:17:56.248453 zmc_m1[34394].DB3-zm_packetqueue.cpp/204 [stream index 1 ?= video_stream_id 0, keyframe 0, keep_keyframes 1,  counts 35 > pre_event_count 25 at begin 1]
07/18/22 08:17:56.248459 zmc_m1[34394].DB2-zm_packetqueue.cpp/497 [Incrementing 0x559ab625ffe0, queue size 70, end? 0]
07/18/22 08:17:56.248465 zmc_m1[34394].DB2-zm_packetqueue.cpp/507 [At end]
07/18/22 08:17:56.248471 zmc_m1[34394].DB4-zm_packetqueue.cpp/403 [Locking in get_packet using it 0x559ab625ffe0 queue end? 1]
07/18/22 08:17:56.248477 zmc_m1[34394].DB4-zm_packetqueue.cpp/409 [Have Lock in get_packet]
07/18/22 08:17:56.248483 zmc_m1[34394].DB2-zm_packetqueue.cpp/412 [waiting.  Queue size 70 it == end? 1]
07/18/22 08:17:56.250033 zmc_m1[34349].DB2-zm_ffmpeg_camera.cpp/238 [ffmpeg_camera in: pts: 237454020 * 1/90000=2638.378000, dts: 237454020, size: 6358, stream_index: 0, video flags: 0000, keyframe(0) pos: -1, duration: 3600]
07/18/22 08:17:56.250065 zmc_m1[34349].DB4-zm_monitor.cpp/2388 [Back from capture result=1 image count 65960]
07/18/22 08:17:56.250074 zmc_m1[34349].DB2-zm_monitor.cpp/2416 [Have packet stream_index:0 ?= videostream_id: 0 q.vpktcount 35 event? 1 image_count 65960]
07/18/22 08:17:56.250081 zmc_m1[34349].DB2-zm_packetqueue.cpp/92 [packet counts for 0 is 36]
[... cut ...]
07/18/22 08:17:56.363567 zmc_m1[34350].DB1-zm_db.cpp/190 [Success running sql query INSERT INTO `Logs` ( `TimeKey`, `Component`, `ServerId`, `Pid`, `Level`, `Code`, `Message`, `File`, `Line` ) VALUES ( 1658125076.264096, 'zmc_m1', 0, 79049, 0, 'INF', 'Backtrace complete, please execute the following command for more information: addr2line -e /usr/bin/zmc 0x559ab4a833d1 0x7f4e6c1f4420 0x559ab4b0d90d 0x559ab4aba078 0x559ab4abb8bc 0x7f4e69de6de4 0x7f4e6c1e8609 0x7f4e69ad3133', 'zm_signal.cpp', 108 )]```

apalacheno, 2022-07-19:
Sorry for the accidental closing of this report. Issue is still open.

dougnazar, 2022-07-26:
Can you recreate the crash with 1.36.21 and send the updated backtrace. Thanks.

apalacheno, 2022-07-28:
Sure. Upgraded to 1.36.21. As soon as an alarm is triggered, zm_signal crashes. Here's the log, this time more precise (I hope).

From this single event only there are 13 MB of logs, so I post only the events immediately preceding and following the crash. Let me know if you need a larger timespan of logs.

ZM is happily capturing the video feed until I manually trigger an alarm state at **13:26:58.796634**. Immediately afterwards ZM crashes at **13:26:59.090684**.

```
07/28/22 13:26:50.180547 zmc_m1[2463].INF-zm_monitor.cpp/1680 [Hauseingang: 3730 - Capturing at 25.06 fps, capturing bandwidth 1077023bytes/sec Analysing at 25.06 fps]
07/28/22 13:26:51.380726 zmc_m1[2463].INF-zm_monitor.cpp/1680 [Hauseingang: 3760 - Capturing at 25.00 fps, capturing bandwidth 541126bytes/sec Analysing at 25.00 fps]
07/28/22 13:26:52.580415 zmc_m1[2463].INF-zm_monitor.cpp/1680 [Hauseingang: 3790 - Capturing at 25.01 fps, capturing bandwidth 541138bytes/sec Analysing at 24.17 fps]
07/28/22 13:26:53.780502 zmc_m1[2463].INF-zm_monitor.cpp/1680 [Hauseingang: 3820 - Capturing at 25.00 fps, capturing bandwidth 538812bytes/sec Analysing at 25.00 fps]
07/28/22 13:26:54.983871 zmc_m1[2463].INF-zm_monitor.cpp/1680 [Hauseingang: 3850 - Capturing at 24.93 fps, capturing bandwidth 539222bytes/sec Analysing at 25.76 fps]
07/28/22 13:26:56.180187 zmc_m1[2463].INF-zm_monitor.cpp/1680 [Hauseingang: 3880 - Capturing at 25.08 fps, capturing bandwidth 1058248bytes/sec Analysing at 25.08 fps]
07/28/22 13:26:57.380482 zmc_m1[2463].INF-zm_monitor.cpp/1680 [Hauseingang: 3910 - Capturing at 24.99 fps, capturing bandwidth 539789bytes/sec Analysing at 24.99 fps]
07/28/22 13:26:58.580320 zmc_m1[2463].INF-zm_monitor.cpp/1680 [Hauseingang: 3940 - Capturing at 25.00 fps, capturing bandwidth 540574bytes/sec Analysing at 25.00 fps]
07/28/22 13:26:58.796634 zmc_m1[2688].INF-zm_monitor.cpp/1980 [Hauseingang: 3945 - Gone into alarm state PreAlarmCount: 0 > AlarmFrameCount:1 Cause:Forced Web]
07/28/22 13:26:58.950882 zmc_m1[2688].INF-zm_monitor.cpp/1987 [Hauseingang: 3944 - Opening new event 192, alarm start]
07/28/22 13:26:59.056696 zmc_m1[4959].INF-zm_videostore.cpp/532 [some options not used, turn on debugging for a list.]
07/28/22 13:26:59.090684 zmc_m1[4959].ERR-zm_signal.cpp/50 [Got signal 11 (Segmentation fault), crashing]
07/28/22 13:26:59.090735 zmc_m1[4959].ERR-zm_signal.cpp/80 [Signal address is (nil), from 0x55a08f5bc88d]
07/28/22 13:26:59.092376 zmc_m1[4959].ERR-zm_signal.cpp/101 [Backtrace 0: /usr/bin/zmc(+0xc4321) [0x55a08f532321]]
07/28/22 13:26:59.092404 zmc_m1[4959].ERR-zm_signal.cpp/101 [Backtrace 1: /lib/x86_64-linux-gnu/libpthread.so.0(+0x14420) [0x7fa0b7e61420]]
07/28/22 13:26:59.092412 zmc_m1[4959].ERR-zm_signal.cpp/101 [Backtrace 2: /usr/bin/zmc(+0x14e88d) [0x55a08f5bc88d]]
07/28/22 13:26:59.092417 zmc_m1[4959].ERR-zm_signal.cpp/101 [Backtrace 3: /usr/bin/zmc(+0xfafc8) [0x55a08f568fc8]]
07/28/22 13:26:59.092422 zmc_m1[4959].ERR-zm_signal.cpp/101 [Backtrace 4: /usr/bin/zmc(+0xfc80c) [0x55a08f56a80c]]
07/28/22 13:26:59.092427 zmc_m1[4959].ERR-zm_signal.cpp/101 [Backtrace 5: /lib/x86_64-linux-gnu/libstdc++.so.6(+0xd6de4) [0x7fa0b5a53de4]]
07/28/22 13:26:59.092433 zmc_m1[4959].ERR-zm_signal.cpp/101 [Backtrace 6: /lib/x86_64-linux-gnu/libpthread.so.0(+0x8609) [0x7fa0b7e55609]]
07/28/22 13:26:59.092438 zmc_m1[4959].ERR-zm_signal.cpp/101 [Backtrace 7: /lib/x86_64-linux-gnu/libc.so.6(clone+0x43) [0x7fa0b5740133]]
07/28/22 13:26:59.092443 zmc_m1[4959].INF-zm_signal.cpp/108 [Backtrace complete, please execute the following command for more information: addr2line -e /usr/bin/zmc 0x55a08f532321 0x7fa0b7e61420 0x55a08f5bc88d 0x55a08f568fc8 0x55a08f56a80c 0x7fa0b5a53de4 0x7fa0b7e55609 0x7fa0b5740133]
07/28/22 13:26:59.100255 zmc_m1[2463].INF-zm_monitor.cpp/3060 [Hauseingang: image_count:3953 - Closing event 192, shutting down]
07/28/22 13:27:01.002176 zmc_m1[5053].DB1-zm_logger.cpp/224 [LogOpts: level=DB4 effective=DB4, screen=OFF, database=INF, logfile=DB9->/var/log/zm/zmc_m1.log, syslog=WAR]
07/28/22 13:27:01.002238 zmc_m1[5053].DB1-zm_utils.cpp/198 [Detected a x86\x86-64 processor with AVX2]
07/28/22 13:27:01.002571 zmc_m1[5053].DB4-zm_image.cpp/396 [Blend: Using SSE2 fast blend function]
07/28/22 13:27:01.002590 zmc_m1[5053].DB4-zm_image.cpp/458 [Delta: Using SSSE3 delta functions]
07/28/22 13:27:01.002598 zmc_m1[5053].DB4-zm_image.cpp/559 [Deinterlace: Using standard functions]
07/28/22 13:27:01.002604 zmc_m1[5053].DB4-zm_image.cpp/572 [Image buffer copy: Using standard memcpy]
07/28/22 13:27:01.002775 zmc_m1[5053].DB2-zm_storage.cpp/62 [Loading Storage for 2 using SELECT `Id`, `Name`, `Path`, `Type`, `Scheme` FROM `Storage` WHERE `Id`=2]
07/28/22 13:27:01.002970 zmc_m1[5053].DB1-zm_storage.cpp/80 [Loaded Storage area 2 'Video (local storage)']
07/28/22 13:27:01.002994 zmc_m1[5053].DB1-zm_monitor.cpp/493 [Have camera type Ffmpeg]
07/28/22 13:27:01.003005 zmc_m1[5053].DB1-zm_monitor.cpp/2213 [Reloading linked monitors for monitor Hauseingang, '(null)']
07/28/22 13:27:01.003014 zmc_m1[5053].DB3-zm_monitor.cpp/570 [Decoding enabled: 1 function 3 Modect savejpegs 0 videowriter 2]
07/28/22 13:27:01.003023 zmc_m1[5053].DB1-zm_packetqueue.cpp/681 [Setting pre_event_video_packet_count to 25]
07/28/22 13:27:01.003030 zmc_m1[5053].DB1-zm_packetqueue.cpp/675 [Setting max_video_packet_count to 0]
[... cut ...]
07/28/22 13:27:06.502357 zmc_m1[5053].DB4-zm_monitor.cpp/2388 [Back from capture result=1 image count 0]
07/28/22 13:27:06.502393 zmc_m1[5053].DB2-zm_monitor.cpp/2416 [Have packet stream_index:1 ?= videostream_id: 0 q.vpktcount 0 event? 0 image_count 0]
07/28/22 13:27:06.502430 zmc_m1[5053].DB4-zm_monitor.cpp/2446 [Not Queueing audio packet]
07/28/22 13:27:06.502570 zmc_m1[5053].DB4-zm_monitor.cpp/1670 [FPS: capture count 0 - last capture count 0 = 0 now:1659007626.502570, last 1659007621.167839, elapsed 5.334731 = capture: 0.000000 fps analysis: 0.000000 fps]
```

connortechnology, 2022-08-05:
We think this is fixed in 1.36.22.

apalacheno, 2022-08-23:
I can confirm this to be fixed in 1.36.24. Thank you very much for looking into this!

PR #3551 (2022-07-28): Fix zmc Segfault on 1.36.20
Wasn't able to reproduce but based on the stacktrace and logs this looks likely.

1.36.20 added packet reordering and attempts to get the stream_index without checking if there actually is an audio_out_stream.

Fixes: #3533
closed 2022-08-23 by apalacheno