ZoneMinder/zoneminder #3555: Duration time on a web table is increased, but duration of records right.

spanchy, 2022-08-01:
**Describe Your Environment**
- Version of ZoneMinder [1.36.21] - Latest on debian repo.
- How you installed ZoneMinder [repository]
- Debian 11.4
- Chrome latest.
- 
**Describe the bug**
The time of events increases by the previous one in each row. All records are 10 min, but duration in row below is increased on a previous durations.

[image]

amimbitassu, 2022-08-04:
Same issue on a fresh install Ubuntu 20.04 Zoneminder 1.36.21

[image]

connortechnology, 2022-08-05:
Fixed by 22331e02cdb4a0813f86a521d0c1b3fcb7e3d6f4.  Included in 1.36.22.

closed 2022-08-05 by connortechnology