ZoneMinder/zoneminder #3564: 1.36.22 compilation error

mattmatician, 2022-08-05:
https://github.com/ZoneMinder/zoneminder/commit/c720af7a6483fbfa9a0628c55ad6808e471a3e9a

`int keyframe_interval = 1;`

appears twice in src/zm_packetqueue.cpp.

Which results in this compilation error:

`#0 87.84 [ 47%] Building CXX object src/CMakeFiles/zm.dir/zm_packetqueue.cpp.o
#0 88.58 /tmp/zoneminder/src/zm_packetqueue.cpp: In member function ‘void PacketQueue::clearPackets(const std::shared_ptr<ZMPacket>&)’:
#0 88.58 /tmp/zoneminder/src/zm_packetqueue.cpp:273:7: error: redeclaration of ‘int keyframe_interval’
#0 88.58   273 |   int keyframe_interval = 1;
#0 88.58       |       ^~~~~~~~~~~~~~~~~
#0 88.58 /tmp/zoneminder/src/zm_packetqueue.cpp:265:7: note: ‘int keyframe_interval’ previously declared here
#0 88.58   265 |   int keyframe_interval = 1;
#0 88.58       |       ^~~~~~~~~~~~~~~~~
#0 88.73 make[2]: *** [src/CMakeFiles/zm.dir/build.make:538: src/CMakeFiles/zm.dir/zm_packetqueue.cpp.o] Error 1
#0 88.73 make[1]: *** [CMakeFiles/Makefile2:484: src/CMakeFiles/zm.dir/all] Error 2
#0 88.73 make: *** [Makefile:136: all] Error 2
`

connortechnology, 2022-08-05:
Fixed.  Botched the release.  1.36.23 is now building.

closed 2022-08-05 by connortechnology

SmokeyBR, 2022-08-06:
problems after updating also looks like same problem after building it
https://forums.zoneminder.com/viewtopic.php?p=127461#p127461