ZoneMinder/zoneminder #3567: v1.36.23 not usable "not running"

sagitt, 2022-08-06:
**Describe Your Environment**
- Version of ZoneMinder: v1.36.23
- How you installed ZoneMinder: PPA
- Full name and version of OS: Ubuntu Server 18.04.6 LTS (On Proxmox Server)
- Browser name and version: Google Chrome

**Describe the bug**
After upgrate to v1.36.23, ZM is not usable. Cameras disconnect evry 20 seconds and the system show "not running", cameras go back and again not running. 

With a snapshot of the previous installed version v1.36.21, all works fine.

SmokeyBR, 2022-08-06:
i can confrim the same also being discussed here https://forums.zoneminder.com/viewtopic.php?t=31958&sid=a9d73302df4623afe392acf410eb1839

sagitt, 2022-08-06:
> i can confrim the same also being discussed here https://forums.zoneminder.com/viewtopic.php?t=31958&sid=a9d73302df4623afe392acf410eb1839

Probably same problem.
Waiting a fix :(

CyberGuerro, 2022-08-06:
Same here on Gentoo....    ....zoneminder starts, but ZMC monitor's processes start and stop repeatedly

SteveGilvarry, 2022-08-06:
Revert to 1.36.21, .24 being worked on to resolve the issue.

seaweeduk, 2022-08-06:
Same issue here, is there a way to revert the package under ubuntu using the repository? All my cameras are restarting every few seconds

MacPro-de, 2022-08-06:
It would be great, if we could install an older version via Ubuntu Repository like this: apt install zoneminder=1.36.22
This does not seem to work. Perhaps it would be an option to revert to 1.36.22 in the repository or simply make 1.36.22 to 1.36.24, so that we can use ZoneMinder again.

Gransi, 2022-08-06:
Same issue here, after the update i get many error message
```
06.08.2022 11:42:28.799415 zmtrigger[2256].ERR [ZoneMinder::Memory:272] [Shared data size conflict in shared_data for monitor CAM1, expected 768, got 760]
06.08.2022 11:42:28.805549 zmtrigger[2256].ERR [ZoneMinder::Memory:272] [Shared data size conflict in shared_data for monitor CAM2, expected 768, got 760]
06.08.2022 11:42:28.813068 zmtrigger[2256].ERR [ZoneMinder::Memory:272] [Shared data size conflict in shared_data for monitor CAM3, expected 768, got 760]
06.08.2022 11:42:28.823953 zmtrigger[2256].ERR [ZoneMinder::Memory:272] [Shared data size conflict in shared_data for monitor CAM4, expected 768, got 760]
06.08.2022 11:42:28.830216 zmtrigger[2256].ERR [ZoneMinder::Memory:272] [Shared data size conflict in shared_data for monitor CAM5, expected 768, got 760]
06.08.2022 11:42:28.871963 zmtrigger[2256].ERR [ZoneMinder::Memory:272] [Shared data size conflict in shared_data for monitor CAM6, expected 768, got 760]
```

seaweeduk, 2022-08-06:
> It would be great, if we could install an older version via Ubuntu Repository like this: apt install zoneminder=1.36.22 This does not seem to work. Perhaps it would be an option to revert to 1.36.22 in the repository or simply make 1.36.22 to 1.36.24, so that we can use ZoneMinder again.

Someone on the forum kindly shared how to download .deb files and downgrade to 1.36.21 here

https://forums.zoneminder.com/viewtopic.php?p=127476#p127476

I'm back to 1.36.21 and everything is running normally again.

sagitt, 2022-08-06:
very well, so the problem is not only for ubuntu server.
Probably the problem is in both .22 and .23.
We wait .24 and still on .21 for now.

thanks all developers for the hard work

SmokeyBR, 2022-08-06:
seems all good after .24

sagitt, 2022-08-06:
> seems all good after .24

Same here

closed 2022-08-06 by sagitt