ZoneMinder/zoneminder #3568: Uncaught Error: Undefined constant "ZM_OPT_USE_API" in AppController.php:62

KeithSBB, 2022-08-06:
**Describe Your Environment**
Fedora 36 sever with nginx and php 8.1.8

Was working great until
after this mornings upgrade to:
 zoneminder-common-1.36.21-1.fc36.x86_64
zoneminder-nginx-1.36.21-1.fc36.x86_64

in /var/log/zoneminder/error.log:
```
Error: Fatal Error (1): Uncaught Error: Undefined constant "ZM_OPT_USE_API" in /usr/share/zoneminder/www/api/app/Controller/AppController.php:62
```

**To Reproduce**
start zoneminder

**Expected behavior**
Note: this particular error idn't use to be fatal, but is now under php 8.1.8

```

KeithSBB, 2022-08-07:
Upon further investigation I found this error occurring before the Undefined constant "ZM_OPT_USE_API" error:
```
2022-08-07 13:09:42 Error: [Error] Class "IntlDateFormatter" not found
```
This is probably causing the routines that load in the ZM_OPT_USE_API to abort.  In Fedora 36 I was able to fix everything by:
```
sudo dnf install php-intl.x86_64
```
No further action was required (I did not have to set extension=php_intl.dll in php.ini)

closed 2022-08-07 by KeithSBB

connortechnology, 2022-08-08:
This is weird as php-intl was added as a requirement in 1.36.20.  It should have been automatically installed. Thoughts @knight-of-ni ?

knight-of-ni, 2022-08-08:
This has already been fixed in https://github.com/rpmfusion/zoneminder/commit/3d4cde7180cf689456f6ada9cd52005386576b83 and new packages have been built.

It can take up to 2 weeks for these packages to reach RPMFusion stable repo.