ZoneMinder/zoneminder #3569: Latest update of Zoneminder web interface doesn't come up.

CaverJ, 2022-08-08:
I am using Rocky Linux 8.6 (RHEL 8.6) using Apache web server. The version of Zoneminder is the one after V1.36.19, as I am unable to any interface to it. To bring up the web interface, I have tried Firefox and Chrome.

The web interface doesn't come up. I tried a fresh OS install and installed Zoneminder again, but it didn't show up on the web browser. The httpd error log file is as follows:

[Sun Aug 07 03:42:01.397683 2022] [lbmethod_heartbeat:notice] [pid 2008:tid 140515384973632] AH02282: No slotmem from mod_heartmonitor
[Sun Aug 07 03:42:01.398248 2022] [mpm_event:notice] [pid 2008:tid 140515384973632] AH00489: Apache/2.4.37 (rocky) OpenSSL/1.1.1k configured -- resuming normal operations
[Sun Aug 07 03:42:01.398255 2022] [core:notice] [pid 2008:tid 140515384973632] AH00094: Command line: '/usr/sbin/httpd -D FOREGROUND'
[Sun Aug 07 07:32:27.488457 2022] [mpm_event:notice] [pid 2008:tid 140515384973632] AH00492: caught SIGWINCH, shutting down gracefully
[Sun Aug 07 07:32:28.586737 2022] [suexec:notice] [pid 1033553:tid 140058738153792] AH01232: suEXEC mechanism enabled (wrapper: /usr/sbin/suexec)
AH00558: httpd: Could not reliably determine the server's fully qualified domain name, using localhost.localdomain. Set the 'ServerName' directive globally to suppress this message
[Sun Aug 07 07:32:28.594972 2022] [lbmethod_heartbeat:notice] [pid 1033553:tid 140058738153792] AH02282: No slotmem from mod_heartmonitor
[Sun Aug 07 07:32:28.596834 2022] [mpm_event:notice] [pid 1033553:tid 140058738153792] AH00489: Apache/2.4.37 (rocky) OpenSSL/1.1.1k configured -- resuming normal operations
[Sun Aug 07 07:32:28.596845 2022] [core:notice] [pid 1033553:tid 140058738153792] AH00094: Command line: '/usr/sbin/httpd -D FOREGROUND'
[Sun Aug 07 08:09:58.611922 2022] [mpm_event:notice] [pid 1033553:tid 140058738153792] AH00492: caught SIGWINCH, shutting down gracefully
[Sun Aug 07 08:11:08.191828 2022] [suexec:notice] [pid 2007:tid 140606360045888] AH01232: suEXEC mechanism enabled (wrapper: /usr/sbin/suexec)
AH00558: httpd: Could not reliably determine the server's fully qualified domain name, using localhost.localdomain. Set the 'ServerName' directive globally to suppress this message
[Sun Aug 07 08:11:08.393476 2022] [lbmethod_heartbeat:notice] [pid 2007:tid 140606360045888] AH02282: No slotmem from mod_heartmonitor
[Sun Aug 07 08:11:08.394734 2022] [mpm_event:notice] [pid 2007:tid 140606360045888] AH00489: Apache/2.4.37 (rocky) OpenSSL/1.1.1k configured -- resuming normal operations
[Sun Aug 07 08:11:08.394743 2022] [core:notice] [pid 2007:tid 140606360045888] AH00094: Command line: '/usr/sbin/httpd -D FOREGROUND'

connortechnology, 2022-08-08:
We have seen issues where the dependency php-intl wasn't automatically installed.

So try 
sudo dnf install php-intl
sudo systemctl reload apache2

CaverJ, 2022-08-08:
That did it. Thanks

closed 2022-08-08 by CaverJ