ZoneMinder/zoneminder #3570: Frame Skip is ignored in RECORD mode

locutus13, 2022-08-09:
**Describe Your Environment**
- Version of ZoneMinder: 1.36.24
- How you installed ZoneMinder: PPA
- Ubuntu 20.04 LTS

- ZoneMinder Source Type: H264 stream @ 2 FPS

**Describe the bug**
Any value I could think of in the field "Frame Skip" is completely ignored when saving event images. There are always 1200 images for 10 minutes of 2 FPS video. 

**To Reproduce**
Steps to reproduce the behavior:
1. Connect to camera with H264 stream @ 2 FPS and set Zoneminder to RECORD mode
2. Selecting or deselecting  "Analysis Enabled" makes no difference, Decoding remains active
3. In the STORAGE tab select under "Save JPEGs" "Frames only" - disable the Video writer
4. Under MISC: Section Length to 600seconds and enter any other value than 0 into the "Frame Skip" field and/or "Motion Frame Skip"
5. Click SAVE
6. Watch how every event creates 1200 images.

**Expected behavior**
Skipped frames should not be recorded or saved into JPG files. 

**Debug Logs**
no errors or warnings have been recorded in the log

connortechnology, 2022-08-09:
frame_skip appears to no longer be implemented.

motion_frame_skip just affects doing motion analysis.

singhharsh1708, 2026-08-30:
Independently confirmed against master (aa30a5a), and one thing that may help the reimplement-vs-remove call: the removal surface is larger than it looks.

`FrameSkip` currently lives in:

- `src/zm_monitor.h:628` (declaration), `zm_monitor.cpp:248` (init), `zm_monitor.cpp:664` (DB load) — the three write-only sites you identified; nothing reads it
- `web/skins/classic/views/monitor.php:1594` — the form control
- `web/skins/classic/views/js/monitor.js.php:165` — a client-side validator that rejects bad input for a field that does nothing, via the `BadFrameSkip` message
- `web/includes/Monitor.php:326` — model default
- `db/zm_create.sql.in:742` — `smallint(5) unsigned NOT NULL default '0'`
- **25 language files**, each carrying both `FrameSkip` and `BadFrameSkip`

So a clean removal is a schema migration plus 25 translation files, and it drops two translated strings that translators have already paid for. That is not an argument against removing it — just that "low-risk" is cheaper in code than in churn.

I'd be glad to do either, whichever you decide:

- **Remove**: migration to drop the column, strip the form control, the validator and the model default, and remove the two keys from the lang files. I'd leave the C++ member removal in the same PR since it is three lines.
- **Reimplement**: scoped as you describe, to the JPEG/Frames save path only, leaving passthrough untouched. I would want your steer on whether the skip counter belongs alongside `motion_frame_skip` in the analysis loop or at the point frames are written.

One caveat worth stating plainly: I don't have a ZoneMinder deployment or cameras, so I can verify a removal by inspection and CI but I could not properly exercise a reimplementation against real streams. That probably argues for me taking the removal if that's the direction, and leaving the reimplementation to someone who can test it.

connortechnology, 2026-09-01:
Verified all of this against master. Both of you are right that it's dead, and there are two references neither list caught.

## Confirmed

`frame_skip` is write-only — exactly three C++ sites, all writes:

```
src/zm_monitor.h:628    int frame_skip;   // How many frames to skip in continuous modes
src/zm_monitor.cpp:248  frame_skip(0),
src/zm_monitor.cpp:664  frame_skip = atoi(dbrow[col]);
```

(@SteveGilvarry's `zm_monitor.h:593` is stale; @singhharsh1708's `:628` is the current line.)

## The history, which I think settles the decision

```
2003-10-08  e498c9118  "Added frame skip for continuous modes"   - implemented
2017-12-01  2e3fda488  read commented out in Analyse()
2021-02-03  cb60b6c14  commented-out remnant deleted
```

FrameSkip has had no effect since December 2017 — nearly nine years — and in that time nobody has filed "FrameSkip doesn't work" apart from this issue. The form has been advertising it that entire time. That is about as clear a demand signal as we are going to get.

## Two references missed above

- **`scripts/ZoneMinder/lib/ZoneMinder/Monitor.pm:135` and `:262`** — `FrameSkip` is in `%fields` and `%defaults`. `%fields` builds the INSERT/UPDATE column lists in `Object.pm:214-284`, so dropping the column without editing this breaks every Perl-side monitor save. The Perl model isn't in either list above.
- **`web/ajax/status.php:87`** — `'FrameSkip' => true` in the monitor-field allow-list for the status endpoint.

Also worth noting: `db/zm_update-0.9.15.sql:11` is history and must not be touched, and the CakePHP API is schema-driven, so dropping the column silently changes monitor payloads for any consumer reading `Monitor.FrameSkip`. That deserves a release note. There are no docs or `web/api/` references.

## Decision: remove

Reimplementation is hard to justify. @SteveGilvarry's passthrough point is the decisive half — you cannot drop frames from an H264 GOP without re-encoding, so it could only ever apply to JPEG/Frames storage, which is a shrinking minority of deployments. The adjacent controls already cover the motivating cases: MotionFrameSkip and AnalysisFPSLimit for CPU, the video writer for disk.

@singhharsh1708 — yes please, and thank you for the offer. But I would like it split into two PRs rather than one:

**PR 1 — stop advertising it.** Remove the form control (`monitor.php:1595-1596`), the JS validator (`monitor.js.php:165-166`), and the three C++ lines. No schema change, no lang changes, trivially revertable. This is what actually resolves the complaint in this issue, and I can merge it quickly.

**PR 2 — retire the column.** Migration to drop it, plus `Monitor.php:326`, `Monitor.pm:135,262`, `status.php:87`, and the lang keys. Dropping a column is irreversible for anyone who downgrades, and it is the piece that needs the API note. We do drop columns — 15 migrations already do — it just does not need to ride along with the UI fix.

One push-back on the framing: removing two keys from 25 lang files does not waste translator effort. A control that silently does nothing costs users more than two retired strings cost translators. The 25-file churn is mechanical and should not weigh on the decision — the sequencing risk above is the real reason to stage it.

And your read on scope is right: a removal is verifiable by inspection and CI, which is exactly the kind of change that does not need a camera to validate.