ZoneMinder/zoneminder #3599: Warning Bug for Mocord set event time

amimbitassu, 2022-09-23:
**Describe Your Environment**
- Zoneminder 1.36.26
- PPA Install
- Ubuntu 22.04.5
- Edge Version? (Up to date)

- Hikvision
- 8fps
- 1920x1080
- RTSP TCP source

**Describe the bug**
All cameras 600seconds event times and options are set to finish events once time is reached.

9/23/22, 10:39:59 AM GMT+3	zmc_m5		33204	WAR	Laboratuvar: 1055770 - event 12842, has exceeded desired section length. 1663918799 - 1663918199 = 600 >= 600	zm_monitor.cpp	2092

For record cameras, there is no issue, for mocord cameras, 600 seconds of events are created but because of the >= comparison, this warning is displayed in the logs.

**To Reproduce**
Add rtsp camera.
Set 600seconds of event time.
Set config>EVENT CLOSE MODE > time
mocord cameras throw these errors.

**Expected behavior**
600 seconds of event time is normal. There should be no warning.

amimbitassu, 2022-09-23:
I may be wrong, event times are off by 1-2seconds. I think the timing is off for mocord. Maybe because of the after alarm frames are added and the event cant be closed at correct time?
[image]
[image]

connortechnology, 2022-10-11:
This was fixed in 1.36.27.

commit 0ce8373b (2022-09-25): Only warn about event exceeding section_length if we are not using close_mode=TIME. Fixes #3599
closed 2022-10-11 by connortechnology