ZoneMinder/zoneminder #3605: Unable to add new storage

Matt-M-ironnet, 2022-10-07:
**Describe Your Environment**
- Version of ZoneMinder [release version, development version, or commit]
1.36.27
- How you installed ZoneMinder [e.g. PPA, RPMFusion, from-source, etc]
PPA
- Full name and version of OS
Ubuntu Server 22.04.1
- Browser name and version (if this is an issue with the web interface)
Chrome

**Describe the bug**
As of 1.36.27, I am unable to add any new storage. I have a version of 1.36.26 within a VM that I am still able to do so. Upon attempting to add a new location, I am sent to a blank white page. Refresh asks if I want to resubmit the form. 

**To Reproduce**
Steps to reproduce the behavior:
1. Goto Options, Storage, Add New Storage.
2. Enter any location.
3. Click Save

**Expected behavior**
The addition of a new storage location. 

**Debug Logs**
```

<insert debug logs here, please make sure they are within the ``` quotes so they are formatted properly>

```
Date/Time,Component,Server,PID,Level,Message,File,Line
"10/7/22, 2:28:17 PM CDT",zmdc,,1490,ERR,"'zmwatch.pl' exited abnormally, exit status 9",zmdc.pl,'-
"10/7/22, 2:28:17 PM CDT",zmdc,,2319,INF,"'zmwatch.pl' started at 22/10/07 14:28:17",zmdc.pl,'-
"10/7/22, 2:28:17 PM CDT",zmdc,,1490,INF,"'zmwatch.pl' starting at 22/10/07 14:28:17, pid = 2319",zmdc.pl,'-
"10/7/22, 2:28:17 PM CDT",zmdc,,1490,INF,"Starting pending process, zmwatch.pl",zmdc.pl,'-
"10/7/22, 2:28:15 PM CDT",zmc_m1,,1518,INF,"Living Room: 66300 - Capturing at 14.33 fps, capturing bandwidth 104003bytes/sec Analysing at 0.00 fps",zm_monitor.cpp,1680
"10/7/22, 2:28:08 PM CDT",zmc_m1,,1518,INF,"Living Room: 66200 - Capturing at 15.00 fps, capturing bandwidth 98731bytes/sec Analysing at 0.00 fps",zm_monitor.cpp,1680
"10/7/22, 2:28:01 PM CDT",zmc_m1,,1518,INF,"Living Room: 66100 - Capturing at 14.85 fps, capturing bandwidth 96862bytes/sec Analysing at 0.00 fps",zm_monitor.cpp,1680
"10/7/22, 2:27:54 PM CDT",zmc_m1,,1518,INF,"Living Room: 66000 - Capturing at 14.85 fps, capturing bandwidth 107807bytes/sec Analysing at 0.00 fps",zm_monitor.cpp,1680
"10/7/22, 2:27:47 PM CDT",zmc_m1,,1518,INF,"Living Room: 65900 - Capturing at 15.02 fps, capturing bandwidth 97032bytes/sec Analysing at 0.00 fps",zm_monitor.cpp,1680
"10/7/22, 2:27:47 PM CDT",zmdc,,1490,ERR,"'zmwatch.pl' exited abnormally, exit status 9",zmdc.pl,'-
"10/7/22, 2:27:46 PM CDT",zmdc,,1490,INF,"'zmwatch.pl' starting at 22/10/07 14:27:46, pid = 2316",zmdc.pl,'-
"10/7/22, 2:27:46 PM CDT",zmdc,,2316,INF,"'zmwatch.pl' started at 22/10/07 14:27:46",zmdc.pl,'-
"10/7/22, 2:27:46 PM CDT",zmdc,,1490,INF,"Starting pending process, zmwatch.pl",zmdc.pl,'-
"10/7/22, 2:27:41 PM CDT",zmc_m1,,1518,INF,"Living Room: 65800 - Capturing at 14.84 fps, capturing bandwidth 96790bytes/sec Analysing at 0.00 fps",zm_monitor.cpp,1680
"10/7/22, 2:27:34 PM CDT",zmc_m1,,1518,INF,"Living Room: 65700 - Capturing at 14.86 fps, capturing bandwidth 107510bytes/sec Analysing at 0.00 fps",zm_monitor.cpp,1680
"10/7/22, 2:27:27 PM CDT",zmc_m1,,1518,INF,"Living Room: 65600 - Capturing at 15.24 fps, capturing bandwidth 100722bytes/sec Analysing at 0.00 fps",zm_monitor.cpp,1680
"10/7/22, 2:27:21 PM CDT",zmc_m1,,1518,INF,"Living Room: 65500 - Capturing at 14.76 fps, capturing bandwidth 95395bytes/sec Analysing at 0.00 fps",zm_monitor.cpp,1680
"10/7/22, 2:27:16 PM CDT",zmdc,,1490,ERR,"'zmwatch.pl' exited abnormally, exit status 9",zmdc.pl,'-
"10/7/22, 2:27:16 PM CDT",zmdc,,2313,INF,"'zmwatch.pl' started at 22/10/07 14:27:16",zmdc.pl,'-
"10/7/22, 2:27:16 PM CDT",zmdc,,1490,INF,"'zmwatch.pl' starting at 22/10/07 14:27:16, pid = 2313",zmdc.pl,'-
"10/7/22, 2:27:16 PM CDT",zmdc,,1490,INF,"Starting pending process, zmwatch.pl",zmdc.pl,'-
"10/7/22, 2:27:14 PM CDT",zmc_m1,,1518,INF,"Living Room: 65400 - Capturing at 14.98 fps, capturing bandwidth 107183bytes/sec Analysing at 0.00 fps",zm_monitor.cpp,1680
"10/7/22, 2:27:07 PM CDT",zmc_m1,,1518,INF,"Living Room: 65300 - Capturing at 15.32 fps, capturing bandwidth 99654bytes/sec Analysing at 0.00 fps",zm_monitor.cpp,1680
"10/7/22, 2:27:01 PM CDT",zmc_m1,,1518,INF,"Living Room: 65200 - Capturing at 14.61 fps, capturing bandwidth 93876bytes/sec Analysing at 0.00 fps",zm_monitor.cpp,1680
"10/7/22, 2:26:54 PM CDT",zmc_m1,,1518,INF,"Living Room: 65100 - Capturing at 15.06 fps, capturing bandwidth 108007bytes/sec Analysing at 0.00 fps",zm_monitor.cpp,1680
"10/7/22, 2:26:47 PM CDT",zmc_m1,,1518,INF,"Living Room: 65000 - Capturing at 15.06 fps, capturing bandwidth 98537bytes/sec Analysing at 0.00 fps",zm_monitor.cpp,1680
"10/7/22, 2:26:46 PM CDT",zmdc,,1490,ERR,"'zmwatch.pl' exited abnormally, exit status 9",zmdc.pl,'-
"10/7/22, 2:26:46 PM CDT",zmdc,,2310,INF,"'zmwatch.pl' started at 22/10/07 14:26:46",zmdc.pl,'-
"10/7/22, 2:26:46 PM CDT",zmdc,,1490,INF,"'zmwatch.pl' starting at 22/10/07 14:26:46, pid = 2310",zmdc.pl,'-
"10/7/22, 2:26:46 PM CDT",zmdc,,1490,INF,"Starting pending process, zmwatch.pl",zmdc.pl,'-
"10/7/22, 2:26:41 PM CDT",zmc_m1,,1518,INF,"Living Room: 64900 - Capturing at 14.76 fps, capturing bandwidth 95753bytes/sec Analysing at 0.00 fps",zm_monitor.cpp,1680
"10/7/22, 2:26:34 PM CDT",zmc_m1,,1518,INF,"Living Room: 64800 - Capturing at 14.77 fps, capturing bandwidth 106793bytes/sec Analysing at 0.00 fps",zm_monitor.cpp,1680
"10/7/22, 2:26:27 PM CDT",zmc_m1,,1518,INF,"Living Room: 64700 - Capturing at 14.95 fps, capturing bandwidth 96280bytes/sec Analysing at 0.00 fps",zm_monitor.cpp,1680
"10/7/22, 2:26:20 PM CDT",zmc_m1,,1518,INF,"Living Room: 64600 - Capturing at 15.10 fps, capturing bandwidth 97235bytes/sec Analysing at 0.00 fps",zm_monitor.cpp,1680
"10/7/22, 2:26:16 PM CDT",zmdc,,1490,ERR,"'zmwatch.pl' exited abnormally, exit status 9",zmdc.pl,'-
"10/7/22, 2:26:16 PM CDT",zmdc,,2308,INF,"'zmwatch.pl' started at 22/10/07 14:26:16",zmdc.pl,'-
"10/7/22, 2:26:16 PM CDT",zmdc,,1490,INF,"'zmwatch.pl' starting at 22/10/07 14:26:16, pid = 2308",zmdc.pl,'-
"10/7/22, 2:26:16 PM CDT",zmdc,,1490,INF,"Starting pending process, zmwatch.pl",zmdc.pl,'-
"10/7/22, 2:26:14 PM CDT",zmc_m1,,1518,INF,"Living Room: 64500 - Capturing at 14.43 fps, capturing bandwidth 104766bytes/sec Analysing at 0.00 fps",zm_monitor.cpp,1680
"10/7/22, 2:26:07 PM CDT",zmc_m1,,1518,INF,"Living Room: 64400 - Capturing at 14.98 fps, capturing bandwidth 98013bytes/sec Analysing at 0.00 fps",zm_monitor.cpp,1680
"10/7/22, 2:26:00 PM CDT",zmc_m1,,1518,INF,"Living Room: 64300 - Capturing at 14.70 fps, capturing bandwidth 105781bytes/sec Analysing at 0.00 fps",zm_monitor.cpp,1680
[... 35 lines cut ...]
"10/7/22, 2:23:46 PM CDT",zmc_m1,,1518,INF,"Living Room: 62300 - Capturing at 14.99 fps, capturing bandwidth 108122bytes/sec Analysing at 0.00 fps",zm_monitor.cpp,1680
"10/7/22, 2:23:46 PM CDT",zmdc,,1490,ERR,"'zmwatch.pl' exited abnormally, exit status 9",zmdc.pl,'-
"10/7/22, 2:23:45 PM CDT",zmdc,,2292,INF,"'zmwatch.pl' started at 22/10/07 14:23:45",zmdc.pl,'-
"10/7/22, 2:23:45 PM CDT",zmdc,,1490,INF,"'zmwatch.pl' starting at 22/10/07 14:23:45, pid = 2292",zmdc.pl,'-
"10/7/22, 2:23:45 PM CDT",zmdc,,1490,INF,"Starting pending process, zmwatch.pl",zmdc.pl,'-
"10/7/22, 2:23:39 PM CDT",zmc_m1,,1518,INF,"Living Room: 62200 - Capturing at 14.86 fps, capturing bandwidth 95345bytes/sec Analysing at 0.00 fps",zm_monitor.cpp,1680
"10/7/22, 2:23:32 PM CDT",zmc_m1,,1518,INF,"Living Room: 62100 - Capturing at 14.87 fps, capturing bandwidth 96957bytes/sec Analysing at 0.00 fps",zm_monitor.cpp,1680
"10/7/22, 2:23:26 PM CDT",zmc_m1,,1518,INF,"Living Room: 62000 - Capturing at 14.99 fps, capturing bandwidth 107030bytes/sec Analysing at 0.00 fps",zm_monitor.cpp,1680
"10/7/22, 2:23:19 PM CDT",zmc_m1,,1518,INF,"Living Room: 61900 - Capturing at 14.98 fps, capturing bandwidth 97912bytes/sec Analysing at 0.00 fps",zm_monitor.cpp,1680
"10/7/22, 2:23:15 PM CDT",zmdc,,1490,ERR,"'zmwatch.pl' exited abnormally, exit status 9",zmdc.pl,'-
"10/7/22, 2:23:15 PM CDT",zmdc,,2290,INF,"'zmwatch.pl' started at 22/10/07 14:23:15",zmdc.pl,'-
"10/7/22, 2:23:15 PM CDT",zmdc,,1490,INF,"'zmwatch.pl' starting at 22/10/07 14:23:15, pid = 2290",zmdc.pl,'-
"10/7/22, 2:23:15 PM CDT",zmdc,,1490,INF,"Starting pending process, zmwatch.pl",zmdc.pl,'-
"10/7/22, 2:23:12 PM CDT",zmc_m1,,1518,INF,"Living Room: 61800 - Capturing at 15.03 fps, capturing bandwidth 96636bytes/sec Analysing at 0.00 fps",zm_monitor.cpp,1680
"10/7/22, 2:23:06 PM CDT",zmc_m1,,1518,INF,"Living Room: 61700 - Capturing at 14.98 fps, capturing bandwidth 107316bytes/sec Analysing at 0.00 fps",zm_monitor.cpp,1680
[... cut ...]
"10/7/22, 2:22:45 PM CDT",zmdc,,1490,INF,"'zmwatch.pl' starting at 22/10/07 14:22:45, pid = 2287",zmdc.pl,'-
"10/7/22, 2:22:45 PM CDT",zmdc,,1490,INF,"Starting pending process, zmwatch.pl",zmdc.pl,'-
"10/7/22, 2:22:39 PM CDT",zmc_m1,,1518,INF,"Living Room: 61300 - Capturing at 15.02 fps, capturing bandwidth 97324bytes/sec Analysing at 0.00 fps",zm_monitor.cpp,1680
"10/7/22, 2:22:32 PM CDT",zmc_m1,,1518,INF,"Living Room: 61200 - Capturing at 14.82 fps, capturing bandwidth 96055bytes/sec Analysing at 0.00 fps",zm_monitor.cpp,1680
```

connortechnology, 2022-10-07:
Thanks for the report.  Fixes are incoming and will be released tonight as 1.36.28.

closed 2022-10-07 by connortechnology

Matt-M-3, 2022-10-07:
Awesome! Question for anyone else, what's the turn around time for it to hit Ubuntu's PPA?

connortechnology, 2022-10-07:
Packages should be building now.  Can take 15 minutes to a day though. Depending on how busy they are.

commit 3f8352c2 (2022-10-07): update storage modal to fix buttons not being in form. Also remove duplicate view field and make button action be save instead of Save. Fixes #3605
commit 371f1adb (2022-10-07): update storage modal to fix buttons not being in form. Also remove duplicate view field and make button action be save instead of Save. Fixes #3605