ZoneMinder/zoneminder #3606: Ajax error when trying to view logs

Fredrick-III, 2022-10-08:
**Describe Your Environment**
- ZoneMinder: 1.36.28
- How you installed ZoneMinder: PPA
- Ubuntu 22.04.1 LTS
- Chrome 106.0.5249.103 and Firefox 105.0.2

**Describe the bug**
Just updated to 1.36.28 and noticed the log page only displays "Loading, please wait...".
Brower console shows Ajax request failed. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the ZoneMinder log webpage.
2. See continuous "Loading, please wait..."

**Expected behavior**
Expect to see log entries populate.

**Debug Logs**
```
/var/log/apache2/error.log:

[Fri Oct 07 17:44:19.028318 2022] [php:error] [pid 13965] [client 172.20.20.3:53372] PHP Parse error:  syntax error, unexpected token "else", expecting end of file in /usr/share/zoneminder/www/ajax/log.php on line 31, referer: https://[fqdn]/zm/?view=log
```

spyjo, 2022-10-08:
same here after update to 1.36.28

`2022/10/08 09:40:51 [error] 777#777: *7097481 FastCGI sent in stderr: "PHP message: PHP Parse error:  syntax error, unexpected 'else' (T_ELSE), expecting end of file in /usr/local/share/zoneminder/www/ajax/log.php on line 31" while reading response header from upstream, client: 192.168.1.12, server: , request: "GET /zm/index.php?view=request&request=log&task=query&search=&offset=0&limit=10 HTTP/1.1", upstream: "fastcgi://unix:/var/run/php7-fpm.sock:", host: "xxxxxxx.yyyyyyy.com", referrer: "http://xxxxxxx.yyyyyyy.com/zm/index.php?view=log"
`

vmatrixx, 2022-10-08:
Affects me too.

dt3ft, 2022-10-11:
Hi,
Same here:
[Tue Oct 11 22:48:45.809547 2022] [php:error] [pid 930] [client ?.?.?.?:51111] PHP Parse error:  syntax error, unexpected token "else", expecting end of file in /usr/share/zoneminder/www/ajax/log.php on line 31, referer: http://?.ddns.net:8080/zm/index.php?view=log

Fredrick-III, 2022-10-13:
Fixed in 1.36.29

closed 2022-10-13 by Fredrick-III