ZoneMinder/zoneminder #3610: Cannot view events in Firefox

rnveach, 2022-10-21:
**Describe Your Environment**
Version: v1.36.31 
Installed by: sudo apt install -y zoneminder

OS:
````
Description:    Ubuntu 18.04.6 LTS
Release:        18.04
Codename:       bionic
````
Browser: Firefox 106.0.1

**If the issue concerns a camera**
Doesn't

**Describe the bug**
Logging into the console, clicking on Event numbers for one of the cameras, click on one of the IDs.
Instead of seeing the picture from the camera, the spot is blank. I do see a red bar on the bottom and event information on the left. None of the controls seems to do anything.

**To Reproduce**
See previous.

**Expected behavior**
I expect to be able to view the events from my camera, motion and all.

Please note, when switching the browser to Microsoft Edge, the picture displays just fine.

Microsoft Edge
Version 106.0.1370.47 (Official build) (64-bit)

**Debug Logs**
N/A

connortechnology, 2022-10-21:
Is this fixed by setting Options->High B/W->WEB_H_CAN_STREAM to yes instead of auto?

rnveach, 2022-10-21:
No. I assume I do not need to restart ZM. The top left says "High" so I assume I am viewing in high mode.

Also to note, under the controls it says: `Progress: NaN:NaN:NaNs Zoom: 0x`

connortechnology, 2022-10-21:
I'm wondering if there is a javascript error preventing required code from running. Can you right click, Inspect Element, go to console tab and see if there are errors listed there?  This all works fine for me here.

rnveach, 2022-10-22:
Yes, there are errors.

Firefox:
````
22:00:52.045 Content Security Policy: The page’s settings blocked the loading of a resource at inline (“script-src”). moz-extension:58:146
22:00:52.045 Content Security Policy: The page’s settings blocked the loading of a resource at inline (“script-src”). moz-extension:58:361
22:00:56.008 zoomrotate: Start [videojs.zoomrotate.js:1:9](http://rveach.no-ip.org:81/zm/js/videojs.zoomrotate.js)
22:00:56.008 zoomrotate: Init defaults [videojs.zoomrotate.js:5:13](http://rveach.no-ip.org:81/zm/js/videojs.zoomrotate.js)
22:00:56.008 zoomrotate: Init Extend [videojs.zoomrotate.js:10:13](http://rveach.no-ip.org:81/zm/js/videojs.zoomrotate.js)
22:00:56.008 zoomrotate: End [videojs.zoomrotate.js:72:9](http://rveach.no-ip.org:81/zm/js/videojs.zoomrotate.js)
22:00:56.013 unreachable code after return statement
[skins_classic_views_js_event-classic-1666048354.js:487:2](http://rveach.no-ip.org:81/zm/cache/skins_classic_views_js_event-classic-1666048354.js)
22:00:56.057 No cue frames for event [skins_classic_views_js_event-classic-1666048354.js:102:13](http://rveach.no-ip.org:81/zm/cache/skins_classic_views_js_event-classic-1666048354.js)
22:01:03.498
getCmdResponse stream error: socket_sendto( /run/zm/zms-426699s.sock ) failed: Connection refused [js_logger-classic-1666048354.js:93:11](http://rveach.no-ip.org:81/zm/cache/js_logger-classic-1666048354.js)
22:01:03.502 Got an error from getCmdResponse [skins_classic_views_js_event-classic-1666048354.js:252:13](http://rveach.no-ip.org:81/zm/cache/skins_classic_views_js_event-classic-1666048354.js)
22:01:03.502
Object { result: "Error", message: "socket_sendto( /run/zm/zms-426699s.sock ) failed: Connection refused" }
[skins_classic_views_js_event-classic-1666048354.js:253:13](http://rveach.no-ip.org:81/zm/cache/skins_classic_views_js_event-classic-1666048354.js)
22:01:03.524 success [skins_classic_views_js_event-classic-1666048354.js:254:13](http://rveach.no-ip.org:81/zm/cache/skins_classic_views_js_event-classic-1666048354.js)
22:01:03.854
getCmdResponse stream error: socket_sendto( /run/zm/zms-426699s.sock ) failed: Connection refused [js_logger-classic-1666048354.js:93:11](http://rveach.no-ip.org:81/zm/cache/js_logger-classic-1666048354.js)
22:01:03.857 Got an error from getCmdResponse [skins_classic_views_js_event-classic-1666048354.js:252:13](http://rveach.no-ip.org:81/zm/cache/skins_classic_views_js_event-classic-1666048354.js)
22:01:03.857
Object { result: "Error", message: "socket_sendto( /run/zm/zms-426699s.sock ) failed: Connection refused" }
[skins_classic_views_js_event-classic-1666048354.js:253:13](http://rveach.no-ip.org:81/zm/cache/skins_classic_views_js_event-classic-1666048354.js)
22:01:03.857 success [skins_classic_views_js_event-classic-1666048354.js:254:13](http://rveach.no-ip.org:81/zm/cache/skins_classic_views_js_event-classic-1666048354.js)
````

Edge:
````
videojs.zoomrotate.js:1 zoomrotate: Start
videojs.zoomrotate.js:5 zoomrotate: Init defaults
videojs.zoomrotate.js:10 zoomrotate: Init Extend
videojs.zoomrotate.js:72 zoomrotate: End
skins_classic_views_…e-1666048354.js:102 No cue frames for event
````

rnveach, 2022-10-22:
Yes, there are errors.

Firefox:
````
22:00:52.045 Content Security Policy: The page’s settings blocked the loading of a resource at inline (“script-src”). moz-extension:58:146
22:00:52.045 Content Security Policy: The page’s settings blocked the loading of a resource at inline (“script-src”). moz-extension:58:361
22:00:56.008 zoomrotate: Start [videojs.zoomrotate.js:1:9](http://myhost:81/zm/js/videojs.zoomrotate.js)
22:00:56.008 zoomrotate: Init defaults [videojs.zoomrotate.js:5:13](http://myhost:81/zm/js/videojs.zoomrotate.js)
22:00:56.008 zoomrotate: Init Extend [videojs.zoomrotate.js:10:13](http://myhost:81/zm/js/videojs.zoomrotate.js)
22:00:56.008 zoomrotate: End [videojs.zoomrotate.js:72:9](http://myhost:81/zm/js/videojs.zoomrotate.js)
22:00:56.013 unreachable code after return statement
[skins_classic_views_js_event-classic-1666048354.js:487:2](http://myhost:81/zm/cache/skins_classic_views_js_event-classic-1666048354.js)
22:00:56.057 No cue frames for event [skins_classic_views_js_event-classic-1666048354.js:102:13](http://myhost:81/zm/cache/skins_classic_views_js_event-classic-1666048354.js)
22:01:03.498
getCmdResponse stream error: socket_sendto( /run/zm/zms-426699s.sock ) failed: Connection refused [js_logger-classic-1666048354.js:93:11](http://myhost:81/zm/cache/js_logger-classic-1666048354.js)
22:01:03.502 Got an error from getCmdResponse [skins_classic_views_js_event-classic-1666048354.js:252:13](http://myhost:81/zm/cache/skins_classic_views_js_event-classic-1666048354.js)
22:01:03.502
Object { result: "Error", message: "socket_sendto( /run/zm/zms-426699s.sock ) failed: Connection refused" }
[skins_classic_views_js_event-classic-1666048354.js:253:13](http://myhost:81/zm/cache/skins_classic_views_js_event-classic-1666048354.js)
22:01:03.524 success [skins_classic_views_js_event-classic-1666048354.js:254:13](http://myhost:81/zm/cache/skins_classic_views_js_event-classic-1666048354.js)
22:01:03.854
getCmdResponse stream error: socket_sendto( /run/zm/zms-426699s.sock ) failed: Connection refused [js_logger-classic-1666048354.js:93:11](http://myhost:81/zm/cache/js_logger-classic-1666048354.js)
22:01:03.857 Got an error from getCmdResponse [skins_classic_views_js_event-classic-1666048354.js:252:13](http://myhost:81/zm/cache/skins_classic_views_js_event-classic-1666048354.js)
22:01:03.857
Object { result: "Error", message: "socket_sendto( /run/zm/zms-426699s.sock ) failed: Connection refused" }
[skins_classic_views_js_event-classic-1666048354.js:253:13](http://myhost:81/zm/cache/skins_classic_views_js_event-classic-1666048354.js)
22:01:03.857 success [skins_classic_views_js_event-classic-1666048354.js:254:13](http://myhost:81/zm/cache/skins_classic_views_js_event-classic-1666048354.js)
````

I tried running firefox in troubleshoot mode (safe mode basically) and it did not change anything.

Edge:
````
videojs.zoomrotate.js:1 zoomrotate: Start
videojs.zoomrotate.js:5 zoomrotate: Init defaults
videojs.zoomrotate.js:10 zoomrotate: Init Extend
videojs.zoomrotate.js:72 zoomrotate: End
skins_classic_views_…e-1666048354.js:102 No cue frames for event
````

myankele, 2023-04-27:
I am having the same problem: clicking on event count (or pretty much any Event filtering operation), hangs in Loading, Please Wait ... state.
ZM Version is 1.37.37.  Ubuntu Jammy

Looked at the Inspect Console:  In firefox, there is a Content Security Policy error.  I turned that to false in about:config.  Restarted Browser.  Nothing changed, still there is the same error.
In Edge:
There is no Content Security Error.  But, there is a Server error (500):

This is my URL:
http://192.168.1.222/zm/index.php?view=events&page=1&filter%5BQuery%5D%5Bterms%5D%5B0%5D%5Battr%5D=Monitor&filter%5BQuery%5D%5Bterms%5D%5B0%5D%5Bop%5D=%3D&filter%5BQuery%5D%5Bterms%5D%5B0%5D%5Bval%5D=4&filter%5BQuery%5D%5Bsort_asc%5D=1&filter%5BQuery%5D%5Bsort_field%5D=StartDateTime&filter%5BQuery%5D%5Bskip_locked%5D=&filter%5BQuery%5D%5Blimit%5D=0

This is the console error (I tried to expand and copy it here):

csrf-magic.js:41 
                
        GET http://192.168.1.222/zm/index.php?view=request&request=events&task=query&filter%5BQuery%5D%5Bterms%5D%5B0%5D%5Battr%5D=Monitor&filter%5BQuery%5D%5Bterms%5D%5B0%5D%5Bop%5D=%3D&filter%5BQuery%5D%5Bterms%5D%5B0%5D%5Bval%5D=4&filter%5BQuery%5D%5Bterms%5D%5B1%5D%5Battr%5D=StartDate&filter%5BQuery%5D%5Bterms%5D%5B1%5D%5Bop%5D=%3E%3D&filter%5BQuery%5D%5Bterms%5D%5B1%5D%5Bval%5D=&filter%5BQuery%5D%5Bsort_asc%5D=1&filter%5BQuery%5D%5Bsort_field%5D=StartDateTime&filter%5BQuery%5D%5Bskip_locked%5D=&filter%5BQuery%5D%5Blimit%5D=0&search=&sort=StartDateTime&order=asc&offset=0&limit=10&filter%5BQuery%5D%5Bterms%5D%5B0%5D%5Battr%5D=Monitor&filter%5BQuery%5D%5Bterms%5D%5B0%5D%5Bop%5D=%3D&filter%5BQuery%5D%5Bterms%5D%5B0%5D%5Bval%5D%5B%5D=4&undefined=&filter%5BQuery%5D%5Bterms%5D%5B1%5D%5Bcnj%5D=and&filter%5BQuery%5D%5Bterms%5D%5B1%5D%5Battr%5D=StartDate&filter%5BQuery%5D%5Bterms%5D%5B1%5D%5Bop%5D=%3E%3D&filter%5BQuery%5D%5Bterms%5D%5B1%5D%5Bval%5D= 500 (Internal Server Error)
send @ csrf-magic.js:41
send @ skins_classic_js_jquery.min-dark-1682113953.js:2
ajax @ skins_classic_js_jquery.min-dark-1682113953.js:2
ajaxRequest @ skins_classic_views_js_events-dark-1682113953.js:46
calculateObjectValue @ skins_classic_js_bootstrap-table-1.21.1_bootstrap-table.min-dark-1682113953.js:10
value @ skins_classic_js_bootstrap-table-1.21.1_bootstrap-table.min-dark-1682113953.js:10
initServer @ skins_classic_js_bootstrap-table-1.21.1_extensions_cookie_bootstrap-table-cookie-dark-1682113953.js:3019
value @ skins_classic_js_bootstrap-table-1.21.1_bootstrap-table.min-dark-1682113953.js:10
value @ skins_classic_js_bootstrap-table-1.21.1_bootstrap-table.min-dark-1682113953.js:10
onSearch @ skins_classic_js_bootstrap-table-1.21.1_extensions_cookie_bootstrap-table-cookie-dark-1682113953.js:3212
value @ skins_classic_js_bootstrap-table-1.21.1_bootstrap-table.min-dark-1682113953.js:10
(anonymous) @ skins_classic_js_bootstrap-table-1.21.1_bootstrap-table.min-dark-1682113953.js:10
each @ skins_classic_js_jquery.min-dark-1682113953.js:2
each @ skins_classic_js_jquery.min-dark-1682113953.js:2
i.default.fn.bootstrapTable @ skins_classic_js_bootstrap-table-1.21.1_bootstrap-table.min-dark-1682113953.js:10
initPage @ skins_classic_views_js_events-dark-1682113953.js:423
(anonymous) @ skins_classic_views_js_events-dark-1682113953.js:443
e @ skins_classic_js_jquery.min-dark-1682113953.js:2
t @ skins_classic_js_jquery.min-dark-1682113953.js:2
setTimeout (async)
(anonymous) @ skins_classic_js_jquery.min-dark-1682113953.js:2
c @ skins_classic_js_jquery.min-dark-1682113953.js:2
fireWith @ skins_classic_js_jquery.min-dark-1682113953.js:2
fire @ skins_classic_js_jquery.min-dark-1682113953.js:2
c @ skins_classic_js_jquery.min-dark-1682113953.js:2
fireWith @ skins_classic_js_jquery.min-dark-1682113953.js:2
ready @ skins_classic_js_jquery.min-dark-1682113953.js:2
$ @ skins_classic_js_jquery.min-dark-1682113953.js:2
skins_classic_views_js_events-dark-1682113953.js:60 error {readyState: 4, getResponseHeader: ƒ, getAllResponseHeaders: ƒ, setRequestHeader: ƒ, overrideMimeType: ƒ, …}

myankele, 2023-04-30:
Nevermind, my issues fixed with 1.37.40.
Thanks!

rnveach, 2023-05-01:
@myankele 
> my issues fixed with 1.37.40.

Is that the correct version number? My ubuntu says 1.36.33 is the latest and https://github.com/ZoneMinder/zoneminder/releases does not seem to go above this version.

myankele, 2023-05-01:
Instead of this step:  sudo add-apt-repository ppa:iconnor/zoneminder-1.36
You do :  sudo add-apt-repository ppa:iconnor/zoneminder-master

It's not recommended, since this one changes daily and is released to collect bug data by the beta testers.

connortechnology, 2026-06-12:
Initial size is taken from MOnitor->Width and Height.  If it is playing using html video tag, it will re-adjust to whatever is stored in the mp4 itself, so if different aspect ratio may change.  It may be instructive to compare viewed html for the video/img tag, and what is actually in the mp4, using ffprobe.