ZoneMinder/zoneminder #3611: Old events are not always erased

rnveach, 2022-10-21:
**Describe Your Environment**
Version: v1.36.31 
Installed by: sudo apt install -y zoneminder

OS:
````
Description:    Ubuntu 18.04.6 LTS
Release:        18.04
Codename:       bionic
````
Browser: Firefox 106.0.1

**If the issue concerns a camera**
Doesn't

**Describe the bug**
ZM randomly leaves folders and files in `/var/cache/zoneminder/events` even though `PurgeWhenFull` is running when disk percent is at 90% or greater.

When this occurs, ZM act like the files don't exist in it's database but continue to take up HD space. I one time found the majority of these files took up so much space, ZM wasn't keeping recordings for long and I had to manually delete them.

Example:
````
drwxr-xr-x    4 www-data www-data 36864 Jul 10 10:34 2022-06-28/
drwxr-xr-x    5 www-data www-data 36864 Jul 11 05:08 2022-06-29/
drwxr-xr-x    5 www-data www-data 36864 Jul 12 02:28 2022-06-30/
drwxr-xr-x    3 www-data www-data 20480 Jul 26 02:12 2022-07-15/
drwxr-xr-x    3 www-data www-data 36864 Aug  2 14:29 2022-07-23/
drwxr-xr-x    3 www-data www-data 36864 Aug 14 18:59 2022-08-04/
drwxr-xr-x    3 www-data www-data 36864 Sep  5 21:07 2022-08-29/
drwxr-xr-x    3 www-data www-data 36864 Sep 16 08:56 2022-09-06/
drwxr-xr-x    3 www-data www-data 36864 Sep 23 00:26 2022-09-10/
drwxr-xr-x  770 www-data www-data 20480 Oct 20 11:09 2022-10-08/
drwxr-xr-x  381 www-data www-data 12288 Oct  9 23:52 2022-10-09/
drwxr-xr-x  720 www-data www-data 20480 Oct 10 23:54 2022-10-10/
drwxr-xr-x 1445 www-data www-data 36864 Oct 11 23:58 2022-10-11/
drwxr-xr-x 1028 www-data www-data 28672 Oct 12 23:58 2022-10-12/
drwxr-xr-x 1311 www-data www-data 36864 Oct 13 23:50 2022-10-13/
drwxr-xr-x 1196 www-data www-data 36864 Oct 14 23:58 2022-10-14/
drwxr-xr-x  489 www-data www-data 12288 Oct 15 23:57 2022-10-15/
drwxr-xr-x 1162 www-data www-data 36864 Oct 16 23:58 2022-10-16/
drwxr-xr-x 2279 www-data www-data 61440 Oct 17 23:58 2022-10-17/
drwxr-xr-x 1535 www-data www-data 36864 Oct 18 23:59 2022-10-18/
drwxr-xr-x 1134 www-data www-data 36864 Oct 19 23:59 2022-10-19/
drwxr-xr-x  743 www-data www-data 20480 Oct 20 14:38 2022-10-20/
````

Everything before `2022-10-08` are things that ZM should have deleted as shown by the jumps. There is always some recording daily as I work from home. Some of the excess folders only had a few event folders in them.

**To Reproduce**
Leave ZM on recording for days with `PurgeWhenFull` running.

**Expected behavior**
I expect ZM to keep account of its save and delete them as part of the purge process when they get too old.

**Debug Logs**
N/A

knight-of-ni, 2022-10-21:
what is the status of OPT_FAST_DELETE?

rnveach, 2022-10-21:
`OPT_FAST_DELETE` is unchecked on the under Options, Systems tab.

connortechnology, 2022-10-21:
Going to need debug logs from zmfilter.pl.  Turn on debug under Options->Logging->DEBUG.  restart ZM.  Wait a few minutes, the in /var/log/zm/zmfilter_1.log (If your purge when full filter is a different id, then correct the # in the filename).  Post results here.

Next question is are those event files existing in the db?  Look in the data directory.  The directory names correspond to an event ID.  Does the event exist in the db? (cutnpaste it into a filter and click list).  If not, then the filter will never delete them. It will only delete events it knows about.

In the case zmaudit.pl can be used to clear up the fs.

rnveach, 2022-10-21:
> Next question is are those event files existing in the db? Look in the data directory. The directory names correspond to an event ID. Does the event exist in the db? (cutnpaste it into a filter and click list). 

I use MySQL and can connect to the database and list the tables. What query do I run that shows what you are asking me to confirm?

> If not, then the filter will never delete them. It will only delete events it knows about.

I can test when given the above, but I am going to assume it doesn't know about them. My first post mentioned that this one time got serious enough that the HD was entirely full of these undeleted events and was keeping very few new events for long. For example, I seem to keep 13-14 days worth from the current folder structure, but I was probably restricted to less than 5 when the issue occurred and thats how I knew something was going on.

This issue has been going on for a few versions, but I am just reporting it now. Besides viewing events and looking at cameras, I don't do much in the UI with events.

> Going to need debug logs from zmfilter.pl. Turn on debug under Options->Logging->DEBUG. restart ZM. Wait a few minutes, the in /var/log/zm/zmfilter_1.log (If your purge when full filter is a different id, then correct the # in the filename). 

The purge is set to run automatically, so do you need me to keep DEBUG on until the purge runs?

Assuming the issue is only when purge runs and starts deleting things, I will need some days before I can do this and report. I just upgraded from a 250 MB HD to 2 TB, so there is now plenty of empty space that needs to be filled.

connortechnology, 2022-10-22:
You could just run zmaudit.pl and see if they go away.  You coulud run zmaudit all the time (not recommended, it is too I/O intensive.) Or just run it from cron on weekends or something.

The SQL would be SELECT Id FROM Events WHERE Id=Id # from filesystem

The issue to be fixed is why deleting these events fails to delete the filesystem contents.  Something will have been logged at the time.

connortechnology, 2022-12-06:
I had a hard drive die and and during cleanup noticed nothing being logged despite the fs being read-only. So I dug in and discovered that we weren't capturing stderr and so not logging failures to delete events.  

So 1.36.33 may shed some light on what is going on.

rnveach, 2023-06-23:
I rechecked and saw some directories that I thought should be erased.

They do appear in the database, I was able to query them using the query above. They are the oldest records in the database.

I was also able to query the events in ZM's filter and noticed all the events in question had no end date and the PurgeWhenFull specifically says to ignore items without an end date.

I manually deleted them in ZM by first viewing the event and then clicking the delete icon, but ZM didn't remove the directories but ZM did remove all the files in them. I manually removed the directories from the console.

I was starting to think maybe these were just old / fluke logs from the HD transfer but I did a search in the DB for other ones with no end date and I see entries for April, May, and June (not today). Only 6 events in the entire DB have no end date of 223,000 events. I just leave ZM on Modect and don't really change it, so I don't know what is special with these dates.

connortechnology, 2023-06-23:
PurgeWhenFull does not come with a rule about end date by default. That has been added by the user.  

The filter can only match events that match the rules... 

Otherwise.. run zmaudit from time to time... 

There might be something here... but seriously... we seem to be beating around the bush... who cares if it left an empty directory?  The cpu/io/ram cost to constantly scan for such things is huge.

rnveach, 2023-06-23:
Ok, if I did add it, I can't remember why. I am not fully concerned about leaving an empty directory after manually deleting events. I was just noting it.

My question is why does ZM sometimes leave some rare events as left "opened" indefinitely? They have a length and are no longer being added to.

connortechnology, 2023-06-23:
If the endtime is not filled in, it means zm crashed before doing so.  When closing an event, the last thing it does is a db update setting the endtime.

rnveach, 2023-06-23:
Ok, I removed the end date check from the filter. I will close this for now and if I have something new, I will re-open.

Thank you for your help.

closed 2023-06-23 by rnveach