ZoneMinder/zoneminder #3612: DB update from 1.37.23 to 1.37.25 failed

romale, 2022-10-24:
ZM 1.37.23
Centos7
Repo: http://zmrepo.connortechnology.com/el/$releasever/$basearch

After update ZM via yum, zmupdate.pl fail with error message:
```
Initiating database upgrade to version 1.37.25 from version 1.37.23

Do you wish to take a backup of your database prior to upgrading?
This may result in a large file in /var/lib/zoneminder/temp if you have a lot of events.
Press 'y' for a backup or 'n' to continue : n

Upgrading database to version 1.37.25
Upgrading DB to 1.37.25 from 1.37.23
ERROR 1067 (42000) at line 10: Invalid default value for 'UpdatedOn'
Output: 
Command 'mysql -umyuser -p'mypassword' -hlocalhost zm < /usr/share/zoneminder/db/zm_update-1.37.25.sql' exited with status: 1

```

romale, 2022-10-27:
By default centos7 has
```
mysql -V
mysql  Ver 15.1 Distrib 5.5.68-MariaDB, for Linux (x86_64) using readline 5.1
```

But from internet information:
```
From the [MySQL 5.5 manual](https://dev.mysql.com/doc/refman/5.5/en/data-type-defaults.html):

    You cannot set the default for a date column to be the value of a function such as NOW() or CURRENT_DATE. The exception is that you can specify CURRENT_TIMESTAMP as the default for a TIMESTAMP column.

Therefore, what you want to achieve will work in MySQL 5.5 if you add a TIMESTAMP column instead of a DATE column.

The changes in 5.6.x that allow the functionality are documented [here](https://dev.mysql.com/doc/refman/5.6/en/timestamp-initialization.html), and I'll quote the relevant summary for completeness:

    As of MySQL 5.6.5, TIMESTAMP and DATETIME columns can be automatically initializated and updated to the current date and time (that is, the current timestamp). Before 5.6.5, this is true only for TIMESTAMP, and for at most one TIMESTAMP column per table.
```

So, sql schema upgrade on default rhlel7/centos7/... will not work, because of sql command has "default NOW() " :
```
cat /usr/share/zoneminder/db/zm_update-1.37.25.sql
SET @s = (SELECT IF(
    (SELECT COUNT(*) FROM INFORMATION_SCHEMA.COLUMNS WHERE table_schema = DATABASE()
     AND table_name = 'Monitor_Status'
     AND column_name = 'UpdatedOn'
    ) > 0,
"SELECT 'Column UpdatedOn already exists in Monitor_Status'",
"ALTER TABLE `Monitor_Status` ADD `UpdatedOn`     datetime NOT NULL default NOW() AFTER CaptureBandwidth"
));

PREPARE stmt FROM @s;
EXECUTE stmt;
```

connortechnology, 2022-10-27:
Please try latest with ac55528f8.  

Please note that we are officially NOT supporting centos 7 for 1.37/1.38.  We are also deprecating support for ubuntu xenial and debian... stretch I think.  

I'm amazed you got this far with it.

closed 2022-10-31 by romale