ZoneMinder/zoneminder #3613: ZM doesn't create symlinks in events directory such as monitor name.

spanchy, 2022-10-26:
- latest from repo. version is 1.36.31
- repository debian 11
- Debian 11

Earlier in the ZM version, that soft created a symbolic link with the monitor name for my ID-monitor, now there is no such thing. Has a bug?
[image]
[image]

How I can setup my configuration for create symlink?

connortechnology, 2022-10-26:
The symlink is created when saving the monitor.  It is used NOWHERE else in ZoneMinder. It is there purely for user convenience.  Some filesystems don't support symlinks, in which case it will not be created.  Which is why we don't use it anywhere else, because we can't count on it.

spanchy, 2022-10-26:
> The symlink is created when saving the monitor.  It is used NOWHERE else in ZoneMinder. It is there purely for user convenience.  Some filesystems don't support symlinks, in which case it will not be created.  Which is why we don't use it anywhere else, because we can't count on it.

Hello, I'm have the installed zm (4 different server) and have setting archive drive on ext2, ext4, xfs filesystems - symlink not creating after saving monitor. I know that zm not used this symlinks, but I need them.

connortechnology, 2022-10-28:
So I took a look at the code and found I could make some improvements.  So you can try out the version in my proposed ppa (ppa:iconnor/zoneminder-proposed), to get a pre-release of 1.36.32.  See if it solves your problems.

spanchy, 2022-12-07:
> With the release of 1.36/32, those fixes should be generally available to debian using packages at zmrepo.
> 
> 
> 
> Can you update and report?

Hello, yes, of course. I add reply later.

spanchy, 2022-12-30:
@connortechnology works fine. Thank you! Close it.

closed 2022-12-30 by connortechnology