ZoneMinder/zoneminder #3614: 1.36.31 memory issue

DanwithArduino, 2022-10-26:
**Describe Your Environment**
- Version of ZoneMinder [1.36.31]
- How you installed ZoneMinder [PPA]
- Full name and version of OS [Ubuntu 20.04.5]
- Running on a Dell R210 ii with dedicated OS SSD and storage HDD's. 

**Describe the bug**
After updating to 1.31.31 from 1.31.26, ZM appears to devour all memory and lock up the server. It does not appear the server ever establishes any sort of successful boot. Tail of the server logs is below. 

**Expected behavior**
I expect to be able to turn the server on and not have all of the memory consumed. 

**Debug Logs**
```

zmc_m1[2535]: WAR [zmc_m1] [Decoding is not keeping up. We are 355 seconds behind capture.]
zmc_m12[2204]: WAR [zmc_m12] [Decoding is not keeping up. We are 413 seconds behind capture.]
zmc_m13[2367]: WAR [zmc_m13] [Decoding is not keeping up. We are 364 seconds behind capture.]
zmc_m3[1981]: WAR [zmc_m3] [Decoding is not keeping up. We are 416 seconds behind capture.]
zmc_m2[1941]: WAR [zmc_m2] [Decoding is not keeping up. We are 1364 seconds behind capture.]
zmc_m19[2562]: WAR [zmc_m19] [Decoding is not keeping up. We are 442 seconds behind capture.]
zmc_m5[2376]: WAR [zmc_m5] [Decoding is not keeping up. We are 416 seconds behind capture.]
zmc_m16[2013]: WAR [zmc_m16] [Decoding is not keeping up. We are 425 seconds behind capture.]
zmc_m8[2003]: WAR [zmc_m8] [Decoding is not keeping up. We are 419 seconds behind capture.]
zmc_m5[2376]: WAR [zmc_m5] [Decoding is not keeping up. We are 416 seconds behind capture.]
zmc_m1[2535]: WAR [zmc_m1] [Decoding is not keeping up. We are 355 seconds behind capture.]
zmc_m12[2204]: WAR [zmc_m12] [Decoding is not keeping up. We are 412 seconds behind capture.]
zmc_m13[2367]: WAR [zmc_m13] [Decoding is not keeping up. We are 364 seconds behind capture.]
zmc_m13[2367]: WAR [zmc_m13] [Decoding is not keeping up. We are 364 seconds behind capture.]
zmc_m3[1981]: WAR [zmc_m3] [Decoding is not keeping up. We are 416 seconds behind capture.]
zmc_m2[1941]: WAR [zmc_m2] [Decoding is not keeping up. We are 1364 seconds behind capture.]
zmc_m1[2535]: WAR [zmc_m1] [Decoding is not keeping up. We are 355 seconds behind capture.]
zmc_m11[1356]: WAR [zmc_m11] [Decoding is not keeping up. We are 2731 seconds behind capture.]
zmc_m13[2367]: WAR [zmc_m13] [Decoding is not keeping up. We are 364 seconds behind capture.]
zmc_m4[2347]: WAR [zmc_m4] [Decoding is not keeping up. We are 351 seconds behind capture.]
zmc_m12[2204]: WAR [zmc_m12] [Decoding is not keeping up. We are 412 seconds behind capture.]
zmc_m19[2562]: WAR [zmc_m19] [Decoding is not keeping up. We are 442 seconds behind capture.]
zmc_m16[2013]: WAR [zmc_m16] [Decoding is not keeping up. We are 425 seconds behind capture.]
zmc_m2[1941]: WAR [zmc_m2] [Decoding is not keeping up. We are 1364 seconds behind capture.]
zmc_m1[2535]: WAR [zmc_m1] [Decoding is not keeping up. We are 355 seconds behind capture.]
zmc_m3[1981]: WAR [zmc_m3] [Decoding is not keeping up. We are 416 seconds behind capture.]
zmc_m5[2376]: WAR [zmc_m5] [Decoding is not keeping up. We are 416 seconds behind capture.]
zmc_m13[2367]: WAR [zmc_m13] [Decoding is not keeping up. We are 364 seconds behind capture.]
zmc_m4[2347]: WAR [zmc_m4] [Decoding is not keeping up. We are 351 seconds behind capture.]
zmc_m12[2204]: WAR [zmc_m12] [Decoding is not keeping up. We are 412 seconds behind capture.]
zmc_m19[2562]: WAR [zmc_m19] [Decoding is not keeping up. We are 442 seconds behind capture.]
zmc_m11[1356]: WAR [zmc_m11] [Decoding is not keeping up. We are 2731 seconds behind capture.]
zmc_m2[1941]: WAR [zmc_m2] [Decoding is not keeping up. We are 1363 seconds behind capture.]
zmc_m1[2535]: WAR [zmc_m1] [Decoding is not keeping up. We are 355 seconds behind capture.]
zmc_m16[2013]: WAR [zmc_m16] [Decoding is not keeping up. We are 425 seconds behind capture.]
zmc_m3[1981]: WAR [zmc_m3] [Decoding is not keeping up. We are 416 seconds behind capture.]
zmc_m10[1352]: INF [zmc_m10] [axis-accc8e561241 - group 3 - cam 4: 71600 - Capturing at 25.01 fps, capturing bandwidth 894bytes/sec Analysing at 1.75 fps]
zmc_m13[2367]: WAR [zmc_m13] [Decoding is not keeping up. We are 364 seconds behind capture.]
zmc_m4[2347]: WAR [zmc_m4] [Decoding is not keeping up. We are 351 seconds behind capture.]
zmc_m5[2376]: WAR [zmc_m5] [Decoding is not keeping up. We are 416 seconds behind capture.]
zmdc[1286]: INF [Starting pending process, zmc -m 6]
zmc_m12[2204]: WAR [zmc_m12] [Decoding is not keeping up. We are 413 seconds behind capture.]
zmc_m19[2562]: WAR [zmc_m19] [Decoding is not keeping up. We are 443 seconds behind capture.]
zmc_m2[1941]: WAR [zmc_m2] [Decoding is not keeping up. We are 1364 seconds behind capture.]
zmc_m1[2535]: INF [zmc_m1] [Byberry Truck Parking and Loading: 6400 - Capturing at 12.30 fps, capturing bandwidth 59612bytes/sec Analysing at 0.00 fps]
zmc_m11[1356]: WAR [zmc_m11] [Decoding is not keeping up. We are 2732 seconds behind capture.]
zmc_m16[2013]: WAR [zmc_m16] [Decoding is not keeping up. We are 426 seconds behind capture.]
zmc_m9[1348]: INF [zmc_m9] [axis-accc8e561241 - group 3 - cam 3: 43000 - Capturing at 14.97 fps, capturing bandwidth 514bytes/sec Analysing at 1.80 fps]
zmc_m7[1340]: INF [zmc_m7] [axis-accc8e561241 - group 3 - cam 1: 43000 - Capturing at 14.98 fps, capturing bandwidth 515bytes/sec Analysing at 1.80 fps]
zmc_m17[1376]: INF [zmc_m17] [axis-accc8e56123f - group 2 - cam 4: 43000 - Capturing at 14.98 fps, capturing bandwidth 514bytes/sec Analysing at 1.20 fps]
zmc_m14[1368]: INF [zmc_m14] [axis-accc8e561242 - group 4 - cam 4: 43000 - Capturing at 14.99 fps, capturing bandwidth 515bytes/sec Analysing at 1.80 fps]
zmc_m8[2003]: INF [zmc_m8] [Rear Building - West view rear loading dock: 30100 - Capturing at 9.94 fps, capturing bandwidth 36076bytes/sec Analysing at 0.00 fps]
zmwatch[1399]: ERR [Memory map file '/dev/shm/zm.mmap.6' does not exist in zmMemAttach.  zmc might not be running.]
zmc_m3[1981]: INF [zmc_m3] [Production - Drum and Pail: 29500 - Capturing at 14.99 fps, capturing bandwidth 56898bytes/sec Analysing at 0.45 fps]
zmc_m13[2367]: INF [zmc_m13] [East Parking - Towars Byberry: 9800 - Capturing at 10.12 fps, capturing bandwidth 10134bytes/sec Analysing at 0.10 fps]
zmc_m10[1352]: INF [zmc_m10] [axis-accc8e561241 - group 3 - cam 4: 71700 - Capturing at 10.60 fps, capturing bandwidth 378bytes/sec Analysing at 0.00 fps]
zmc_m4[2347]: INF [zmc_m4] [Front Warehouse Main Asile: 10700 - Capturing at 9.24 fps, capturing bandwidth 8220bytes/sec Analysing at 0.28 fps]
zmc_m5[2376]: INF [zmc_m5] [Store: 10000 - Capturing at 11.02 fps, capturing bandwidth 71750bytes/sec Analysing at 0.11 fps]
zmc_m12[2204]: INF [zmc_m12] [CEO Office and West Office Door: 16900 - Capturing at 11.43 fps, capturing bandwidth 49410bytes/sec Analysing at 0.11 fps]

```

connortechnology, 2022-10-26:
Interesting, as there were no real changes to the c++ code between .26 and .31.  Changes were all to the web UI php code.

The logs are clear, decoding is not keeping up, telling me you don't have anywhere near enough cpu.  We are going to need actual debug logs.  Turn on debugging in Options->Logging->DEBUG

DanwithArduino, 2022-10-26:
I don't think I can get to the UI at all. I just ran update/upgrade again in case an underlying package was unhappy, and within a few seconds swap was nearly fully loaded. After trying to get to the UI, swap steadily increased up to the 4gb limit. CPU fluctuating between 3-15% utilization. Everything was fine - relatively - prior to the upgrade. The server definitely ran burdened but it ran.

connortechnology, 2022-10-26:
once the machine is up, login at the console and do sudo systemctl stop zoneminder. 

My other thought was: if you were in fact, coming from 1.34.26 and NOT 1.36.26 and didn't read the release notes... you wouldn't have updated your buffer settings.  Go into each monitor and set a value for MaxImageBuffer and set ImageBuffer to 3.

DanwithArduino, 2022-10-27:
No it was 1.36. I've been consistently within 30days of a release for the past year or so. Not sure what my issue with the last update and reboot but everything seems ok now.

closed 2022-10-27 by DanwithArduino

badrpc, 2022-12-02:
After switching from 1.36.12 (current FreeBSD port) to 1.36.32 I noticed that my machine started running out of swap space. I limited the amount of memory available to zoneminder jail and this resolved the machine wide issue but zoneminder was no longer able to record more than a few frames. The zmc process was trying to allocate memory above the limit, was denied and all recording were shown unfinished in UI. I went back to 1.36.12 with the same limit configured and now I can see a couple of recordings with over a 100 frames.

Logs contain something like this:
```
12/02/22 10:01:34.182801 zmc_m3[201410].INF-zm_monitor.cpp/1981 [Monitor-1: 283 - Gone into alarm state PreAlarmCount: 0 > AlarmFrameCount:1 Cause:Motion All]
12/02/22 10:01:34.286206 zmc_m3[201410].INF-zm_monitor.cpp/1987 [Monitor-1: 281 - Opening new event 116, alarm start]
12/02/22 10:01:34.589951 zmc_m3[201410].INF-zm_monitor.cpp/2020 [Monitor-1: 286 - Gone into alert state]
12/02/22 10:01:34.778408 zmc_m3[201410].INF-zm_monitor.cpp/2004 [Monitor-1: 290 - Gone back into alarm state]
12/02/22 10:01:35.286112 zmc_m3[201409].FAT-zm_image.h/55 [Memory allocation failed: Invalid argument]
12/02/22 10:01:35.286503 zmc_m3[201410].FAT-zm_image.h/55 [Memory allocation failed: Invalid argument]
```

Followed I believe by a restart of zmc. My machine is a bit low on memory compared to the load but 1.36.12 certainly manages to work where 1.36.32 does not.