ZoneMinder/zoneminder #3616: zmc -m 1 crashes

radoslavv, 2022-10-30:
**Describe Your Environment**
- Version of ZoneMinder: **latest**
- How you installed ZoneMinder: **git clone + compile**
- Full name and version of OS: **Ubuntu 20.04**
- Browser name and version (if this is an issue with the web interface): **irrelevant**

**If the issue concerns a camera**
irrelevant

**Describe the bug**
zm -m 1 crashes
```
10/30/22 02:27:32.606322 zmc_m1[20353].WAR-zm_monitor.cpp/807 [Monitor 1 has incorrect zone_count 0 != 2]
10/30/22 02:27:32.618066 zmc_m1[20353].ERR-zm_signal.cpp/50 [Got signal 7 (Bus error), crashing]
10/30/22 02:27:32.618143 zmc_m1[20353].ERR-zm_signal.cpp/80 [Signal address is 0x7fa10b984000, from 0x7fa12253c1ca]
10/30/22 02:27:32.618382 zmc_m1[20353].ERR-zm_signal.cpp/101 [Backtrace 0: zmc(+0xb530f) [0x555e4503930f]]
10/30/22 02:27:32.618410 zmc_m1[20353].ERR-zm_signal.cpp/101 [Backtrace 1: /lib/x86_64-linux-gnu/libc.so.6(+0x42520) [0x7fa1223dd520]]
10/30/22 02:27:32.618442 zmc_m1[20353].ERR-zm_signal.cpp/101 [Backtrace 2: /lib/x86_64-linux-gnu/libc.so.6(+0x1a11ca) [0x7fa12253c1ca]]
10/30/22 02:27:32.618462 zmc_m1[20353].ERR-zm_signal.cpp/101 [Backtrace 3: zmc(+0x6292d) [0x555e44fe692d]]
10/30/22 02:27:32.618481 zmc_m1[20353].ERR-zm_signal.cpp/101 [Backtrace 4: zmc(main+0x5a2) [0x555e44fb12f2]]
10/30/22 02:27:32.618500 zmc_m1[20353].ERR-zm_signal.cpp/101 [Backtrace 5: /lib/x86_64-linux-gnu/libc.so.6(+0x29d90) [0x7fa1223c4d90]]
10/30/22 02:27:32.618524 zmc_m1[20353].ERR-zm_signal.cpp/101 [Backtrace 6: /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x80) [0x7fa1223c4e40]]
10/30/22 02:27:32.618544 zmc_m1[20353].ERR-zm_signal.cpp/101 [Backtrace 7: zmc(+0x30355) [0x555e44fb4355]]

```
Same issue has docker image with latest code [bkjaya1952/zoneminder-master-docker-latest](https://hub.docker.com/r/bkjaya1952/zoneminder-master-docker-latest)

**To Reproduce**
After clean installation define new camera with 4K image size (does not necessarily needs to be alive as module crashes before any connection happens).

**Expected behavior**
NOT crash.

**Debug Logs**
valgrind -v --leak-check=full zmc -m 1 2>&1 | tee valgrind.txt
```
==20229== Memcheck, a memory error detector
==20229== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==20229== Using Valgrind-3.18.1-42b08ed5bd-20211015 and LibVEX; rerun with -h for copyright info
==20229== Command: zmc -m 1
==20229== 
--20229-- Valgrind options:
--20229--    --leak-check=full
--20229--    -v
--20229-- Contents of /proc/version:
--20229--   Linux version 4.14.24-qnap (root@Q64-100-75) (gcc version 4.9.2 (toolchain config: [gcc-4.9.2 binutils-2.25 glibc-2.21])) #1 SMP Wed Aug 24 05:56:44 CST 2022
--20229-- 
--20229-- Arch and hwcaps: AMD64, LittleEndian, amd64-cx16-lzcnt-rdtscp-sse3-ssse3-avx-avx2-bmi-f16c-rdrand-rdseed
--20229-- Page sizes: currently 4096, max supported 4096
--20229-- Valgrind library directory: /usr/libexec/valgrind
--20229-- Reading syms from /usr/local/bin/zmc
--20229-- Reading syms from /usr/lib/x86_64-linux-gnu/ld-linux-x86-64.so.2
--20229--   Considering /usr/lib/debug/.build-id/61/ef896a699bb1c2e4e231642b2e1688b2f1a61e.debug ..
--20229--   .. build-id is valid
--20229-- Reading syms from /usr/libexec/valgrind/memcheck-amd64-linux
--20229--    object doesn't have a symbol table
--20229--    object doesn't have a dynamic symbol table
--20229-- Scheduler: using generic scheduler lock implementation.
--20229-- Reading suppressions file: /usr/libexec/valgrind/default.supp
==20229== embedded gdbserver: reading from /tmp/vgdb-pipe-from-vgdb-to-20229-by-???-on-ubuntu_22.04
==20229== embedded gdbserver: writing to   /tmp/vgdb-pipe-to-vgdb-from-20229-by-???-on-ubuntu_22.04
==20229== embedded gdbserver: shared mem   /tmp/vgdb-pipe-shared-mem-vgdb-20229-by-???-on-ubuntu_22.04
==20229== 
==20229== TO CONTROL THIS PROCESS USING vgdb (which you probably
==20229== don't want to do, unless you know exactly what you're doing,
==20229== or are doing some strange experiment):
==20229==   /usr/bin/vgdb --pid=20229 ...command...
==20229== 
==20229== TO DEBUG THIS PROCESS USING GDB: start GDB like this
==20229==   /path/to/gdb zmc
[... 677 lines cut ...]
--20229-- REDIR: 0x7491440 (libc.so.6:__memchr_avx2) redirected to 0x484fd50 (memchr)
--20229-- REDIR: 0x73bc200 (libc.so.6:__strstr_sse2_unaligned) redirected to 0x4853a50 (strstr)
--20229-- REDIR: 0x7490f00 (libc.so.6:__strncmp_avx2) redirected to 0x484f3e0 (strncmp)
--20229-- REDIR: 0x7491710 (libc.so.6:__rawmemchr_avx2) redirected to 0x4853360 (rawmemchr)
--20229-- REDIR: 0x7491c00 (libc.so.6:__memcmp_avx2_movbe) redirected to 0x4852010 (bcmp)
--20229-- REDIR: 0x7498910 (libc.so.6:__memcpy_chk_avx_unaligned_erms) redirected to 0x48532c0 (__memmove_chk)
--20229-- REDIR: 0x70739b0 (libstdc++.so.6:operator new(unsigned long)) redirected to 0x4848fa0 (operator new(unsigned long))
--20229-- REDIR: 0x742c990 (libc.so.6:__strcpy_chk) redirected to 0x48533a0 (__strcpy_chk)
--20229-- REDIR: 0x7495580 (libc.so.6:__strchrnul_avx2) redirected to 0x4853330 (strchrnul)
--20229-- REDIR: 0x7071c60 (libstdc++.so.6:operator delete(void*, unsigned long)) redirected to 0x484bb00 (operator delete(void*, unsigned long))
--20229-- REDIR: 0x7071c50 (libstdc++.so.6:operator delete(void*)) redirected to 0x484b840 (operator delete(void*))
--20229-- REDIR: 0x7490730 (libc.so.6:__strcspn_sse42) redirected to 0x4853c70 (strcspn)
--20229-- REDIR: 0x7490990 (libc.so.6:__strspn_sse42) redirected to 0x4853d50 (strspn)
--20229-- REDIR: 0x74922b0 (libc.so.6:__strcasecmp_avx) redirected to 0x484f560 (strcasecmp)
--20229-- REDIR: 0x7073a10 (libstdc++.so.6:operator new[](unsigned long)) redirected to 0x484a280 (operator new[](unsigned long))
--20229-- REDIR: 0x7071c80 (libstdc++.so.6:operator delete[](void*)) redirected to 0x484ca20 (operator delete[](void*))
--20229-- REDIR: 0x7498900 (libc.so.6:__mempcpy_avx_unaligned_erms) redirected to 0x4853440 (mempcpy)
--20229-- REDIR: 0x7496e30 (libc.so.6:__strcpy_avx2) redirected to 0x484ee60 (strcpy)
--20229-- REDIR: 0x7490870 (libc.so.6:__strpbrk_sse42) redirected to 0x4853c00 (strpbrk)
--20229-- REDIR: 0x70739f0 (libstdc++.so.6:operator new(unsigned long, std::nothrow_t const&)) redirected to 0x4849820 (operator new(unsigned long, std::nothrow_t const&))
--20229-- REDIR: 0x74971c0 (libc.so.6:__strncpy_avx2) redirected to 0x484efe0 (strncpy)
--20229-- REDIR: 0x739f160 (libc.so.6:posix_memalign) redirected to 0x484df60 (posix_memalign)
--20229-- REDIR: 0x7497b20 (libc.so.6:__stpcpy_avx2) redirected to 0x4852130 (stpcpy)

```

radoslavv, 2022-10-30:
```
==20229== 
==20229== HEAP SUMMARY:
==20229==     in use at exit: 50,221,392 bytes in 1,697 blocks
==20229==   total heap usage: 9,643 allocs, 7,946 frees, 51,521,483 bytes allocated
==20229== 
==20229== Searching for pointers to 1,676 not-freed blocks
==20229== Checked 229,094,552 bytes
==20229== Skipped 65,609,728 bytes due to read errors
==20229== 
==20229== 35 bytes in 1 blocks are possibly lost in loss record 307 of 772
==20229==    at 0x4848899: malloc (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==20229==    by 0x693A4DF: ??? (in /usr/lib/x86_64-linux-gnu/libmysqlclient.so.21.2.31)
==20229==    by 0x68EE138: mysql_select_db (in /usr/lib/x86_64-linux-gnu/libmysqlclient.so.21.2.31)
==20229==    by 0x1408FE: zmDbConnect() (in /usr/local/bin/zmc)
==20229==    by 0x13505E: main (in /usr/local/bin/zmc)
==20229== 
==20229== 39 bytes in 1 blocks are possibly lost in loss record 308 of 772
==20229==    at 0x4848899: malloc (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==20229==    by 0x693A4DF: ??? (in /usr/lib/x86_64-linux-gnu/libmysqlclient.so.21.2.31)
==20229==    by 0x68EC3FF: ??? (in /usr/lib/x86_64-linux-gnu/libmysqlclient.so.21.2.31)
==20229==    by 0x68F1FA3: ??? (in /usr/lib/x86_64-linux-gnu/libmysqlclient.so.21.2.31)
==20229==    by 0x68EDFC7: mysql_real_connect (in /usr/lib/x86_64-linux-gnu/libmysqlclient.so.21.2.31)
==20229==    by 0x140A5A: zmDbConnect() (in /usr/local/bin/zmc)
==20229==    by 0x13505E: main (in /usr/local/bin/zmc)
==20229== 
==20229== 39 bytes in 1 blocks are possibly lost in loss record 309 of 772
==20229==    at 0x4848899: malloc (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==20229==    by 0x693A4DF: ??? (in /usr/lib/x86_64-linux-gnu/libmysqlclient.so.21.2.31)
==20229==    by 0x68EC442: ??? (in /usr/lib/x86_64-linux-gnu/libmysqlclient.so.21.2.31)
==20229==    by 0x68F1FA3: ??? (in /usr/lib/x86_64-linux-gnu/libmysqlclient.so.21.2.31)
==20229==    by 0x68EDFC7: mysql_real_connect (in /usr/lib/x86_64-linux-gnu/libmysqlclient.so.21.2.31)
==20229==    by 0x140A5A: zmDbConnect() (in /usr/local/bin/zmc)
==20229==    by 0x13505E: main (in /usr/local/bin/zmc)
==20229== 
==20229== 40 bytes in 1 blocks are possibly lost in loss record 333 of 772
==20229==    at 0x4848899: malloc (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==20229==    by 0x693A4DF: ??? (in /usr/lib/x86_64-linux-gnu/libmysqlclient.so.21.2.31)
==20229==    by 0x68EC14C: ??? (in /usr/lib/x86_64-linux-gnu/libmysqlclient.so.21.2.31)
==20229==    by 0x68EC33A: ??? (in /usr/lib/x86_64-linux-gnu/libmysqlclient.so.21.2.31)
==20229==    by 0x68F1FA3: ??? (in /usr/lib/x86_64-linux-gnu/libmysqlclient.so.21.2.31)
==20229==    by 0x68EDFC7: mysql_real_connect (in /usr/lib/x86_64-linux-gnu/libmysqlclient.so.21.2.31)
==20229==    by 0x140A5A: zmDbConnect() (in /usr/local/bin/zmc)
==20229==    by 0x13505E: main (in /usr/local/bin/zmc)
==20229== 
==20229== 56 bytes in 1 blocks are possibly lost in loss record 602 of 772
==20229==    at 0x4848899: malloc (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==20229==    by 0x693A4DF: ??? (in /usr/lib/x86_64-linux-gnu/libmysqlclient.so.21.2.31)
==20229==    by 0x68DE859: mysql_server_init (in /usr/lib/x86_64-linux-gnu/libmysqlclient.so.21.2.31)
==20229==    by 0x68EE48E: mysql_init (in /usr/lib/x86_64-linux-gnu/libmysqlclient.so.21.2.31)
==20229==    by 0x1407AB: zmDbConnect() (in /usr/local/bin/zmc)
==20229==    by 0x13505E: main (in /usr/local/bin/zmc)
==20229== 
==20229== 64 bytes in 1 blocks are possibly lost in loss record 609 of 772
==20229==    at 0x4849899: operator new(unsigned long, std::nothrow_t const&) (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==20229==    by 0x68F5DE8: mysql_options4 (in /usr/lib/x86_64-linux-gnu/libmysqlclient.so.21.2.31)
==20229==    by 0x68EEFAE: ??? (in /usr/lib/x86_64-linux-gnu/libmysqlclient.so.21.2.31)
==20229==    by 0x68F1FA3: ??? (in /usr/lib/x86_64-linux-gnu/libmysqlclient.so.21.2.31)
==20229==    by 0x68EDFC7: mysql_real_connect (in /usr/lib/x86_64-linux-gnu/libmysqlclient.so.21.2.31)
==20229==    by 0x140A5A: zmDbConnect() (in /usr/local/bin/zmc)
==20229==    by 0x13505E: main (in /usr/local/bin/zmc)
==20229== 
==20229== 72 bytes in 1 blocks are possibly lost in loss record 610 of 772
==20229==    at 0x484DA83: calloc (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==20229==    by 0x693A492: ??? (in /usr/lib/x86_64-linux-gnu/libmysqlclient.so.21.2.31)
==20229==    by 0x68F1170: ??? (in /usr/lib/x86_64-linux-gnu/libmysqlclient.so.21.2.31)
==20229==    by 0x68F1FA3: ??? (in /usr/lib/x86_64-linux-gnu/libmysqlclient.so.21.2.31)
==20229==    by 0x68EDFC7: mysql_real_connect (in /usr/lib/x86_64-linux-gnu/libmysqlclient.so.21.2.31)
==20229==    by 0x140A5A: zmDbConnect() (in /usr/local/bin/zmc)
==20229==    by 0x13505E: main (in /usr/local/bin/zmc)
[... 139 lines cut ...]
==20229==    by 0x140A5A: zmDbConnect() (in /usr/local/bin/zmc)
==20229==    by 0x13505E: main (in /usr/local/bin/zmc)
==20229== 
==20229== 16,416 bytes in 1 blocks are possibly lost in loss record 766 of 772
==20229==    at 0x4848899: malloc (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==20229==    by 0x693A4DF: ??? (in /usr/lib/x86_64-linux-gnu/libmysqlclient.so.21.2.31)
==20229==    by 0x6931F50: ??? (in /usr/lib/x86_64-linux-gnu/libmysqlclient.so.21.2.31)
==20229==    by 0x69344BD: ??? (in /usr/lib/x86_64-linux-gnu/libmysqlclient.so.21.2.31)
[... cut ...]
==20229==    definitely lost: 0 bytes in 0 blocks
==20229==    indirectly lost: 0 bytes in 0 blocks
==20229==      possibly lost: 27,896 bytes in 23 blocks
==20229==    still reachable: 50,191,480 bytes in 1,653 blocks
==20229==         suppressed: 0 bytes in 0 blocks
==20229== Reachable blocks (those to which a pointer was found) are not shown.
==20229== To see them, rerun with: --leak-check=full --show-leak-kinds=all
==20229== 
==20229== ERROR SUMMARY: 23 errors from 23 contexts (suppressed: 0 from 0)
```

radoslavv, 2022-10-30:
also issue on version 1.36.31:

```
10/31/22 00:41:49.834595 zmc_m1[15681].INF-zmc.cpp/220 [Starting Capture version 1.36.31]
10/31/22 00:41:49.854607 zmc_m1[15681].ERR-zm_signal.cpp/50 [Got signal 7 (Bus error), crashing]
10/31/22 00:41:49.854649 zmc_m1[15681].ERR-zm_signal.cpp/80 [Signal address is 0x7f1d6e112000, from 0x7f1d7eb831ca]
10/31/22 00:41:49.854829 zmc_m1[15681].ERR-zm_signal.cpp/101 [Backtrace 0: /usr/bin/zmc(+0x761dc) [0x5652f0ef01dc]]
10/31/22 00:41:49.854844 zmc_m1[15681].ERR-zm_signal.cpp/101 [Backtrace 1: /lib/x86_64-linux-gnu/libc.so.6(+0x42520) [0x7f1d7ea24520]]
10/31/22 00:41:49.854857 zmc_m1[15681].ERR-zm_signal.cpp/101 [Backtrace 2: /lib/x86_64-linux-gnu/libc.so.6(+0x1a11ca) [0x7f1d7eb831ca]]
10/31/22 00:41:49.854867 zmc_m1[15681].ERR-zm_signal.cpp/101 [Backtrace 3: /usr/bin/zmc(+0x44d3d) [0x5652f0ebed3d]]
10/31/22 00:41:49.854877 zmc_m1[15681].ERR-zm_signal.cpp/101 [Backtrace 4: /usr/bin/zmc(+0x18742) [0x5652f0e92742]]
10/31/22 00:41:49.854887 zmc_m1[15681].ERR-zm_signal.cpp/101 [Backtrace 5: /lib/x86_64-linux-gnu/libc.so.6(+0x29d90) [0x7f1d7ea0bd90]]
10/31/22 00:41:49.854897 zmc_m1[15681].ERR-zm_signal.cpp/101 [Backtrace 6: /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x80) [0x7f1d7ea0be40]]
10/31/22 00:41:49.854907 zmc_m1[15681].ERR-zm_signal.cpp/101 [Backtrace 7: /usr/bin/zmc(+0x1ac95) [0x5652f0e94c95]]
10/31/22 00:41:49.854917 zmc_m1[15681].INF-zm_signal.cpp/108 [Backtrace complete, please execute the following command for more information: addr2line -e /usr/bin/zmc 0x5652f0ef01dc 0x7f1d7ea24520 0x7f1d7eb831ca 0x5652f0ebed3d 0x5652f0e92742 0x7f1d7ea0bd90 0x7f1d7ea0be40 0x5652f0e94c95]
```
```
root@ubuntu_22:/var/log/zm# addr2line -e /usr/bin/zmc 0x5652f0ef01dc 0x7f1d7ea24520 0x7f1d7eb831ca 0x5652f0ebed3d 0x5652f0e92742 0x7f1d7ea0bd90 0x7f1d7ea0be40 0x5652f0e94c95
??:0
??:0
??:0
??:0
??:0
??:0
??:0
??:0
```

radoslavv, 2022-10-30:
valgrind -v --leak-check=full zmc -m 1 2>&1 | tee valgrind.txt

```
==15884== Memcheck, a memory error detector
==15884== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==15884== Using Valgrind-3.18.1-42b08ed5bd-20211015 and LibVEX; rerun with -h for copyright info
==15884== Command: zmc -m 1
==15884== 
--15884-- Valgrind options:
--15884--    -v
--15884--    --leak-check=full
--15884-- Contents of /proc/version:
--15884--   Linux version 4.14.24-qnap (root@Q64-100-75) (gcc version 4.9.2 (toolchain config: [gcc-4.9.2 binutils-2.25 glibc-2.21])) #1 SMP Wed Aug 24 05:56:44 CST 2022
--15884-- 
--15884-- Arch and hwcaps: AMD64, LittleEndian, amd64-cx16-lzcnt-rdtscp-sse3-ssse3-avx-avx2-bmi-f16c-rdrand-rdseed
--15884-- Page sizes: currently 4096, max supported 4096
--15884-- Valgrind library directory: /usr/libexec/valgrind
--15884-- Reading syms from /usr/bin/zmc
--15884-- Reading syms from /usr/lib/x86_64-linux-gnu/ld-linux-x86-64.so.2
--15884--   Considering /usr/lib/debug/.build-id/61/ef896a699bb1c2e4e231642b2e1688b2f1a61e.debug ..
--15884--   .. build-id is valid
--15884-- Reading syms from /usr/libexec/valgrind/memcheck-amd64-linux
--15884--    object doesn't have a symbol table
--15884--    object doesn't have a dynamic symbol table
--15884-- Scheduler: using generic scheduler lock implementation.
--15884-- Reading suppressions file: /usr/libexec/valgrind/default.supp
==15884== embedded gdbserver: reading from /tmp/vgdb-pipe-from-vgdb-to-15884-by-???-on-ubuntu_22.04
==15884== embedded gdbserver: writing to   /tmp/vgdb-pipe-to-vgdb-from-15884-by-???-on-ubuntu_22.04
==15884== embedded gdbserver: shared mem   /tmp/vgdb-pipe-shared-mem-vgdb-15884-by-???-on-ubuntu_22.04
==15884== 
==15884== TO CONTROL THIS PROCESS USING vgdb (which you probably
==15884== don't want to do, unless you know exactly what you're doing,
==15884== or are doing some strange experiment):
==15884==   /usr/bin/vgdb --pid=15884 ...command...
==15884== 
==15884== TO DEBUG THIS PROCESS USING GDB: start GDB like this
==15884==   /path/to/gdb zmc
==15884== and then give GDB the following command
==15884==   target remote | /usr/bin/vgdb --pid=15884
==15884== --pid is optional if only one valgrind process is running
==15884== 
--15884-- REDIR: 0x402aa60 (ld-linux-x86-64.so.2:strlen) redirected to 0x580bcec2 (???)
--15884-- REDIR: 0x402a830 (ld-linux-x86-64.so.2:index) redirected to 0x580bcedc (???)
--15884-- Reading syms from /usr/libexec/valgrind/vgpreload_core-amd64-linux.so
--15884--    object doesn't have a symbol table
--15884-- Reading syms from /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so
--15884--    object doesn't have a symbol table
==15884== WARNING: new redirection conflicts with existing -- ignoring it
--15884--     old: 0x0402aa60 (strlen              ) R-> (0000.0) 0x580bcec2 ???
--15884--     new: 0x0402aa60 (strlen              ) R-> (2007.0) 0x0484ee30 strlen
--15884-- REDIR: 0x4027240 (ld-linux-x86-64.so.2:strcmp) redirected to 0x484fcd0 (strcmp)
--15884-- REDIR: 0x402afc0 (ld-linux-x86-64.so.2:mempcpy) redirected to 0x4853840 (mempcpy)
--15884-- Reading syms from /usr/lib/x86_64-linux-gnu/libz.so.1.2.11
--15884--    object doesn't have a symbol table
--15884-- Reading syms from /usr/lib/x86_64-linux-gnu/libjpeg.so.8.2.2
--15884--    object doesn't have a symbol table
--15884-- Reading syms from /usr/lib/x86_64-linux-gnu/libcrypto.so.3
--15884--    object doesn't have a symbol table
--15884-- Reading syms from /usr/lib/x86_64-linux-gnu/libpcre.so.3.13.3
--15884--    object doesn't have a symbol table
--15884-- Reading syms from /usr/lib/x86_64-linux-gnu/libmysqlclient.so.21.2.31
--15884--    object doesn't have a symbol table
--15884-- Reading syms from /usr/lib/x86_64-linux-gnu/libavformat.so.58.76.100
--15884--    object doesn't have a symbol table
--15884-- Reading syms from /usr/lib/x86_64-linux-gnu/libavcodec.so.58.134.100
--15884--    object doesn't have a symbol table
--15884-- Reading syms from /usr/lib/x86_64-linux-gnu/libavutil.so.56.70.100
--15884--    object doesn't have a symbol table
--15884-- Reading syms from /usr/lib/x86_64-linux-gnu/libswscale.so.5.9.100
--15884--    object doesn't have a symbol table
[... 644 lines cut ...]
==15884==    by 0x12AFF2: zmDbConnect() (in /usr/bin/zmc)
==15884==    by 0x120499: main (in /usr/bin/zmc)
==15884== 
==15884== 16,416 bytes in 1 blocks are possibly lost in loss record 764 of 768
==15884==    at 0x4848899: malloc (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==15884==    by 0x4E434DF: ??? (in /usr/lib/x86_64-linux-gnu/libmysqlclient.so.21.2.31)
==15884==    by 0x4E3AF50: ??? (in /usr/lib/x86_64-linux-gnu/libmysqlclient.so.21.2.31)
==15884==    by 0x4E3D4BD: ??? (in /usr/lib/x86_64-linux-gnu/libmysqlclient.so.21.2.31)
==15884==    by 0x4DF8680: ??? (in /usr/lib/x86_64-linux-gnu/libmysqlclient.so.21.2.31)
==15884==    by 0x4DFAFA3: ??? (in /usr/lib/x86_64-linux-gnu/libmysqlclient.so.21.2.31)
==15884==    by 0x4DF6FC7: mysql_real_connect (in /usr/lib/x86_64-linux-gnu/libmysqlclient.so.21.2.31)
==15884==    by 0x12AFF2: zmDbConnect() (in /usr/bin/zmc)
==15884==    by 0x120499: main (in /usr/bin/zmc)
==15884== 
==15884== LEAK SUMMARY:
==15884==    definitely lost: 0 bytes in 0 blocks
==15884==    indirectly lost: 0 bytes in 0 blocks
==15884==      possibly lost: 27,896 bytes in 23 blocks
==15884==    still reachable: 8,520,916 bytes in 1,645 blocks
==15884==         suppressed: 0 bytes in 0 blocks
==15884== Reachable blocks (those to which a pointer was found) are not shown.
==15884== To see them, rerun with: --leak-check=full --show-leak-kinds=all
==15884== 
==15884== ERROR SUMMARY: 23 errors from 23 contexts (suppressed: 0 from 0)
```

connortechnology, 2022-10-31:
I don't know why you keep posting leak check output instead of ZoneMinder's debug output which might actually contain useful info. 

Signal 7 normally occurs when you are trying to allocate more SHM  than exists.  What is your ImageBuffers set to?  In 1.34 and below it had to be high.  Now it should be 3.

radoslavv, 2022-10-31:
as ZM debug did not show anything:
```
root@ubuntu_22:/var/log/zm# addr2line -e /usr/bin/zmc 0x5652f0ef01dc 0x7f1d7ea24520 0x7f1d7eb831ca 0x5652f0ebed3d 0x5652f0e92742 0x7f1d7ea0bd90 0x7f1d7ea0be40 0x5652f0e94c95
??:0
??:0
??:0
??:0
??:0
??:0
??:0
??:0
```
if any other output needed, please suggest

SteveGilvarry, 2022-10-31:
https://zoneminder.readthedocs.io/en/latest/userguide/options/options_logging.html

Turn on debug logging for _zmc_m1 target, details mostly explained in docs above.

radoslavv, 2022-10-31:
DEBUG logging into file for 1.36.31 - still cycled same messages (IP camera is on):
```
10/31/22 14:26:40.826918 zmc_m1[513].DB1-zm_logger.cpp/224 [LogOpts: level=DB9 effective=DB9, screen=OFF, database=INF, logfile=DB9->zm_dbg_all.log, syslog=WAR]
10/31/22 14:26:40.826989 zmc_m1[513].DB1-zm_utils.cpp/198 [Detected a x86\x86-64 processor with AVX2]
10/31/22 14:26:40.827277 zmc_m1[513].DB5-zm_db.cpp/166 [Success]
10/31/22 14:26:40.827298 zmc_m1[513].DB4-zm_image.cpp/396 [Blend: Using SSE2 fast blend function]
10/31/22 14:26:40.827309 zmc_m1[513].DB4-zm_image.cpp/458 [Delta: Using SSSE3 delta functions]
10/31/22 14:26:40.827318 zmc_m1[513].DB4-zm_image.cpp/559 [Deinterlace: Using standard functions]
10/31/22 14:26:40.827326 zmc_m1[513].DB4-zm_image.cpp/572 [Image buffer copy: Using standard memcpy]
10/31/22 14:26:40.827521 zmc_m1[513].DB1-zm_storage.cpp/91 [No id passed to Storage constructor.  Using default path /var/cache/zoneminder/events instead]
10/31/22 14:26:40.827544 zmc_m1[513].DB1-zm_monitor.cpp/493 [Have camera type Ffmpeg]
10/31/22 14:26:40.827558 zmc_m1[513].DB1-zm_monitor.cpp/2215 [Reloading linked monitors for monitor PIR, '(null)']
10/31/22 14:26:40.827569 zmc_m1[513].DB3-zm_monitor.cpp/570 [Decoding enabled: 1 function 5 Mocord savejpegs 3 videowriter 0]
10/31/22 14:26:40.827587 zmc_m1[513].DB1-zm_packetqueue.cpp/714 [Setting pre_event_video_packet_count to 5]
10/31/22 14:26:40.827596 zmc_m1[513].DB1-zm_packetqueue.cpp/708 [Setting max_video_packet_count to 0]
10/31/22 14:26:40.827605 zmc_m1[513].DB1-zm_monitor.cpp/658 [mem.size(8) SharedData=760 TriggerData=560 VideoStoreData=4128 timestamps=48 images=3x33177600 = 99532800 total=99538360]
10/31/22 14:26:40.827615 zmc_m1[513].DB2-zm_camera.cpp/68 [New camera id: 1 width: 3840 line size: 15360 height: 2160 colours: 4 subpixelorder: 8 capture: 1]
10/31/22 14:26:40.827629 zmc_m1[513].INF-zm_ffmpeg.cpp/84 [Enabling ffmpeg logs, as LOG_DEBUG+LOG_FFMPEG are enabled in options]
10/31/22 14:26:40.827679 zmc_m1[513].DB3-zm_monitor.cpp/2207 [Reloading zones for monitor PIR have 0]
10/31/22 14:26:40.849712 zmc_m1[514].DB1-zm_db.cpp/190 [Success running sql query INSERT INTO `Logs` ( `TimeKey`, `Component`, `ServerId`, `Pid`, `Level`, `Code`, `Message`, `File`, `Line` ) VALUES ( 1667222800.827629, 'zmc_m1', 0, 513, 0, 'INF', 'Enabling ffmpeg logs, as LOG_DEBUG+LOG_FFMPEG are enabled in options', 'zm_ffmpeg.cpp', 84 )]
10/31/22 14:26:40.849992 zmc_m1[513].DB1-zm_zone.cpp/839 [Got 1 zones for monitor PIR]
10/31/22 14:26:40.850019 zmc_m1[513].DB5-zm_zone.cpp/870 [Parsing polygon 0,0 3839,0 3839,2159 0,2159]
10/31/22 14:26:40.850029 zmc_m1[513].DB3-zm_zone.cpp/765 [Got coordinate 0,0 from polygon string]
10/31/22 14:26:40.850038 zmc_m1[513].DB3-zm_zone.cpp/765 [Got coordinate 3839,0 from polygon string]
10/31/22 14:26:40.850046 zmc_m1[513].DB3-zm_zone.cpp/765 [Got coordinate 3839,2159 from polygon string]
10/31/22 14:26:40.850054 zmc_m1[513].DB3-zm_zone.cpp/765 [Got coordinate 0,2159 from polygon string]
10/31/22 14:26:40.850061 zmc_m1[513].DB3-zm_zone.cpp/777 [Successfully parsed polygon string 0,2159]
10/31/22 14:26:40.852752 zmc_m1[513].DB1-zm_image.cpp/2486 [vertices 4]
10/31/22 14:26:40.887223 zmc_m1[513].DB1-zm_monitor.cpp/2209 [Reloading zones for monitor PIR have 1]
10/31/22 14:26:40.887267 zmc_m1[513].DB1-zm_monitor.cpp/694 [Loaded monitor 1(PIR), 1 zones]
10/31/22 14:26:40.887298 zmc_m1[513].DB2-zmc.cpp/217 [1 monitors loaded]
10/31/22 14:26:40.887307 zmc_m1[513].INF-zmc.cpp/220 [Starting Capture version 1.36.31]
10/31/22 14:26:40.887328 zmc_m1[513].DB3-zm_monitor.cpp/917 [Connecting to monitor.  Purpose is 1]
10/31/22 14:26:40.887343 zmc_m1[513].DB3-zm_monitor.cpp/930 [Success opening mmap file at (/dev/shm/zm.mmap.1)]
10/31/22 14:26:40.887353 zmc_m1[513].DB3-zm_monitor.cpp/963 [MMap file size is 99538360]
10/31/22 14:26:40.887364 zmc_m1[513].DB1-zm_monitor.cpp/968 [Unable to map file /dev/shm/zm.mmap.1 (99538360 bytes) to locked memory, trying unlocked]
10/31/22 14:26:40.887375 zmc_m1[513].DB1-zm_monitor.cpp/971 [Mapped file /dev/shm/zm.mmap.1 (99538360 bytes) to unlocked memory]
10/31/22 14:26:40.887383 zmc_m1[513].DB3-zm_monitor.cpp/1004 [Aligning shared memory images to the next 64 byte boundary]
10/31/22 14:26:40.887392 zmc_m1[513].DB3-zm_monitor.cpp/1014 [Allocated 3 3 image buffers]
```

(btw found docker image zoneminderhq/zoneminder:latest-ubuntu18.04 with ZM version 1.36.4 and there capturing from IP camera works correctly)

radoslavv, 2022-10-31:
also tried GDB
```
(gdb@sourceware.org) or on GDB's IRC channel (#gdb on Freenode).
root@ubuntu_22:/# gdb --args zmc "-m 1"
GNU gdb (Ubuntu 12.1-0ubuntu1~22.04) 12.1
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from zmc...
(No debugging symbols found in zmc)
(gdb) run
Starting program: /usr/bin/zmc -m\ 1
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffed8f5640 (LWP 22201)]

Thread 1 "zmc" received signal SIGBUS, Bus error.
__memset_avx2_unaligned_erms () at ../sysdeps/x86_64/multiarch/memset-vec-unaligned-erms.S:351
351     ../sysdeps/x86_64/multiarch/memset-vec-unaligned-erms.S: No such file or directory.
```
seems similar issue like here [zmc -m 1 exited abnormally, exit status 7 #3132](https://github.com/ZoneMinder/zoneminder/issues/3132)