ZoneMinder/zoneminder #3617: GUI version 1.37.25 vs released 1.36.31

radoslavv, 2022-10-30:
**Describe Your Environment**
- Version of ZoneMinder: **1.37.25** vs [1.36.31](https://github.com/ZoneMinder/zoneminder/releases/tag/1.36.31)
- How you installed ZoneMinder: **git + compile from source**
- Full name and version of OS: **Ubuntu 20.04**
- Browser name and version: **irrelevant**

**If the issue concerns a camera**
irrelevant

**Describe the bug**
different version in release notes [1.36.31](https://github.com/ZoneMinder/zoneminder/releases/tag/1.36.31)
[image]
different version shown in GUI 1.37.25
[image]
+ Log 1.37.25
[image]
Unclear which one is valid

**To Reproduce**
compile from sources 

**Expected behavior**
identical version in GUI and release notes

**Debug Logs**
not applicable

connortechnology, 2022-10-30:
1.36.31 is the stable release.
1.37.25 is the latest development version, which has NOT been released.  

If you want to follow stable, follow the tags, or the release-1.36 branch.

closed 2022-10-30 by connortechnology

radoslavv, 2022-10-30:
OK, thx, clear now