ZoneMinder/zoneminder #3618: zm config files placed in /etc instead of /etc/zm

radoslavv, 2022-10-30:
**Describe Your Environment**
- Version of ZoneMinder: **1.37.25 (shown in web GUI)**
- How you installed ZoneMinder: **git + compile from sources**
- Full name and version of OS: **Ubuntu 22.04**
- Browser name and version (if this is an issue with the web interface)

**If the issue concerns a camera**
irrelevant

**Describe the bug**
zm.conf file and conf.d files placed in /etc/ instead of /etc/zm/

**To Reproduce**
compile from sources
1. cmake .
2. make
3. make install

check files location
```
root@ubuntu_22:/# ls -la /etc/zm.conf /etc/conf.d
-rw-r--r-- 1 root root 1855 Oct 23 22:19 /etc/zm.conf

/etc/conf.d:
total 75
drwxr-xr-x  2 root root        6 Oct 30 01:39 .
drwxr-xr-x 67 root root      136 Oct 30 02:21 ..
-rw-r--r--  1 root root     1989 Oct 30 01:39 01-system-paths.conf
-rw-r--r--  1 root root      395 Oct 23 22:16 02-multiserver.conf
-rw-r--r--  1 root root      740 Oct 23 22:16 README
-rw-r-----  1 root www-data  199 Oct 28 20:29 zmcustom.conf
```
also in zm.conf file is not /etc/zm path
```
root@ubuntu_22:/# cat /etc/zm.conf |grep -B1 ZM_PATH_CONF
# Path to ZoneMinder configuration (this file only at present)
ZM_PATH_CONF=/etc
```
**Expected behavior**
all have to be placed in folder /etc/zm
/etc/zm
/etc/zm/zm.conf
/etc/zm/conf.d/*

**Debug Logs**
not applicable

connortechnology, 2022-10-30:
You have not configured cmake to put it where you want it.  This is not a bug. If you want to build something similar to our debian/ubuntu packages, use: 

cmake -DCMAKE_INSTALL_PREFIX=/usr \
  -DBUILD_MAN=OFF \
  -DCMAKE_SKIP_RPATH=ON \
  -DCMAKE_VERBOSE_MAKEFILE=OFF \
  -DCMAKE_COLOR_MAKEFILE=ON \
  -DZM_RUNDIR=/var/run/zm \
  -DZM_SOCKDIR=/var/run/zm \
  -DZM_TMPDIR=/var/tmp/zm \
  -DZM_LOGDIR=/var/log/zm \
  -DZM_WEBDIR=/usr/share/zoneminder/www \
  -DZM_CONTENTDIR=/var/cache/zoneminder \
  -DZM_PATH_ZMS=/zm/cgi-bin/zms \
  -DZM_CGIDIR=/usr/lib/zoneminder/cgi-bin \
  -DZM_CACHEDIR=/var/cache/zoneminder/cache \
  -DZM_WEB_USER=www-data \
  -DZM_WEB_GROUP=www-data \
  -DCMAKE_INSTALL_SYSCONFDIR=etc/zm \
  -DZM_CONFIG_DIR=/etc/zm

closed 2022-10-30 by connortechnology

radoslavv, 2022-10-30:
OK,
`-DZM_CONFIG_DIR=/etc/zm`
is important, thx