ZoneMinder/zoneminder #3619: ZM_PATH_ZMS=/cgi-bin/nph-zms instead of ZM_PATH_ZMS=/zm/cgi-bin/nph-zms

radoslavv, 2022-10-30:
**Describe Your Environment**
- Version of ZoneMinder: **1.37.25** (shown in web GUI)
- How you installed ZoneMinder: git + sources compile
- Full name and version of OS: **Ubuntu 22.04**
- Browser name and version: irrelevant

**If the issue concerns a camera**
irrelevant

**Describe the bug**
incorrect path in 01-system-paths.conf, missing /zm prefix in path

**To Reproduce**
compile from sources
1. cmake .
2. make
3. make install

check 01-system-paths.conf for ZM_PATH_ZMS
```
root@ubuntu_22:/etc/conf.d# cat 01-system-paths.conf | grep -B1 ZM_PATH_ZMS
# ZoneMinder url path to the zms streaming server
ZM_PATH_ZMS=/cgi-bin/nph-zms
```
**Expected behavior**
prefix /zm to be part of ZM_PATH_ZMS
```
# ZoneMinder url path to the zms streaming server
ZM_PATH_ZMS=/zm/cgi-bin/nph-zms
```

**Debug Logs**
not applicable

connortechnology, 2022-10-30:
Kindly stop spamming our issues with your lack of configuration.

closed 2022-10-30 by connortechnology

radoslavv, 2022-10-30:
Apologies if you are considering this as spamming, in default configuration binary is in right place, but incorrect in config file.
I would say it is a bug.

connortechnology, 2022-10-30:
No.  You need to tell it where you want things.  Every distro has different thoughts on the matter.