ZoneMinder/zoneminder #3620: Dependency on PHP 7 -- EOL, No security support after November 2022

jeffsf, 2022-10-31:
Installing ZoneMinder through apt results in a dependency on PHP7, which is EOLed and about to fall out of even security support.

```
jeff@deb-devel:~$ cat /etc/os-release 
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```

```
jeff@deb-devel:~$ sudo apt install zoneminder
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following additional packages will be installed:
  cakephp cakephp-scripts fonts-glyphicons-halflings libapache2-mod-php libapache2-mod-php7.4 libappconfig-perl libb64-0d
  libclass-inspector-perl libclass-load-perl libclass-mix-perl libclass-std-fast-perl libclass-std-perl
  libconvert-binhex-perl libcrypt-eksblowfish-perl libcrypt-rijndael-perl libdata-entropy-perl libdata-float-perl
  libdevice-serialport-perl libemail-date-format-perl libhttp-lite-perl libimage-base-bundle-perl libimage-info-perl
  libio-interface-perl libio-pty-perl libio-sessiondata-perl libio-socket-multicast-perl libjwt-gnutls0 libmime-lite-perl
  libmime-tools-perl libmime-types-perl libnet-sftp-foreign-perl libnumber-bytes-human-perl libossp-uuid-perl libossp-uuid16
  libphp-serialization-perl libsoap-lite-perl libsoap-wsdl-perl libsys-cpu-perl libsys-meminfo-perl libsys-mmap-perl
  libtask-weaken-perl libtemplate-perl liburi-encode-perl libxmlrpc-lite-perl php-apcu php-apcu-bc php-cli php-common php-gd
  php-mysql php7.4-cli php7.4-common php7.4-gd php7.4-json php7.4-mysql php7.4-opcache php7.4-readline zoneminder-doc
Suggested packages:
  php-mcrypt php-pgsql | php5-pgsql php-sqlite3 | php5-sqlite php-pear libmojolicious-perl uuid libapache2-mod-perl2
  libnet-jabber-perl libtemplate-plugin-gd-perl libtemplate-plugin-xml-perl fcgiwrap
The following NEW packages will be installed:
  cakephp cakephp-scripts fonts-glyphicons-halflings libapache2-mod-php libapache2-mod-php7.4 libappconfig-perl libb64-0d
  libclass-inspector-perl libclass-load-perl libclass-mix-perl libclass-std-fast-perl libclass-std-perl
  libconvert-binhex-perl libcrypt-eksblowfish-perl libcrypt-rijndael-perl libdata-entropy-perl libdata-float-perl
  libdevice-serialport-perl libemail-date-format-perl libhttp-lite-perl libimage-base-bundle-perl libimage-info-perl
  libio-interface-perl libio-pty-perl libio-sessiondata-perl libio-socket-multicast-perl libjwt-gnutls0 libmime-lite-perl
  libmime-tools-perl libmime-types-perl libnet-sftp-foreign-perl libnumber-bytes-human-perl libossp-uuid-perl libossp-uuid16
  libphp-serialization-perl libsoap-lite-perl libsoap-wsdl-perl libsys-cpu-perl libsys-meminfo-perl libsys-mmap-perl
  libtask-weaken-perl libtemplate-perl liburi-encode-perl libxmlrpc-lite-perl php-apcu php-apcu-bc php-cli php-common php-gd
  php-mysql php7.4-cli php7.4-common php7.4-gd php7.4-json php7.4-mysql php7.4-opcache php7.4-readline zoneminder
  zoneminder-doc
0 upgraded, 59 newly installed, 0 to remove and 2 not upgraded.
Need to get 18.0 MB of archives.
After this operation, 70.8 MB of additional disk space will be used.
Do you want to continue? [Y/n] 
```

https://www.php.net/supported-versions.php

jeffsf, 2022-10-31:
Note that there are recommendations for php5 packages, suggesting a very outdated set of dependencies in the build.

connortechnology, 2022-10-31:
ZoneMinder depends on php packages. We don't specify version.  bullseye comes with 7.4 and always will.  You don't mention what version of zoneminder you are installing, but if I was to assume the package that comes with debian bullseye, then that is Debian's problem, not ours.

closed 2022-12-06 by connortechnology