ZoneMinder/zoneminder #3622: ZoneMinder 1.36.30+ No Longer Works On CentOS 7

gandihar, 2022-10-31:
**Describe Your Environment**
- CentOS 7 (Latest), setup as per Zoneminder docs.
  - httpd-2.4.6-97.el7.centos.5.x86_64
  - php-5.4.16-48.el7.x86_64
  - php-cli-5.4.16-48.el7.x86_64
  - php-common-5.4.16-48.el7.x86_64
  - php-gd-5.4.16-48.el7.x86_64
  - php-intl-5.4.16-48.el7.x86_64
  - php-mysql-5.4.16-48.el7.x86_64
  - php-pdo-5.4.16-48.el7.x86_64
  - php-pear-1.9.4-22.el7.noarch
  - php-pecl-apcu-4.0.11-1.el7.x86_64
  - php-process-5.4.16-48.el7.x86_64
  - php-xml-5.4.16-48.el7.x86_64
- Zoneminder 1.36.30+ (Tested on .31) installed from RPMFusion repo
  - zoneminder-1.36.31-1.el7.x86_64
  - zoneminder-common-1.36.31-1.el7.x86_64
  - zoneminder-httpd-1.36.31-1.el7.x86_64
- Any browser

**Describe the bug**
After upgrading to ZoneMinder 1.36.30+ (I went from .28 to .31) the ZoneMinder web interface is entirely blank. As in a single blank line is returned to the browser for any page requested. After digging through Apache and ZoneMinder error logs, it looks like the new PHP syntax added to /usr/share/zoneminder/www/includes/session.php is using newer PHP syntax (PHP 7+) than what is shipped with CentOS 7 (PHP 5.4). 

**To Reproduce**
1. Install ZoneMinder on CentOS 7 as per docs.
2. Try to hit the web interface.
3. Blank page.
4. Web server error logs show something like this:
```
2022-10-31 15:46:49 Error: Fatal Error (4): syntax error, unexpected ':', expecting ';' or '{' in [/usr/share/zoneminder/www/includes/session.php, line 129]
```

**Expected behavior**
The ZoneMinder login page or console.

**Hacky Work-around**
Going through /usr/share/zoneminder/www/includes/session.php and removing the PHP 7 function return hints.
i.e. `public function open($path, $name): bool {` becomes `public function open($path, $name) {`

If PHP 5 is being deprecated, please update the docs to include SCL instructions for installing PHP 7 on CentOS 7. If not, please remove PHP 7+ function return hints.

connortechnology, 2022-12-06:
This should have been fixed in 1.36.32.

gandihar, 2022-12-06:
Confirmed working. TYVM

commit f20ef49c (2022-11-01): Remove return type from session class methods.  not supported in php5.4 Fixes #3622