ZoneMinder/zoneminder #3624: Error in package for el8 in zmrepo

pavkamlc, 2022-11-05:
**Describe Your Environment**
- Version of ZoneMinder [1.37.25-1.169.20221104gitg59898b9a0.el8]
- How you installed ZoneMinder [zmrepo]
- Full name and version of OS [Rocky 8]

**Describe the bug**
After upgrade to this version i get this error:

[05-Nov-2022 12:52:00 Europe/Prague] PHP Warning:  require_once(Monitor_Permission.php): failed to open stream: No such file or directory in /usr/share/zoneminder/www/includes/User.php on line 6
[05-Nov-2022 12:52:00 Europe/Prague] PHP Fatal error:  require_once(): Failed opening required 'Monitor_Permission.php' (include_path='.:/usr/share/pear:/usr/share/php') in /usr/share/zoneminder/www/includes/User.php on line 6

**Solution**
When I download Monitor_Permission.php into includes folder from git, ZM is running now.

kb9mfd, 2022-11-05:
Same for el7, did the same actions to fix.

connortechnology, 2022-12-06:
I think we can close this, as 1.37.27 includes this file,

closed 2022-12-06 by connortechnology