ZoneMinder/zoneminder #3625: Can no longer view monitors after upgrade

kb9mfd, 2022-11-05:
**Describe Your Environment**
- Version of ZoneMinder [release version, development version, or commit]  - 1.37.25
- How you installed ZoneMinder [e.g. PPA, RPMFusion, from-source, etc] - zmrepo
- Full name and version of OS - CentOS Linux release 7.9.2009 (Core)
- Browser name and version (if this is an issue with the web interface) - any

**Describe the bug**
I ran yum to update my system and it upgraded zoneminder to 1.37.25 and completely broke it. Unfortunately I did not write down what the previous version was, but the update script said it updated the database from 1.37.1 to1.37.25. I had to find Monitor_Permissions.php in the GIT repository as it's not in the rpm. I tried to reinstall the rpm and downloaded the SRPM and that file is in neither. I also had to run the zm_update-1.x.x.sql files manually to fix a large number of missing table fields. Once the interface is up, I still have no monitors, it is recording and I can view past recordings, but I cannot view anything live. The  log shows "Insufficient privileges for request user 1 admin for monitor x" from the zms.cpp. I check the user for permissions and set every available field to "Edit" and verified the sql tables Groups_Permissions and Monitors_Permissions and the fields are set to 'Edit'. In the debug log it shows the queries are returning "Inherit" instead of "Edit" and not giving permission to view the monitors. I cannot seem to locate any idea why, here is an example from the log - 

11/5/22, 12:54:25 PM CDT | zms_m4 |   | 11800 | ERR | Insufficient privileges for request user 1 admin for monitor 4 | zms.cpp | 44
11/5/22, 12:54:25 PM CDT | zms_m4 |   | 11800 | DB1 | INHERIT from group_permission | zm_user.cpp | 133
11/5/22, 12:54:25 PM CDT | zms_m4 |   | 11800 | DB1 | # of Group_Permissions 1 | zm_user.cpp | 76
11/5/22, 12:54:25 PM CDT | zms_m4 |   | 11800 | DB1 | INHERIT from monitor_permission | zm_user.cpp | 104
11/5/22, 12:54:25 PM CDT | zms_m4 |   | 11800 | DB1 | # of Monitor_Permissions 6 | zm_user.cpp | 71
11/5/22, 12:54:25 PM CDT | zms_m4 |   | 11800 | DB1 | Authenticated user 'admin' | zm_user.cpp | 298

in the SQL table - 

MariaDB [zm]> select * from Monitors_Permissions;
+----+-----------+--------+------------+
| Id | MonitorId | UserId | Permission |
+----+-----------+--------+------------+
|  1 |         1 |      1 | Edit       |
|  2 |         2 |      1 | Edit       |
|  3 |         3 |      1 | Edit       |
|  4 |         4 |      1 | Edit       |
|  5 |         5 |      1 | Edit       |
|  6 |         6 |      1 | Edit       |
+----+-----------+--------+------------+

MariaDB [zm]> select * from Groups_Permissions;
+----+---------+--------+------------+
| Id | GroupId | UserId | Permission |
+----+---------+--------+------------+
|  1 |       1 |      1 | Edit       |
+----+---------+--------+------------+

MariaDB [zm]> select * from Users;
+----+----------+--------------------------------------------------------------+----------+---------+--------+--------+---------+----------+--------+---------+-----------+--------+--------------+-------------+----------------+------------+----------+
| Id | Username | Password                                                     | Language | Enabled | Stream | Events | Control | Monitors | Groups | Devices | Snapshots | System | MaxBandwidth | MonitorIds  | TokenMinExpiry | APIEnabled | HomeView |
+----+----------+--------------------------------------------------------------+----------+---------+--------+--------+---------+----------+--------+---------+-----------+--------+--------------+-------------+----------------+------------+----------+
|  1 | admin    | XXXXXXXXX | en_us    |       1 | View   | Edit   | Edit    | Edit     | Edit   | Edit    | Edit      | Edit   | high         | 1,2,3,4,5,6 |              0 |          1 |          |
+----+----------+--------------------------------------------------------------+----------+---------+--------+--------+---------+----------+--------+---------+-----------+--------+--------------+-------------+----------------+------------+----------+

```

grabbman, 2022-11-06:
I had the same issue today after updating vom 1.37.18 to 1.37.27. After long search and also some inspection in mysql, decided to play with groups and user-permission. After I changed inherit to edit on all monitor permissions under Options - Users my monitor showed up again.

kb9mfd, 2022-11-06:
> I had the same issue today after updating vom 1.37.18 to 1.37.27. After long search and also some inspection in mysql, decided to play with groups and user-permission. After I changed inherit to edit on all monitor permissions under Options - Users my monitor showed up again.

I also did that, as you can see by the SQL, monitors are all set to Edit, still no-go. No idea why, you would think that should of fixed it.

grabbman, 2022-11-06:
I have just rechecked, and I had Monitor_Permissions only for another user before update. 
But I think the problem is Groups, as I get an error for this table and so I think it will not give the access to the "default-group" because I had no groups before:

`mysql> select * from Groups;
ERROR 1064 (42000): You have an error in your SQL syntax; check the manual that corresponds to your MySQL server version for the right syntax to use near 'Groups' at line 1
`

kb9mfd, 2022-11-06:
> I have just rechecked, and I had Monitor_Permissions only for another user before update. But I think the problem is Groups, as I get an error for this table and so I think it will not give the [...]

I actually do have a group, here is my output -

```
MariaDB [zm]> select * from Groups;
+----+---------+----------+
| Id | Name    | ParentId |
+----+---------+----------+
|  1 | Monitor |     NULL |
+----+---------+----------+
1 row in set (0.00 sec)
```

so, in Groups_Permissions I have permission for it (see in the original post), but when I look at the logs it says it returns as "Inherit" It seems like there is some kind of issue with the enum. Both in the c++ side in zm_user.cpp and in the sql table that column is set to be a enum, but apparently it's not returning correctly.

connortechnology, 2022-12-06:
When we do the SQL query, we do Select Permission+0.  This results in PERM_INHERIT being =1, etc.  

I think I may have fixed this already, although I am still tracking some weirdness on one server.  Can you update and report?

alien8r, 2022-12-12:
Is this part of the same after upgrade issue where the console screen won't show monitors but Cycle, Montage, and Montage Review continue to work?
Trying to get cameras to show up ZM produces this error:
SQL-ERR dbFetchAll no result, statement was 'SELECT Id FROM MonitorsWHERE Id IN (9,2,7,6,5,10,8,1,11)'

connortechnology, 2022-12-12:
We are not aware of any such issue.  Looks like that sql is missing a space between Monitors and WHERE.  Not sure where in the code that SQL is generated.

closed 2026-06-12 by SteveGilvarry