ZoneMinder/zoneminder #3627: "Cannot write to event folder", but events are written just fine. (Picking up #3080)

NormHenderson, 2022-11-07:
**Describe Your Environment**
- Version of ZoneMinder 1.37.18
- How you installed ZoneMinder PPA: deb http://ppa.launchpad.net/iconnor/zoneminder-master/ubuntu focal main
- Ubuntu 20.04.4 LTS

Zoneminder runs as www-data

ZM_DIR_EVENTS is /Security/events, /Security is mounted from a Synology NAS on top of /Media:
```
# mount |grep nfs4
10.20.84.101:/volume1/Media on /Media type nfs4 (rw,relatime,vers=4.1,rsize=131072,wsize=131072,namlen=255,hard,proto=tcp,timeo=600,retrans=2,sec=sys,clientaddr=10.20.84.202,local_lock=none,addr=10.20.84.101)
10.20.84.101:/volume1/Media/Security on /Security type nfs4 (rw,relatime,vers=4.1,rsize=131072,wsize=131072,namlen=255,hard,proto=tcp,timeo=600,retrans=2,sec=sys,clientaddr=10.20.84.202,local_lock=none,addr=10.20.84.101)

```
Directory ownership as seen from the server running ZM:
```
# ls -ld /Security
drwxrwxrwx 1 root root 24 Nov  7 15:53 /Security
# ls -l /Security
drwxrwxrwx 1 www-data www-data 140 Nov  7 15:55 events  (also rwxrwxrwx for all subdirectories and files)
drwxrwxrwx 1 www-data www-data   0 May 20  2017 images

```
*Describe the bug**
Log file is filling with many times:
11/7/22, 5:22:34 PM GMT+1 | web_php |   | 694740 | WAR | Cannot write to event folder /Security/events. Check that it exists and is owned by the web account user. | /usr/share/zoneminder/www/index.php | 171

However, events are being created for example:
drwxrwxrwx 1 www-data www-data  4450 Nov  7 17:22 1385923
drwxrwxrwx 1 www-data www-data  1686 Nov  7 17:22 1385925
drwxrwxrwx 1 www-data www-data  2324 Nov  7 17:22 1385926

**This bug appears similar to #3080 which wasn't analyzed, the submitter desisted.**

**To Reproduce**
Steps to reproduce the behavior:
I wish I knew ...

**Expected behavior**
No warning messages

**Debug Logs**

```
[zm_debug.log.655863.log](https://github.com/ZoneMinder/zoneminder/files/9953622/zm_debug.log.655863.log)
[zm_debug.log.655864.log](https://github.com/ZoneMinder/zoneminder/files/9953623/zm_debug.log.655864.log)
[zm_debug.log.694740.log](https://github.com/ZoneMinder/zoneminder/files/9953624/zm_debug.log.694740.log)
[zm_debug.log.802304.log](https://github.com/ZoneMinder/zoneminder/files/9953625/zm_debug.log.802304.log)
```

connortechnology, 2022-11-15:
Hi Norm,

I have to wonder if it is an apparmor or some other SELINUX restriction... 
I have used nfs4 storage in the past without major issue.  
I am also tempted to just remove that check.  It is old and meant to be useful, but if monitors can't save events, there will be lots of other errors in the logs.  

You might want to update.  1.37.18 is fairly old now. I have just pushed a change to remove that check so it shouldn't log anymore.

NormHenderson, 2022-11-16:
Thank you very much. I have upgraded using the PPA (1.37.27) and the messages seem to have stopped.

closed 2022-11-16 by NormHenderson