ZoneMinder/zoneminder #3628: Can't delete zones. Error: actions can no longer be performed without POST in apache log.

waynieack, 2022-11-09:
**Describe Your Environment**
- Version of ZoneMinder 1.37.25
- How you installed ZoneMinder PPA: deb-src http://ppa.launchpad.net/iconnor/zoneminder-master/ubuntu focal main
- Ubuntu 20.04.5 LTS
- apache2 2.4.41
- PHP Version 7.4.3
- Firefox 106.0.5

**Describe the bug**
Can't delete zones from the web console.

**To Reproduce**
Steps to reproduce the behavior:
1.  Open zoneminder web console
2. Click the zone number next to the monitor in the zone column to open the zone list for the monitor. 
3. Click the check box next to any zone, and click "DELETE". 
4. The zone is still there after clicking delete. 

**Expected behavior**
The zone should be deleted after clicking delete. 

**Debug Logs**
```

[Wed Nov 09 10:41:16.204236 2022] [php7:notice] [pid 234731] [client 192.168.1.133:61496] ERR [actions can no longer be performed without POST.], referer: http://192.168.1.35/zm/index.php?view=zones&mid=12

```

Optimaximal, 2022-11-15:
I'm also seeing this issue in v1.36.31

connortechnology, 2022-11-15:
Fixes are available in ppa:iconnor/zoneminder-proposed and will be included in 1.36.32 when released.

waynieack, 2022-11-16:
As a work around for version 1.37.25:

 I updated file "/www/skins/classic/views/zones.php",  line 52 from:

 `<form name="contentForm" id="contentForm" method="get" action="?">`

to

 `<form name="contentForm" id="contentForm" method="post" action="?">`

connortechnology, 2022-11-16:
Looks like I didn't cherry-pick the fix from release-1.36.  I have done so now. Today's build should work fine.

closed 2022-11-21 by connortechnology