ZoneMinder/zoneminder #3639: eval violated CSP script-src

Eldertoon, 2022-11-25:
**Describe Your Environment**
- Version of ZoneMinder: 1.34.23-1 amd64 from stable (bullseye) Debian
- How you installed ZoneMinder: .deb
- Full name and version of OS: Debian 11 bullseye
- Browser name and version: Firefox 107 64 bits

**Describe the bug**
tail web_js.log is filled with lines like:
11/25/22 17:09:35.942290 web_js[57181].ERR [x.x.x.x] [eval violated CSP script-src] at ?view=options line

They are only created when I am logged on the web interface

x.x.x.x: Firefox client IP
57181: /usr/sbin/apache2 -k start

**To Reproduce**
I am unsure where this issue comes from. I discovered this behavior after loosing a disk (which included /var/lib/mysql/ + /var/cache/zoneminder/). Everything is back up from scratch (all cameras recording, no issue browsing the /zm pages) but Log keep beeing red with a lot of those messages.

**Expected behavior**
No error? Note I also have another instance of ZoneMinder running and I did not find any difference (apache enabled modules, php&apache versions, ...).

**Debug Logs**
I am unsure what log could be interesting. I suspect I am missing a folder, a right or something in configuration.

connortechnology, 2022-11-25:
These are normally caused by a browser extension.  You might try disabling them and see if it goes away.

Also 1.34 is ancient and no longer supported.  Please upgrade to 1.36.  You will likely need to use our packages https://zmrepo.zoneminder.com to do so.

Eldertoon, 2022-11-25:
That would have been the last place I would have searched for!

I confirm, using Edge/Internet Explorer do not reproduce the issue, I share the same extension in Firefox and in Chrome...

Thank you a lot, and I will plan to move to 1.36 now.

closed 2022-11-25 by Eldertoon